Koenkk/zigbee2mqtt #806: Use of CC2530 et.al programmed with router firmware

mcsSolutions, 2019-01-04:
I am trying to install a router to evaluate behavior and understand how to grow beyond the 15 devices that are supported by the CC2531 as a coordinator.  I had initial problem of understanding how to program the GBAN RF TO USB (CC2530 CC2591) device.  With persistence and lots or reading in this forum able to overcome it.

soldered wires and jumpered to CC Debugger per pictures at https://github.com/Koenkk/zigbee2mqtt/issues/124
router firmware router-cc2530-cc2591-diag.hex for CC2530+RFX2401 from https://github.com/Koenkk/zigbee2mqtt/blob/4cbf9afcdee90a3983c5a208735a89a75e52bbde/docs/information/cc_sniffer_devices.md

No issues with the programming.

Removed and reinserted the GBAN card and observed COM3 appears. Started Termite to access COM3. No communications showing through COM3.

Used Xiaomi water leak sensor that was previously paired with CC2531 coordinator and tried pairing within inches of GBAN antennae.  Nothing still shows up on COM3 terminal window.

I captured my database.db at https://pastebin.com/naYCg8J0.  Best I can tell there is nothing about a router there and only the set of water leak sensors.

Reinserting the GBAN shows the D1 LED blink twice, but still nothing on COM3.  I assume it is at least alive, but not certain.

Do the wires to ground remain attached when using the device or should they be removed after flashing is complete?

How do I know if the router is working or not?

In the last several weeks I have lost the ability to follow hyperlinks in the forum.  If I click on the link that contains https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size, I always get directed to https://github.com/Koenkk/zigbee2mqtt.  This is the master source repository for zigbee2mqtt and then some high level description below.  I remember before that I could get to a Wiki and then link to other pages of interest.  Has something changed with navigation in github?

While I have this specific hurdle of the router operation, I also need to understand how to use the router to expand the size of the network, but unable to navigate to where I expect the information to be.

danpowell88, 2019-01-05:
Wiki has moved https://koenkk.github.io/zigbee2mqtt/

mcsSolutions, 2019-01-11:
Thank your for the link, but the primary question still exists.  How do I know a router is working?