Koenkk/zigbee2mqtt #807: Support of OpenWRT

Sindbad6, 2019-01-05:
Unsupported device
First check: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Sindbad6, 2019-01-05:
Hi,
I use my OpenWrt router as access points and I don't want to add other devices for a zigbee bridge. So I thought it would be useful to connect the cc2531 to the APs. Unfortunately I'm not common with the integration of software into OpenWrt. So is there somebody, who love this idea also and has experience with this?

JLFN, 2019-01-06:
https://github.com/Koenkk/zigbee2mqtt/issues/665

closed 2019-03-14 by stale[bot]