Koenkk/zigbee2mqtt #808: Osram Plug Smart+ turns itself off

florianj1, 2019-01-05:
My Zigbee network consists of two Osram Plug Smart+ and one OSRAM Flex RGBW. The RGB-Strip works fine but the Plugs turn off by itself approximately twice a day.

The log output of this situation where the Plug truns off for no reason looks like [this](https://pastebin.com/phaUYDeN)

Additional Info:
- zigbee2mqtt version 1.0.1 (docker)
- Distance between all devices: less than 2 meter

Koenkk, 2019-01-06:
Looking at your log, indeed no command is send by zigbee2mqtt. 

I see the linkquality is quite low, perhaps this could be the problem, does it also happen when having the plug close to the coordinator?

lolorc, 2019-01-06:
I have 5 osram plugs, 2 are very close to the coordinator, I've never observed such behavior.
Could it be a group related issue ? was the plug bound to another running coordinator/hub  ?

florianj1, 2019-01-06:
@Koenkk : I have installed one of the plugs right next to the coordinator to increase the signal strength. 

@lolorc  Both plugs are brand new, thus I assumed they aren't paired to another coordinator. In order to get a clean state I have created a new docker container with a new configuration.yaml and had reset both plugs by pressing the button until the relay toggles.

Tomorrow we will see if anything changes

Koenkk, 2019-01-09:
Perhaps an other device in the network is turning your device off. Could you sniff the traffic while this happens? (https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/how_tos/how_to_sniff_zigbee_traffic.md)

closed 2019-03-17 by stale[bot]

mpuff, 2019-08-06:
i have also now 3 osram smart plug, with the same issue, has some one found the reason?

florianj1, 2019-08-09:
A firmware update to "CC2531_MAX_STABILITY_20190315" improved the situation, but the issue is still present. I haven't tried a more recent firmware yet.