Koenkk/zigbee2mqtt #809: Unsupported device: Securifi Peanut Smart Plug

roger-, 2019-01-05:
I'm trying to get the [Securifi Peanut Smart Plug](https://www.amazon.com/Securifi-compatible-Remotely-Monitor-Appliances/dp/B00TC9NC82) working, primarily to act as a router to extend the range of my current setup, but zigbee2mqtt doesn't support it and reports the modelID as `undefined`.

[Log here](https://pastebin.com/5CJgWWFY).

I'm running the latest version of zigbee2mqtt with [this](https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex) firmware on a CC2531 USB stick. I have 6 Xiaomi sensors that are all connected and working.

A couple questions:

1) how do I add support for a modelID that's unknown?
2) will it still function as a router even though it's supported? Is it paired even though it's unsupported?
3) is there any reason to update my firmware (my network is small)? I tried the latest one before (after deleting the second to last line in the hex file and flashing with an Arduino) but it didn't work.

Thanks!

Koenkk, 2019-01-06:
1. Not, we first need a modelID, can you try repairing while having it close to the coordinator?
2. Yes, router functionality has nothing to do with zigbee2mqtt, it's part of zigbee itself.
3. no

roger-, 2019-01-06:
Will try, but how do I repair?

Koenkk, 2019-01-06:
- Stop zigbee2mqtt
- Remove all lines containing `0x000d6f000a798c7b` from `database.db`
- Start zigbee2mqtt
- Pair

Koenkk, 2019-01-07:
Could you add the modelId yourself?

```js
{
   "id":2,
   "type":"Router",
   "ieeeAddr":"0x000d6f000a798c7b",
   "nwkAddr":4957,
   "manufId":4098,
   "modelId": "Securifi",
   "manufName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�",
   "powerSource":"Unknown",
   "epList":[
      1
   ],
   "status":"online",
   "joinTime":1546791322,
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":0,
         "inClusterList":[
            0,
            1,
            3,
            4,
            5,
            6,
            2820,
            2821
         ],
         "outClusterList":[
            0,
            1,
            3,
            4,
            5,
            6,
            25,
            2820,
            2821
         ],
         "clusters":{
            "genBasic":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "zclVersion":1,
                  "stackVersion":0,
                  "manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�",
                  "powerSource":0
               }
            },
            "genPowerCfg":{
               "dir":{
                  "value":3
               },
               "attrs":{

               }
            },
            "genIdentify":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "identifyTime":0
               }
            },
            "genGroups":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "nameSupport":0
               }
            },
            "genScenes":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "count":0,
                  "currentScene":0,
                  "currentGroup":0,
                  "sceneValid":0,
                  "nameSupport":0
               }
            },
            "genOnOff":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "onOff":0
               }
            },
            "genOta":{
               "dir":{
                  "value":2
               },
               "attrs":{

               }
            },
            "haElectricalMeasurement":{
               "dir":{
                  "value":3
               },
               "attrs":{
                  "measurementType":8,
                  "acFrequency":0,
                  "acFrequencyMultiplier":1,
                  "acFrequencyDivisor":1,
                  "rmsVoltage":0,
                  "rmsCurrent":0,
                  "activePower":0,
                  "powerFactor":0,
                  "acVoltageMultiplier":180,
                  "acVoltageDivisor":39321,
                  "acCurrentMultiplier":72,
                  "acCurrentDivisor":39321,
                  "acPowerMultiplier":10252,
                  "acPowerDivisor":39321
               }
            },
            "haDiagnostic":{
               "dir":{
                  "value":3
               },
               "attrs":{

               }
            }
         }
      }
   },
   "_id":"vF2UsvBEuvxyRThs"
}
```

roger-, 2019-01-07:
To the database.db file? I tried adding a modelId to the three lines the device showed up in but it ignored them and created three new ones.

roger-, 2019-01-09:
Thanks, that does seem to work (I had to restart zigbee2mqtt a couple times though) when I manually press the "on" button:

```json
  zigbee2mqtt:info 1/9/2019, 1:49:59 AM Succesfully configured 0x000d6f000a798c7b 0x000d6f000a798c7b
  zigbee2mqtt:debug 1/9/2019, 1:50:21 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:info 1/9/2019, 1:50:21 AM MQTT publish: topic 'zigbee2mqtt/0x000d6f000a798c7b', payload '{"state":"ON","linkquality":70}'
  zigbee2mqtt:debug 1/9/2019, 1:50:21 AM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:debug 1/9/2019, 1:50:26 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:info 1/9/2019, 1:50:26 AM MQTT publish: topic 'zigbee2mqtt/0x000d6f000a798c7b', payload '{"state":"OFF","linkquality":68}'
  zigbee2mqtt:debug 1/9/2019, 1:50:26 AM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
```

However I also get this error related to power measurements:

```json
  zigbee2mqtt:debug 1/9/2019, 1:52:09 AM Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}'
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 1/9/2019, 1:52:09 AM Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsVoltage":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsVoltage":0}}'
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 1/9/2019, 1:52:09 AM Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"activePower":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"activePower":0}}'
  zigbee2mqtt:warn 1/9/2019, 1:52:09 AM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
```
I noticed that the converter you posted doesn't use the `acPowerMultiplier`, etc. fields. Not ssure if that's related the 0 values.

Also nothing shows up in home assistant via MQTT discovery.

Koenkk, 2019-01-09:
Home Assistant discovery requires some extra changes, when you connect something to the plug, is the power consumption still 0? (`{"activePower":0}}`)

henworth, 2019-01-09:
I've gotten this to work by adding `fz.SP120_power` to `fromZigbee`, so the full line looks like:

`fromZigbee: [fz.ignore_electrical_change, fz.SP120_power, fz.generic_state, fz.ignore_onoff_change],`

Also, adding `'PP-WHT-US': [configurations.switch, configurations.sensor_power],` to the `mapping` variable in `lib/extension/homeassistant.js` gets it to show up in home assistant.

I can now toggle it on/off from home assistant and toggling the switch on the device updates status in HA as well. However, the power fields in the sensor entity are just 0. Not sure if that means it isn't working or not supported or I don't have enough load on it. Either way, it's essentially working now!

roger-, 2019-01-10:
@henworth

Thanks, tried that and now it shows up in HA and the switch works! 

@Koenkk

About two and a half minutes after turning on the plug (with a load) I get this message:

```js
Jan 10 01:05:46 raspberrypi npm[32478]:   zigbee2mqtt:debug 1/10/2019, 1:05:46 AM Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0,"rmsVoltage":0,"activePower":0}}' of device 'PP-WHT-US' (0x000d6f000a798c7b)
Jan 10 01:05:46 raspberrypi npm[32478]:   zigbee2mqtt:info 1/10/2019, 1:05:46 AM MQTT publish: topic 'zigbee2mqtt/0x000d6f000a798c7b', payload '{"state":"ON","linkquality":89,"power":0,"current":0,"voltage":0}'
```

So the decoded (?) measurement data is all zeros. 

Where does the data get decoded? Is there somewhere I can see the pre-decoded messages to see if it's a decoding error?

Koenkk, 2019-01-10:
Can you try with:

```js
{
        zigbeeModel: ['PP-WHT-US'],
        model: 'PP-WHT-US',
        vendor: 'Securifi',
        description: 'Peanut smart plug',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.ignore_electrical_change, fz.generic_state, fz.ignore_onoff_change],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const onOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const rmsCurrent = {
                direction: 0, attrId: 1288, dataType: 33, minRepIntval: 0, maxRepIntval: 3, repChange: 0,
            };
            const electricalCfg = [rmsCurrent];
            const actions = [
                (cb) => device.foundation('genOnOff', 'configReport', [onOff], foundationCfg, cb),
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('haElectricalMeasurement', 'configReport', electricalCfg, foundationCfg, cb),
                (cb) => device.bind('haElectricalMeasurement', coordinator, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

henworth, 2019-01-10:
This message comes back:

```zigbee2mqtt:warn 2019-1-10 11:51:26 No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}'
  zigbee2mqtt:warn 2019-1-10 11:51:26 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 2019-1-10 11:51:29 No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}'
  zigbee2mqtt:warn 2019-1-10 11:51:29 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2019-01-10:
@henworth still no current, did you connect a device which consumes power?

henworth, 2019-01-10:
Yep, I have something connected to it.

roger-, 2019-01-10:
Where do `dataType`, `attrId`, etc. come from?

Koenkk, 2019-01-11:
dataType: https://github.com/zigbeer/zcl-id/blob/master/definitions/cluster_defs.json#L1187
attrId: https://github.com/zigbeer/zcl-id/blob/master/definitions/common.json#L219
direction always 0
min/max rep interval: minimum/maximum report interval in seconds
repChange: reportable change (0 doesn't require changes) of the value.

roger-, 2019-01-11:
Thanks, that helps a lot.

Is there a way to view the raw ZB packets? I’d like to know if the device is outputting zeros (perhaps due to old firmware).

roger-, 2019-01-20:
Looking a bit more at the data it seems that the power measurements may be missing.

It doesn't seem to be functioning as a router for my Xaomi devices though. Any ideas why?

Koenkk, 2019-01-22:
@roger- can you repair the xiaomi device while having it close to the plug. It is know that Xiaomi devices are picky about connecting to routers.

roger-, 2019-02-01:
I've tried but it doesn't seem to work. Does the USB coordinator have to be on (it was)? And is there a way to check if anything is connecting to the router?

Koenkk, 2019-02-02:
Yes the coordinator has to be one, otherwise a device cannot join the network. You can use the networkmap to check how devices are connected (https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html)

roger-, 2019-03-09:
I don't think this ever worked properly as a router and I believe that the firmware needs to be updated for measurements to be supported. Works fine as a switch though. 

Can we get the changes above merged as is? What about the missing ID? How can that be dealt with automatically?

Koenkk, 2019-03-09:
The missing ID is something that should be documented (like http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#gledopto). Could you create  a PR?

cycloneZ, 2019-04-13:
@henworth or @roger- 

Were you able to determine the cause of the 'No converter available' issue? The Peanut is now listed as a supported device, and though it did pair I ended up having to modify the database to add the model id manually. It now functions but the logs are flooded by the following every few seconds.

`zigbee2mqtt:warn 4/13/2019, 10:01:16 AM No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}'
zigbee2mqtt:warn 4/13/2019, 10:01:16 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

roger-, 2019-04-13:
For whatever reason yes you still have to add it manually. The measurements don't seem to work unless you upgrade the firmware on your plug, but that requires  a specific hub.

cycloneZ, 2019-04-14:
I don't have any way to upgrade the firmware on the device. Is that the only way to get all of the warning messages to stop appearing in the logs?

roger-, 2019-04-14:
You can probably remove the haElectricalMeasurement stuff from the relevant files (see above).

wixoff, 2019-04-23:
The Peanut Plug entry in devices.js might still need some work.  (After following the new documentation steps I added above that are now merged, they paired correctly and work partially.)

I have two firmware-upgraded plugs that **do not report** power measurements into Home Assistant.  However, Home Assistant has sensors for them that show only blank values, and the z2m log shows current measurements being received:

[image]

The switch operates perfectly and the LQI sensor looks good, but I only get blank values for "Watts":

[image]

The Z2M log has some interesting information.  When I have something plugged into the plug and it's turned on, I'll get log entries about **current** (not power), like this:

```
4/19/2019, 1:50:07 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":34}}' of device 'PP-WHT-US' (0x000d6f000593651d) of endpoint 1
4/19/2019, 1:50:08 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":34}}' of device 'PP-WHT-US' (0x000d6f000593651d) of endpoint 1
4/19/2019, 1:50:11 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":34}}' of device 'PP-WHT-US' (0x000d6f000593651d) of endpoint 1
4/19/2019, 1:50:13 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":34}}' of device 'PP-WHT-US' (0x000d6f000593651d) of endpoint 1
```
I grabbed the log entries above a few days ago.  With nothing plugged in I get "rmsCurrent":0 as expected.  Today I'm getting messages like this:

```
4/23/2019, 9:03:15 AM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}' of device 'PP-WHT-US' (0x000d6f000593651d) of endpoint 1
4/23/2019, 9:03:15 AM - warn: No converter available for 'PP-WHT-US' with cid haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":0}}'
4/23/2019, 9:03:15 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
4/23/2019, 9:03:16 AM - debug: Received zigbee message of type 'attReport' with data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":464}}' of device 'PP-WHT-US' (0x000d6f000549b3b7) of endpoint 1
4/23/2019, 9:03:16 AM - warn: No converter available for 'PP-WHT-US' with cid 'haElectricalMeasurement', type 'attReport' and data '{"cid":"haElectricalMeasurement","data":{"rmsCurrent":464}}'4/23/2019, 9:03:16 AM - warn: Please see: http://bit.ly/2IQeK2A.
```
etc.

In neither case is anything showing up in the Home Assistant sensor.

As you can see, I'm getting spammed with those reports and log entries every couple of seconds.  I'm not sure if this is slowing down my zigbee network, but it sure seems to lag after I added these plugs.

For that "rmsCurrent":34 value, I'm not sure what to make of it.  I have a Tradfri 8.6W bulb plugged in, turned up to 100%.  If I multiply 34 by a conversion factor of 0.001831 (a default value stolen from the SmartThings device handler at https://github.com/pakmanwg/smartthings-peanut-plug/blob/master/devicetypes/pakmanwg/peanut-plug.src/peanut-plug.groovy), RMS voltage (120), and the Power Factor of around 0.8 (typical for decent bulbs), I get 5.98, which is roughly in the ballpark (I should plug the bulb into my Kill-a-Watt to confirm, as 100% white on a 600lm RGBW bulb might not be the highest-current scenario).

That SmartThings handler seems to have a way of caluclating these conversion factors for current, voltage, and power from data provided by the device; see lines 105-138.  I don't see this in our devices.js.

It's also worth noting that the SmartThings is able to report voltage and power in addition to current.

Any thoughts on what needs to change here?  I'm no zigbee expert, but I'm happy to help.

j4c3, 2019-05-02:
I just picked one of these up as well, but haven't dug too much into it yet. 

At the moment, the devices.js file is looking for three fromZigbee converters, fz.ignore_electrical_change, fz.state, fz.ignore_onoff_change. 

The ignore_electrical_change has the right cid, but is looking at devChange types, not attReport.   So, we could easily ignore those reports/warnings in the log.  But that won't get the measurement data. 

When my device joined, in the .db file, there are entries for the things being used by smart things to determine the values.  I didn't do any firmware updates, but I also haven't tested with anything plugged into it.

"haElectricalMeasurement":{  
	"dir":{  
        "value":3
    },
    "attrs":{  
        "measurementType":8,
        "acFrequency":0,
        "acFrequencyMultiplier":1,
        "acFrequencyDivisor":1,
        "rmsVoltage":0,
        "rmsCurrent":0,
        "activePower":0,
        "powerFactor":0,
        "acVoltageMultiplier":180,
        "acVoltageDivisor":39321,
        "acCurrentMultiplier":72,
        "acCurrentDivisor":39321,
        "acPowerMultiplier":10252,
        "acPowerDivisor":39321
    }

wixoff, 2019-05-03:
@j4c3 I actually put together most of an implementation to handle all of those attributes just this afternoon. I’m traveling right now so I can’t test it, but with a little luck I might be able to submit a PR within a few days.

wixoff, 2019-05-13:
Update - not ready yet. Apparently I know enough about zigbee-shepherd to be dangerous, but not enough to actually make a working converter.  That and I haven't had much time to follow up.

In the meantime I can confirm to @j4c3 that my database.db **does** appear to have a valid value for rmsVoltage (with nothing plugged in) at least, after the multiplier and divider is applied, so it appears the upgraded Peanut firmware is absolutely required.

roygbiv856, 2019-05-24:
@wixoff If I buy one of these, can I go through the standard pairing process to get it to work as just a plug without power monitoring? Or are there still extra steps involved with altering the database file?

j4c3, 2019-05-24:
Yes, it still requires extra steps.  In the SecuriFi section here: https://www.zigbee2mqtt.io/getting_started/pairing_devices.html

And, if you aren't going to have monitoring, you may need to make a change to ignore the empty attReport messages. 

A few extra steps, but if you just care about on and off, the price is pretty good.

wixoff, 2019-07-10:
OK, it took a while, but [PR #513 for the power, voltage, and current sensors](https://github.com/Koenkk/zigbee-shepherd-converters/pull/513) is now submitted and pending review.

Koenkk, 2019-07-11:
Closing this as @wixoff completed the support, will be supported in the dev branch within a few hours.

closed 2019-07-11 by Koenkk

dsaul560, 2019-08-15:
> I don't think this ever worked properly as a router and I believe that the firmware needs to be updated for measurements to be supported. Works fine as a switch though.

Is this device working as router. I bought a couple to increase range but devices far aways still getting disconnected.

wixoff, 2019-08-15:
They work for me as routers - in my [network map](https://github.com/rgruebel/ha_zigbee2mqtt_networkmap) I see several end devices connected to Peanut plugs rather than the coordinator.

j4c3, 2019-08-15:
Are your end devices battery powered Xiaomi/Aqara?  I am not 100% on this, but there may be an issue with those dropping off a router, since they don't report within expected time-frames.  I've seen talk of some routers that allow you to extend that time (specifically someone said this about the IKEA one), but I don't know the Peanut Plug does/can.

dsaul560, 2019-08-15:
When I pair sometimes I see in the map connected to the peanut, but after 2-3h it disconnect. Tried multiple times and it pairs but disconnect after few hours, bought 3 of them :/. My firmware is not updated as I don’t have a hub. All end device battery is 100%, they are relatively new. 

My end devices are all Xiaomi battery power, buttons, temp sensor, motion etc. the ones closer.

andrew-d, 2021-08-02:
It looks like the zigpy folks figured out how to perform an OTA update with a firmware file here:
https://github.com/zigpy/zigpy/issues/522

I am not using zigpy right now (and will probably return the plugs, TBQH), but I figured I'd drop this here in case anyone else comes across this issue and wants to try this.

airdrummingfool, 2021-08-02:
@andrew-d support for OTA updates for the Peanut Smart Plug was completed in https://github.com/Koenkk/zigbee-OTA/pull/14 and https://github.com/Koenkk/zigbee-herdsman-converters/pull/2027 and is fully functional.

andrew-d, 2021-08-03:
> @andrew-d support for OTA updates for the Peanut Smart Plug was completed in [Koenkk/zigbee-OTA#14](https://github.com/Koenkk/zigbee-OTA/pull/14) and [Koenkk/zigbee-herdsman-converters#2027](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2027) and is fully functional.

Oh, awesome - thank you for the information!