Koenkk/zigbee2mqtt #810: How to add an 'undefined' device to device.js

goldriver2, 2019-01-05:
I have a problem while trying to add a Stelpro Orleans integrated zigbee thermostat, here is part of the log while pairing device:

_zigbee2mqtt:info 2019-1-5 16:01:20 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:20 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:20 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:21 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:21 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:21 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:23 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:23 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:23 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:26 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:26 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:26 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:28 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:28 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:28 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:28 Connecting with device...
  zigbee2mqtt:info 2019-1-5 16:01:28 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-5 16:01:28 Message without device!
  zigbee2mqtt:info 2019-1-5 16:01:28 Device incoming...
  zigbee2mqtt:info 2019-1-5 16:01:28 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-1-5 16:01:28 New device with address 0xf8f005fffff89588 connected!
  zigbee2mqtt:info 2019-1-5 16:01:28 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xf8f005fffff89588"}'
  zigbee2mqtt:warn 2019-1-5 16:01:28 
  zigbee2mqtt:warn 2019-1-5 16:01:28 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-1-5 16:01:28 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-1-5 16:01:28 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Reading the how_to_support_new_devices.htm documents, it mentions that once the device is paired, if it is unsuported we should add the device in the device.js but since the _modelID_ is 'undefined' how should I create it ?

looking in the database.db, I found somewhere that the modelId was STE218, I tried with this in the device.js and rebooted but when I change the temperature on the thermostat nothing isn catch by the zigbeetomqtt.

Any toughts ?

khzd, 2019-01-06:
Hi,
the pairing did not provide the modelId at the header but in the ep record!
just to start , i would suggest to add **"modelId": "ST218",** in the DB after manufId. 
after that restart zigbee2mqtt and will see that ST218 is not supported
Now you can follow https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
 
PS
for clarity here is the dump of the record with the right position:
{
	"id": 2,
	"type": "Router",
	"ieeeAddr": "0xf8f005fffff89588",
	"nwkAddr": 8997,
	"manufId": 4485,
        **"modelId": "ST218",**
	"manufName": "Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
	"powerSource": "Mains (single phase)",
	"epList": [25],
	"status": "online",
	"joinTime": 1546722078,
	"endpoints": {
		"25": {
			"profId": 260,
			"epId": 25,
			"devId": 769,
			"inClusterList": [0, 3, 4, 513, 516],
			"outClusterList": [1026],
			"clusters": {
				"genBasic": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"zclVersion": 1,
						"appVersion": 104,
						"stackVersion": 3,
						"hwVersion": 1,
						"manufacturerName": "Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
						"modelId": "ST218\u0000\u0000\u0000(\u0001](\u0001S\b\u0000\u0001X%\u0004\u0000��\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
						"dateCode": "20170126 00111\u0000",
						"powerSource": 1
					}
				},
				"genIdentify": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"identifyTime": 168,
						"identifyCommissionState": 0
					}
				},
				"genGroups": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"nameSupport": 0
					}
				},
				"hvacThermostat": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"16385": -32768,
						"16412": 4,
						"localTemp": 2300,
						"outdoorTemp": -32768,
						"ocupancy": 1,
						"absMinHeatSetpointLimit": 500,
						"absMaxHeatSetpointLimit": 3000,
						"absMinCoolSetpointLimit": 700,
						"absMaxCoolSetpointLimit": 4000,
						"pIHeatingDemand": 0,
						"systemTypeConfig": 0,
						"localTemperatureCalibration": 0,
						"occupiedCoolingSetpoint": 3500,
						"occupiedHeatingSetpoint": 1900,
						"unoccupiedCoolingSetpoint": 4000,
						"unoccupiedHeatingSetpoint": 1700,
						"minHeatSetpointLimit": 500,
						"maxHeatSetpointLimit": 3000,
						"minCoolSetpointLimit": 700,
						"maxCoolSetpointLimit": 4000,
						"minSetpointDeadBand": 25,
						"remoteSensing": 2,
						"ctrlSeqeOfOper": 2,
						"systemMode": 4
					}
				},
				"hvacUserInterfaceCfg": {
					"dir": {
						"value": 1
					},
					"attrs": {
						"tempDisplayMode": 0,
						"keypadLockout": 0,
						"programmingVisibility": 0
					}
				},
				"msTemperatureMeasurement": {
					"dir": {
						"value": 2
					},
					"attrs": {}
				}
			}
		}
	},
	"_id": "ui9yNcvVLMI917vX"
}

goldriver2, 2019-01-06:
Hi Kaled, first thanks for your help with this, 

 

I modified the database.db to reflect your comments (see the attached files)

 

 

Now when I start zigbeetomqtt I have “zigbee2mqtt:info 2019-1-6 07:09:59 0xf8f005fffff89588 (0xf8f005fffff89588): SORH - Stelpro Thermostat integre zigbee Stelpro  (Router)” in the logfile instead of ‘undefined’

 

But even after adding the following to the device.js:

 

// Stelpro

    {

        zigbeeModel: ['ST218'], 

        model: 'SORH', 

        vendor: 'Stelpro', 

        description: 'Thermostat integre zigbee Stelpro ',  

        supports: 'temperature',  

        fromZigbee: [], 

        toZigbee: [],

    },

 

When playing with temperatures settings on the base heater thermostat, I see nothing in the log, it looks like zigbeetomqtt is not seeing any incoming messages from zigbee or it does not log it

 

Have I missed something along the line ?

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

goldriver2, 2019-01-06:
Ok I started back from the beginning

 

1.	I stopped zigbteetomqtt
2.	I erased the database.db
3.	I remove the information relating to the thermostat in the configuration.yal
4.	I reset the stelpro thermostat

 

1.	I start back the zigbeetomqtt
2.	I looked in database.db and configuration.yaml to make sure there was no signs of the thermostat

 

1.	I activated the pairing on the ST218
2.	I got the above in the LOG
3.	zigbee2mqtt:warn 2019-1-6 08:08:58 Message without device!
4.	  zigbee2mqtt:info 2019-1-6 08:09:05 Connecting with device...
5.	  zigbee2mqtt:info 2019-1-6 08:09:05 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
6.	  zigbee2mqtt:warn 2019-1-6 08:09:05 Message without device!
7.	  zigbee2mqtt:info 2019-1-6 08:09:07 Connecting with device...
8.	  zigbee2mqtt:info 2019-1-6 08:09:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
9.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Message without device!
10.	  zigbee2mqtt:info 2019-1-6 08:09:07 Connecting with device...
11.	  zigbee2mqtt:info 2019-1-6 08:09:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
12.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Message without device!
13.	  zigbee2mqtt:info 2019-1-6 08:09:07 Device incoming...
14.	  zigbee2mqtt:info 2019-1-6 08:09:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
15.	  zigbee2mqtt:info 2019-1-6 08:09:07 New device with address 0xf8f005fffff89588 connected!
16.	  zigbee2mqtt:info 2019-1-6 08:09:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xf8f005fffff89588"}'
17.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Device with modelID 'undefined' is not supported.
18.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
19.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Device with modelID 'undefined' is not supported.
20.	  zigbee2mqtt:warn 2019-1-6 08:09:07 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

 

1.	No I have 3 lines with the same id in the the database.db:
2.	{"id":2,"type":"Router","ieeeAddr":"0xf8f005fffff89588","nwkAddr":54989,"manufId":4485,"epList":[25],"status":"online","joinTime":1546780137,"endpoints":{"25":{"profId":260,"epId":25,"devId":769,"inClusterList":[0,3,4,513,516],"outClusterList":[1026],"clusters":{}}},"_id":"xsLNRH73hBUsQ2Zp"}
3.	{"id":2,"type":"Router","ieeeAddr":"0xf8f005fffff89588","nwkAddr":54989,"manufId":4485,"manufName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","epList":[25],"status":"online","joinTime":1546780137,"endpoints":{"25":{"profId":260,"epId":25,"devId":769,"inClusterList":[0,3,4,513,516],"outClusterList":[1026],"clusters":{}}},"_id":"xsLNRH73hBUsQ2Zp"}
4.	{"id":2,"type":"Router","ieeeAddr":"0xf8f005fffff89588","nwkAddr":54989,"manufId":4485,"manufName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","epList":[25],"status":"online","joinTime":1546780137,"endpoints":{"25":{"profId":260,"epId":25,"devId":769,"inClusterList":[0,3,4,513,516],"outClusterList":[1026],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":104,"stackVersion":3,"hwVersion":1,"manufacturerName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","modelId":"ST218\u0000\u0000\u0000(\u0001](\u0001S\b\u0000\u0001X%\u0004\u0000��\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","dateCode":"20170126 00111\u0000","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":170,"identifyCommissionState":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"hvacThermostat":{"dir":{"value":1},"attrs":{"16385":-32768,"16412":4,"localTemp":2150,"outdoorTemp":-32768,"ocupancy":1,"absMinHeatSetpointLimit":500,"absMaxHeatSetpointLimit":3000,"absMinCoolSetpointLimit":700,"absMaxCoolSetpointLimit":4000,"pIHeatingDemand":0,"systemTypeConfig":0,"localTemperatureCalibration":0,"occupiedCoolingSetpoint":3500,"occupiedHeatingSetpoint":1900,"unoccupiedCoolingSetpoint":4000,"unoccupiedHeatingSetpoint":1700,"minHeatSetpointLimit":500,"maxHeatSetpointLimit":3000,"minCoolSetpointLimit":700,"maxCoolSetpointLimit":4000,"minSetpointDeadBand":25,"remoteSensing":2,"ctrlSeqeOfOper":2,"systemMode":4}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0,"programmingVisibility":0}},"msTemperatureMeasurement":{"dir":{"value":2},"attrs":{}}}}},"_id":"xsLNRH73hBUsQ2Zp"}

 

My question here is on which of those lines do I put the "modelId": "ST218", you mention in your email ???

 

 

1.	I stopped again zigbeetomqtt
2.	I tried adding it to the three lines 
3.	Restarted zigbeemqtt 
4.	Here is what I have in the log
5.	zigbee2mqtt:info 2019-1-6 08:16:46 Currently 1 devices are joined:
6.	  zigbee2mqtt:info 2019-1-6 08:16:46 0xf8f005fffff89588 (0xf8f005fffff89588): unkown - undefined unknown (Router)

 

[... cut ...]
        fromZigbee: [],
        toZigbee: [],
    },
4.	I restarted the zigbetomqtt and now I have zigbee2mqtt:info 2019-1-6 08:26:51 0xf8f005fffff89588 (0xf8f005fffff89588): SORH - Stelpro Thermostat integre zigbee Stelpro  (Router) in the log

 

1.	Next step, I went to the thermostat and played with temperature setting and put it at 22 instead of 21
2.	NOTHING HAPPENS IN THE LOG FILE

 

My question now have I missed something somewhere ?

 

Thanks for any help

 

https://www.avast.com/antivirus

khzd, 2019-01-08:
hi
the last line is the more complete one.
I see another issue about Stelpro, please see the best modelId they put in place.

goldriver2, 2019-01-08:
What do you mean the last line is the more complete one “nothing in the log “ ?

I have also seen the other thread about Stelpro, the thing is they have 2 thermostat, one that is standalone on the wall and the other is integrated to the heatbase, I am almost certain that this is why there are 2 modelId

khzd, 2019-01-09:
hi
the last line **in the DB** is the more complete one, you may delete the first line about your device.

about 2 moedId, if you have only one, indeed you have only one record in the Db, so you may focus on only the las more complte line in the DB.
you have to add the modelId before manufacturerName, ie:
{"id":2,"type":"Router","ieeeAddr":"0xf8f005fffff89588","nwkAddr":54989,**"modelId":"ST218",** "manufId":4485,"manufName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","epList":[25],"status":"online","joinTime":1546780137,"endpoints":{"25":{"profId":260,"epId":25,"devId":769,"inClusterList":[0,3,4,513,516],"outClusterList":[1026],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":104,"stackVersion":3,"hwVersion":1,"manufacturerName":"Stelpro\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","modelId":"ST218\u0000\u0000\u0000(\u0001](\u0001S\b\u0000\u0001X%\u0004\u0000��\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","dateCode":"20170126 00111\u0000","powerSource":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":170,"identifyCommissionState":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"hvacThermostat":{"dir":{"value":1},"attrs":{"16385":-32768,"16412":4,"localTemp":2150,"outdoorTemp":-32768,"ocupancy":1,"absMinHeatSetpointLimit":500,"absMaxHeatSetpointLimit":3000,"absMinCoolSetpointLimit":700,"absMaxCoolSetpointLimit":4000,"pIHeatingDemand":0,"systemTypeConfig":0,"localTemperatureCalibration":0,"occupiedCoolingSetpoint":3500,"occupiedHeatingSetpoint":1900,"unoccupiedCoolingSetpoint":4000,"unoccupiedHeatingSetpoint":1700,"minHeatSetpointLimit":500,"maxHeatSetpointLimit":3000,"minCoolSetpointLimit":700,"maxCoolSetpointLimit":4000,"minSetpointDeadBand":25,"remoteSensing":2,"ctrlSeqeOfOper":2,"systemMode":4}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0,"programmingVisibility":0}},"msTemperatureMeasurement":{"dir":{"value":2},"attrs":{}}}}},"_id":"xsLNRH73hBUsQ2Zp"} 

cheers

henriz, 2019-01-12:
See my answer to this issue:

https://github.com/Koenkk/zigbee2mqtt/issues/844

goldriver2, 2019-01-12:
I looked at the link you provided and implemented it BUT this does not give me more informations neither fix my issue. I already knew what my device ref was (0xf8f005fffff89588)

Looking at the line that was created in the database.db when pairing occured, I saw the following: "modelId":"ST218\u0000\u0000\u0000(\u0001](\u0001S\b\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/\u0000\u0002\u0000"

I tought that this whole line might be the modelId so  I tried to enter this complete line of information in the device.js instead of just entering St218 but still no reaction in the log when I activate the thermostat temperature.

henriz, 2019-01-12:
Maybe you should look a bit better.

After adding your modelID ('ST218') you also have to remove all instances with your device address from database.db, configuration.yaml and state.json, and then repair.

As the device has now already been paired wrong, it's not re-pairing.

goldriver2, 2019-01-12:
Ok, I made sure the **modelId** was created correctly in the device.js with ST218, I stopped zigbee2mqtt, I remove ALL reference to the 0xf8f005fffff89588 device in the database.db and configuration.yaml (state.json was empty) and Restarted zigbee2mqtt, I repaired and still:

zigbee2mqtt:info 2019-1-12 07:45:40 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xf8f005fffff89588"}'
  zigbee2mqtt:warn 2019-1-12 07:45:40 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-1-12 07:45:40 device address '0xf8f005fffff89588'

here is my device.js setup:

 {
        zigbeeModel: ['ST218'], 
        model: 'ST218', 
        vendor: 'Stelpro', 
        description: 'Thermostat integre zigbee Stelpro ',  
        supports: 'temperature',  
        fromZigbee: [],
        toZigbee: [], 
    },

henriz, 2019-01-12:
Then the modelid is not 'ST218'.

If this:
ST218\u0000\u0000\u0000(\u0001](\u0001S\b\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/\u0000\u0002\u000

is what you earlier found in the database.db file, then you have to enter that entire string as modelID.

henriz, 2019-01-12:
There's something going on with that device ID. I don't have the time right now, but some added logging on pairing would help to solve this. If the issue hasn't been solved by then, I'll look into it this evening. It's 14:34 over here now.

goldriver2, 2019-01-12:
Ok I’ll continue my research and post my results before 18:00 your time

Bruno Lalongé

henriz, 2019-01-12:
To give you a pointer, those \u values stand for special characters. You have to find how to put those in a javascript string, and then put that in there.

(the double \\ escapes the single \)

sehraf, 2019-01-12:
Based on https://github.com/Koenkk/zigbee2mqtt/issues/421#issuecomment-425450869 you can just write `\u0000...` (see https://github.com/Koenkk/zigbee-shepherd-converters/pull/89/files#diff-6c9a6acf22f90d1c6e524d9f3c5c1745R82 ). So simple copy and paste *should* have worked ...

henriz, 2019-01-12:
There's also a / in there. Maybe that should have been escaped?

A better way would be to have zigbee2mqtt remove all non-standard ascii chars from the deviceid.

sehraf, 2019-01-12:
If the device id is just used for matching then this might be the easiest approach as long as there are no devices that differ only in weird uni-code characters. @Koenkk

goldriver2, 2019-01-12:
Ok first please bare with me as I am definitly not a programmer 

While doing my test, everytime I erased any lines referencing to the device 0xf8f005fffff89588 in the database.db and in the configuration.yaml (there as never been any data in the  state.json so no made were made to this file.

I also completely reset my Stelpro device to factory setting.

The first time I put my device in pairing mode, the zigbe2mqtt log never showed anything happening, I always need to stop zigbee on the Stelpro and then pair it again. Then only I can see the 
zigbee2mqtt:info 2019-1-12 08:21:09 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xf8f005fffff89588"}'
zigbee2mqtt:warn 2019-1-12 08:21:09 Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 2019-1-12 08:21:09 device address '0xf8f005fffff89588'

if i look at the database.db after this, there are 3 lines reffering to this device, the 2 first lines does not seem complete, only the third is complete.

Sometimes the modelId contains bizzare caracters: "ST218\u0000\u0004\u0000��\u0001\u0015�\u0001\u0013\u000f\u0004\u0001\u0002R0n\u0004\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000

sehraf, 2019-01-12:
I have a bad feeling about this. Why is the model id changing? It looks like the device it actually reading random memory. If this is true then the model id will always be random to a certain extend.
Counting bytes: it always sends 31 bytes as a model id. Looks like someone used 32 byte to store the model id but only set the first 5 or 6 :rofl: #highQualityProgramming

sehraf, 2019-01-12:
While there are [many](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L121) devices that look like `"<modelId>\0<random>"` (trailing `\0` from C strings) there are [some](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L2192) that seem to have a byte model id `"\u0000\u0002\u0000\u0004\u0012"`.
The question is: Are these the real ids that don't change? In the worst case there was exactly one user who has seen this id one time, added it and it was working for him, while it won't work a second time for anybody else.

goldriver2, 2019-01-12:
I would like to be of any help, unfortunatly I have no knowledge of programming but I can understand that it can be a headeach to program something for an unstable modelId !

henriz, 2019-01-12:
Maybe another workaround could be deviced.  You can't develop against unstable or changing ID's. And if there are devices that have a byte model ID, ditching all non-ascii chars won't work without breaking other stuff.

But.. how about making it possible to add exception devices based on the device address. Like a 'custom.db', 'custon.json' or 'custom.yaml' in which an end user can add the network ID and the desired ModelID?

goldriver2, 2019-01-12:
Just to confirm, I brought my RPI rigth nearby the Stelpro thermostat to make sure distance was not an issue and redid 5 pairing, here are the results:

"modelId":"ST218\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/"

"modelId":"ST218\u0000\u0000\u0000(\u0001]
(\u0001S\b\u0000\u0001X%\u0004\u0000��\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"

"modelId":"ST218\u0000@\u0000�\u0000@\u0000��\u0000\u0000�\u0004\u0000��\u0001\u0015\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"

"modelId":"ST218\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0001\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"

"modelId":"ST218\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/\u0000@\u0000�\u0000@\u0000"

sehraf, 2019-01-12:
Thanks @goldriver2 for doing some testing. I think the easiest way is to cut any model id by the first `null` and add some byte model id workaround for the few devices that need it. 
But lets first wait for @Koenkk 's opinion.

Koenkk, 2019-01-13:
I would propose the following:
- First a search with the normal modal ID will be done (so e.g. `ST218\u0000\u0000\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0000\u0007\u0001,c\u0001\u0001/\u0000@\u0000�\u0000@\u0000`
- If no device is found, search again while stripping everything after the first '\u'.

sehraf, 2019-01-13:
:+1: 
What will be the policy with (new) model ids containing `\uXXXX`?
Can the ids like `lumi.sensor_86sw1\u0000lu`be shortened to `lumi.sensor_86sw1`?

henriz, 2019-01-13:
How about the other way around. First search for the complete ID, and if that's not found, take the list of devices and see if the first part of the device string matches.

So, IE, if you have something like ST218\u0000... and so forth. You first use the current method, which won't find it. And then you turn it around an cydle the device array from devices.js and see if the model matches the first part of the searched string.

In devices you would then have the normal 'ST218'  ID, so it will then look if the first 5 characters of the modelI it got, matches that.

goldriver2, 2019-01-13:
I will be more than happy to test the dev branch, one thing before I proceed, I have find out how to clone from dev branch but I have no idea if I have to do a "uninstall" before ugrading and how to do it ? npm uninstall ???

Koenkk, 2019-01-14:
I've updated the dev branch to strip the null characters.

So `ST218` can be used as the zigbeeModel now.

@goldriver2 after switching to the dev branch, do a `rm -rf node_modules && npm install`

goldriver2, 2019-01-15:
Ok lets recap,

I have installed the new zigbeetomqtt 
    - removed the /opt/zigbee2mqtt directory
    - sudo git clone --branch dev https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
    - reconfigured the device.js adding sthe stelpro:
               // Stelpro model ST218
                            {
                            zigbeeModel: ['ST218'],
                            odel: 'ST218',
                            vendor: 'Stelpro',
                            description: 'Stelpro ST218',
                            supports: 'Thermostat temperature and outside temperature',
                            fromZigbee: [],
                            toZigbee: [],
                            },
              - restarted zigbeetomqtt

-I paired the device, in  the log files I still had "Unsuported device"
- stopped zigbetomqtt

I then remembered that @Koenkk told me I had to edit the database.db to move the content of the "modelId" before the "manufName", once done, restarted the zigbetomqtt

Now I have 
zigbee2mqtt:info 2019-1-15 05:49:37 0xf8f005fffff89588 (0xf8f005fffff89588): ST218 - Stelpro Stelpro ST218 (Router)

But when I change the temperature on the thermostat, nothing happens in the log

khzd, 2019-01-15:
Hi
Good progress for modelid challenge :) 
Now you can focus on the second one :
fromZigbee: [],
toZigbee: [],

You have to fill the [] with the good proc.  See thermostat implementations 
Bon courage

goldriver2, 2019-02-15:
I am Back after to where I left for a knee surgery, please bare with me as am am still not a programmer or It guy

As mentionned before, I have been able to pair the Stelpro ST218 with the plugin, now, according to the GITHUB instrcution, I am trying to get some feed back from the thermostat by playing with the down and up button but unfortunatly nothing happens in the log.

looking again at the documentation, it is mentionned "in case your device is not reporting anything, it could be that this device requires additional configuration. " I have looked at the exemples the page links to but I seriously have NO CLUE of where to start and what I would need to add in a "configuiration" section.

I found this exemple for an eCozy thermostat but how the hell can I find all parameters for the specific STelpro ????

help pleaee

// eCozy
    {
        zigbeeModel: ['Thermostat'],
        model: '1TST-EU',
        vendor: 'eCozy',
        description: 'Smart heating thermostat',
        supports: 'temperature, occupancy, un-/occupied heating, schedule',
        fromZigbee: [
            fz.ignore_basic_change, fz.generic_battery_voltage,
            fz.thermostat_att_report, fz.thermostat_dev_change,
        ],
        toZigbee: [
            tz.factory_reset, tz.thermostat_local_temperature, tz.thermostat_local_temperature_calibration,
            tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_setpoint_raise_lower,
            tz.thermostat_remote_sensing, tz.thermostat_control_sequence_of_operation, tz.thermostat_system_mode,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_weekly_schedule_rsp,
            tz.thermostat_relay_status_log, tz.thermostat_relay_status_log_rsp,
        ],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 3);
            const actions = [
                // from https://github.com/ckpt-martin/Hubitat/blob/master/eCozy/eCozy-ZigBee-Thermostat-Driver.groovy
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genTime', coordinator, cb),
                (cb) => device.bind('genPollCtrl', coordinator, cb),
                (cb) => device.bind('hvacThermostat', coordinator, cb),
                (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
                (cb) => device.report('hvacThermostat', 'localTemp', 5, 30, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },

goldriver2, 2019-02-16:
made some progress, I have been able to define a device.js file that now provides me at least the local temperature.

I am now stuck where if I change the temperature on the thermostat, nothing is returned to the zigbee2mqtt

{
        zigbeeModel: ['ST218'],
        model: 'ST218',
        vendor: 'Stelpro',
        description: 'Thermostat ',
        supports: 'temperature and humidity',
        fromZigbee: [
            fz.ignore_basic_change, fz.generic_battery_voltage,
            fz.thermostat_att_report, fz.thermostat_dev_change,
        ],
        toZigbee: [
            tz.factory_reset, tz.thermostat_local_temperature, tz.thermostat_local_temperature_calibration,
            tz.thermostat_occupancy, tz.thermostat_occupied_heating_setpoint,
            tz.thermostat_unoccupied_heating_setpoint, tz.thermostat_setpoint_raise_lower,
            tz.thermostat_remote_sensing, tz.thermostat_control_sequence_of_operation, tz.thermostat_system_mode,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule, tz.thermostat_weekly_schedule_rsp,
            tz.thermostat_relay_status_log, tz.thermostat_relay_status_log_rsp,
        ],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 25);
            const actions = [
                (cb) => device.bind('genBasic', coordinator, cb),
                (cb) => device.bind('genIdentify', coordinator, cb),
                (cb) => device.bind('genGroups', coordinator, cb),
                (cb) => device.bind('hvacThermostat', coordinator, cb),
                (cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('hvacThermostat', 'localTemp', 300, 3600, 0, cb),
                ];

            execute(device, actions, callback); 
         },

goldriver2, 2019-02-22:
Hi

I have no clue as to how to do a PR to add something to GITHUB but for anyone interested, here is the changes I added to my device.js to get it to work:

const devices = [
// Stelpro Orleans baseboard thermostat
{
zigbeeModel: ['ST218'],
model: 'ST218',
vendor: 'Stelpro',
description: 'Baseboard integrated zigbee Thermostat ',
supports: 'temperature ',
fromZigbee: [
fz.thermostat_att_report, fz.thermostat_dev_change,
],
toZigbee: [
tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint,
tz.thermostat_pi_heating_demand,
],
configure: (ieeeAddr, shepherd, coordinator, callback) => {
const device = shepherd.find(ieeeAddr, 25);
const actions = [
(cb) => device.bind('genBasic', coordinator, cb),
(cb) => device.bind('genIdentify', coordinator, cb),
(cb) => device.bind('genGroups', coordinator, cb),
(cb) => device.bind('hvacThermostat', coordinator, cb),
(cb) => device.bind('hvacUserInterfaceCfg', coordinator, cb),
(cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
(cb) => device.report('hvacThermostat', 'localTemp', 300, 3600, 0, cb),
];
execute(device, actions, callback);
},
},

as for the tozigbee.js and fromzigbee.js, I used the already existing ones and everything seems to work

closed 2019-02-22 by goldriver2

BuhtigithuB, 2019-12-29:
Nice job!