Koenkk/zigbee2mqtt #814: Setting parameters of zigbee devices

HerrieMan72, 2019-01-06:
Hi,
I'm in the process of transferring my zigbee devices from Homey to zigbee2mqtt (combined with node-red and domoticz).
I've connected a few devices to zigbee2mqtt (ikea tradfri lamps, xiaomi aqare switch/human-body-sensor/thb-sensor), that was pretty easy.
But I am wondering how I go about setting the parameters of the devices. E.g. in Homey I could set parameters for the xiaomi sensors for motion-timeout (in the human-body-sensor), or for battery-low level warning. I've been looking at the tozigbee converters in the shepherd directory, but for e.g. my human-body-sensor (lumi.sensor_motion.aq2) this lists nothing.
Is setting of device parameters supported in zigbee2mqtt ? if yes, how should I proceed on this ?
regards,
Herman

ryanbeaton, 2019-01-06:
For timeout
https://koenkk.github.io/zigbee2mqtt/configuration/device_specific_configuration.html

There are a few more options that have been added in the dev branch too but I can’t remember them off the top of my head

HerrieMan72, 2019-01-07:
Hi Ryan,
thanks for your reply, that timeout is indeed the most important parameter for now, I'll update the configfile for that. Do you know whether or not this parameter is actually set on the motion-device, or is this a zigbee2mqtt internal timer or so ?
More generic speaking, how are these parameters handled in a zigbee network and by zigbee2mqtt ? can I somehow query my devices to check what parameters can be configured ? or is there some global database with parameters per device ?
And then how are these parameters set onto the devices, is there some generic mechanism in place to facilitate this ?

khzd, 2019-01-07:
Hello

yes, there is a file : configuration.yaml where you will store this kind of data.

there is a more technical database with all pairing informations, all are text file, you can edit them (carrefully)

each time you paire a device records are added to this 2 files, you can later complete with few requirements.

cheers

closed 2019-03-15 by stale[bot]