Koenkk/zigbee2mqtt #816: Blacklist/block devices from joining?

simonvanderveldt, 2019-01-06:
It seems like there are a couple devices in my vicinity that are joined to zigbee2mqtt as soon as I enable joining. Is there a way to block/blacklist specific devices from being allowed to join?

I couldn't find any info on this in the issues or documentation.

Koenkk, 2019-01-08:
Implemented in dev branch, can you verify this feature?

simonvanderveldt, 2019-01-08:
Cool, thanks! I can check it this week, at the latest this weekend. Is that OK?

Koenkk, 2019-01-09:
No problem :)

tb-killa, 2019-03-10:
@Koenkk your implementation use the defined function for remove devices.
As I see this on different devices it seems not work as it should do. 
I think we should rebuild this and use a defined way to prevent them joining again. 

How about add to our devices configuration part something like banned with the addr per device. 
On controller side we should do something like this :
https://github.com/zigbeer/zigbee-shepherd/wiki#API_acceptDevIncoming

This would be a solid way.

Koenkk, 2019-03-26:
Implemented!

Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/configuration/configuration.md (`ban`)

Devices banned through `zigbee2mqtt/bridge/config/ban` will also be added to this list.

simonvanderveldt, 2019-03-26:
Sorry, had some personal things get in the way. Should I still validate/test this?

Koenkk, 2019-03-27:
@simonvanderveldt yes, please :)

closed 2019-04-07 by Koenkk