Koenkk/zigbee2mqtt #817: Aqara switch long press

berrywhite96, 2019-01-06:
Hello, I am using the double click switch from Aqara, everything works as expected. But I have been in a discussion with a user who uses the same switch with I think the Dresden elektronik project and he says that the switch supports long clicks. Zigbee2mqtt only sends single or double click events. It is possible to implement the long click event in a way that it sends several MQTT messages in a short time?
I researched a bit on github and found a repo using zigbee shephard, on the click event it starts a timer and then recognize a long click. Maybe this is an idea.

Thank you.

subzero79, 2019-01-07:
If you are talking about the wall switch with double buttons, which you can press right, left and both, only newer versions of that switch support double and long in each of the buttons, unfortunately there is no difference in the model number. This was reported by a user here before last year. I confirmed this, the seller that i know has this new switch for sale is this [one](https://www.aliexpress.com/item/original-Xiaomi-Aqara-Mijia-Smart-home-Light-Control-Single-Fire-wire-ZigBee-Wireless-Key-Wall-Switch/32878073541.html?spm=a2g0s.9042311.0.0.36aa4c4dwTjU6M)

berrywhite96, 2019-01-07:
Ur right, I meant the wall switch. maybe Aqara just made an update for the triggers. I dont have an official hub, but I cant imagine Aqara have different function for the same model.

Koenkk, 2019-01-07:
@berrywhite96 which model do you have?

berrywhite96, 2019-01-07:
@Koenkk WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice)

berrywhite96, 2019-01-07:
@subzero79 I dont know but that would explain why only one model id exist with different features.
Never mind I never buy a aqara hub so the issue can be closed if no others are interested in this discussion.

AnhDuc85, 2019-01-07:
my WXKG02LM - CMIIT ID 2016DP1247 supports long press, got it few weeks ago and never used it with xiaomi/aqara gateway

AnhDuc85, 2019-01-07:
actually it's on the back of the switch
[image]

berrywhite96, 2019-01-07:
@AnhDuc85 oh ok but I dont want to pull it down from my wall.

AnhDuc85, 2019-01-07:
that's on the package
[image]

berrywhite96, 2019-01-07:
> that's on the package
> [image]

My packaging looks different:
[image]

subzero79, 2019-01-07:
I still have the hub. I can give a try later to see if fw update is available, but I doubt it. Maybe you can ask at home asssitant forum for people still using the hub.

subzero79, 2019-01-07:
Someone posted this on facebook the other day

https://gist.github.com/johan/26913f6aa2f0ed71db4256e09a561709

This emulates single, double and triple click using the shelly1 stock firmware when using the detached button mode. I gave this a try now using node-red and the aqara double switch (the one with no double or long press). It works good so far with single and double click

[image]

this is the flow, needs to be polished for sure

```
[{"id":"dc0cb373.5fb0c","type":"server-state-changed","z":"9ac58995.e96338","name":"","server":"d5ce5cec.92bc8","entityidfilter":"sensor.switch_masterbedroom_0bcb_click","entityidfiltertype":"substring","outputinitially":false,"state_type":"str","haltifstate":"","halt_if_type":"","halt_if_compare":"is","outputs":1,"x":350,"y":300,"wires":[["95313f23.2a4c9"]]},{"id":"95313f23.2a4c9","type":"switch","z":"9ac58995.e96338","name":"","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"left","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":670,"y":300,"wires":[["e44b7451.9a6108"]]},{"id":"e44b7451.9a6108","type":"api-call-service","z":"9ac58995.e96338","name":"","server":"d5ce5cec.92bc8","service_domain":"counter","service":"increment","data":"{\"entity_id\":\"counter.masterbedroom_wall_left_click\"}","render_data":false,"mergecontext":"","output_location":"payload","output_location_type":"msg","x":1040,"y":300,"wires":[[]]},{"id":"2185d5ff.e9096a","type":"trigger-state","z":"9ac58995.e96338","name":"single","server":"d5ce5cec.92bc8","entityid":"counter.masterbedroom_wall_left_click","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"kwlq1up2jr","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"1"},{"id":"851u4kh4jd","targetType":"this_entity","targetValue":"","propertyType":"previous_state","propertyValue":"old_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"0"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"str","x":390,"y":380,"wires":[["134316a4.dc0ec9"],[]]},{"id":"85008f2e.e0767","type":"api-call-service","z":"9ac58995.e96338","name":"Light1","server":"d5ce5cec.92bc8","service_domain":"light","service":"toggle","data":"{\"entity_id\":\"light.ricardo_bed_light\"}","render_data":false,"mergecontext":"","output_location":"","output_location_type":"none","x":1090,"y":380,"wires":[["4a761c47.367344"]]},{"id":"134316a4.dc0ec9","type":"delay","z":"9ac58995.e96338","name":"","pauseType":"delay","timeout":"500","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":590,"y":380,"wires":[["3f1a9708.75b7b8"]]},{"id":"3f1a9708.75b7b8","type":"api-current-state","z":"9ac58995.e96338","name":"leftClick","server":"d5ce5cec.92bc8","halt_if":"1","halt_if_type":"str","halt_if_compare":"is_not","override_topic":true,"override_payload":true,"override_data":true,"entity_id":"counter.masterbedroom_wall_left_click","state_type":"str","outputs":2,"x":780,"y":380,"wires":[["b90d1bac.7df168","28b27878.e68a78"],[]]},{"id":"b90d1bac.7df168","type":"switch","z":"9ac58995.e96338","name":"","property":"data.timeSinceChangedMs","propertyType":"msg","rules":[{"t":"gte","v":"501","vt":"num"}],"checkall":"true","repair":false,"outputs":1,"x":930,"y":380,"wires":[["85008f2e.e0767"]]},{"id":"28b27878.e68a78","type":"debug","z":"9ac58995.e96338","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":930,"y":440,"wires":[]},{"id":"8aab7958.12e988","type":"trigger-state","z":"9ac58995.e96338","name":"double","server":"d5ce5cec.92bc8","entityid":"counter.masterbedroom_wall_left_click","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"4kzp8jonvv6","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"2"},{"id":"200kganwz74","targetType":"this_entity","targetValue":"","propertyType":"previous_state","propertyValue":"old_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"1"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"str","x":390,"y":480,"wires":[["e6f0ff67.8bc22"],[]]},{"id":"e6f0ff67.8bc22","type":"delay","z":"9ac58995.e96338","name":"","pauseType":"delay","timeout":"500","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":590,"y":480,"wires":[["9c4304b0.79ee38"]]},{"id":"9c4304b0.79ee38","type":"api-current-state","z":"9ac58995.e96338","name":"leftClick","server":"d5ce5cec.92bc8","halt_if":"2","halt_if_type":"str","halt_if_compare":"is_not","override_topic":true,"override_payload":true,"override_data":true,"entity_id":"counter.masterbedroom_wall_left_click","state_type":"str","outputs":2,"x":780,"y":480,"wires":[["5d31232e.6b7a6c"],[]]},{"id":"5d31232e.6b7a6c","type":"switch","z":"9ac58995.e96338","name":"","property":"data.timeSinceChangedMs","propertyType":"msg","rules":[{"t":"gte","v":"501","vt":"num"}],"checkall":"true","repair":false,"outputs":1,"x":930,"y":480,"wires":[["207ff6c7.c79eda"]]},{"id":"207ff6c7.c79eda","type":"api-call-service","z":"9ac58995. [... cut ...] ne","x":1090,"y":480,"wires":[["4a761c47.367344"]]},{"id":"4a761c47.367344","type":"api-call-service","z":"9ac58995.e96338","name":"resetCounter","server":"d5ce5cec.92bc8","service_domain":"counter","service":"reset","data":"{\"entity_id\":\"counter.masterbedroom_wall_left_click\"}","render_data":false,"mergecontext":"","output_location":"payload","output_location_type":"msg","x":1270,"y":440,"wires":[[]]},{"id":"d5ce5cec.92bc8","type":"server","z":"","name":"Home Assistant","legacy":false}]
```

berrywhite96, 2019-01-08:
> Someone posted this on facebook the other day
> 
> https://gist.github.com/johan/26913f6aa2f0ed71db4256e09a561709
> 
> This emulates single, double and triple click using the shelly1 stock firmware when using the detached button mode. I gave this a try now using node-red and the aqara double switch (the one with no double or long press). It works good so far with single and double click. Long press won't work.
> 
> [image]
> 
> this is the flow, needs to be polished for sure
> 
> ```
> [{"id":"dc0cb373.5fb0c","type":"server-state-changed","z":"9ac58995.e96338","name":"","server":"d5ce5cec.92bc8","entityidfilter":"sensor.switch_masterbedroom_0bcb_click","entityidfiltertype":"substring","outputinitially":false,"state_type":"str","haltifstate":"","halt_if_type":"","halt_if_compare":"is","outputs":1,"x":350,"y":300,"wires":[["95313f23.2a4c9"]]},{"id":"95313f23.2a4c9","type":"switch","z":"9ac58995.e96338","name":"","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"left","vt":"str"}],"checkall":"true","repair":false,"outputs":1,"x":670,"y":300,"wires":[["e44b7451.9a6108"]]},{"id":"e44b7451.9a6108","type":"api-call-service","z":"9ac58995.e96338","name":"","server":"d5ce5cec.92bc8","service_domain":"counter","service":"increment","data":"{\"entity_id\":\"counter.masterbedroom_wall_left_click\"}","render_data":false,"mergecontext":"","output_location":"payload","output_location_type":"msg","x":1040,"y":300,"wires":[[]]},{"id":"2185d5ff.e9096a","type":"trigger-state","z":"9ac58995.e96338","name":"single","server":"d5ce5cec.92bc8","entityid":"counter.masterbedroom_wall_left_click","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"kwlq1up2jr","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"1"},{"id":"851u4kh4jd","targetType":"this_entity","targetValue":"","propertyType":"previous_state","propertyValue":"old_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"0"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"str","x":390,"y":380,"wires":[["134316a4.dc0ec9"],[]]},{"id":"85008f2e.e0767","type":"api-call-service","z":"9ac58995.e96338","name":"Light1","server":"d5ce5cec.92bc8","service_domain":"light","service":"toggle","data":"{\"entity_id\":\"light.ricardo_bed_light\"}","render_data":false,"mergecontext":"","output_location":"","output_location_type":"none","x":1090,"y":380,"wires":[["4a761c47.367344"]]},{"id":"134316a4.dc0ec9","type":"delay","z":"9ac58995.e96338","name":"","pauseType":"delay","timeout":"500","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":590,"y":380,"wires":[["3f1a9708.75b7b8"]]},{"id":"3f1a9708.75b7b8","type":"api-current-state","z":"9ac58995.e96338","name":"leftClick","server":"d5ce5cec.92bc8","halt_if":"1","halt_if_type":"str","halt_if_compare":"is_not","override_topic":true,"override_payload":true,"override_data":true,"entity_id":"counter.masterbedroom_wall_left_click","state_type":"str","outputs":2,"x":780,"y":380,"wires":[["b90d1bac.7df168","28b27878.e68a78"],[]]},{"id":"b90d1bac.7df168","type":"switch","z":"9ac58995.e96338","name":"","property":"data.timeSinceChangedMs","propertyType":"msg","rules":[{"t":"gte","v":"501","vt":"num"}],"checkall":"true","repair":false,"outputs":1,"x":930,"y":380,"wires":[["85008f2e.e0767"]]},{"id":"28b27878.e68a78","type":"debug","z":"9ac58995.e96338","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":930,"y":440,"wires":[]},{"id":"8aab7958.12e988","type":"trigger-state","z":"9ac58995.e96338","name":"double","server":"d5ce5cec.92bc8","entityid":"counter.masterbedroom_wall_left_click","entityidfiltertype":"exact","debugenabled":false,"constraints":[{"id":"4kzp8jonvv6","targetType":"this_entity","targetValue":"","propertyType":"current_state","propertyValue":"new_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"2"},{"id":"200kganwz74","targetType":"this_entity","targetValue":"","propertyType":"previous_state","propertyValue":"old_state.state","comparatorType":"is","comparatorValueDatatype":"str","comparatorValue":"1"}],"constraintsmustmatch":"all","outputs":2,"customoutputs":[],"outputinitially":false,"state_type":"str","x":390,"y":480,"wires":[["e6f0ff67.8bc22"],[]]},{"id":"e6f0ff67.8bc22","type":"delay","z":"9ac58995.e96338","name":"","pauseType":"delay","timeout":"500","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":590,"y":480,"wires":[["9c4304b0.79ee38"]]},{"id":"9c4304b0.79ee38","type":"api-current-state","z":"9ac58995.e96338","name":"leftClick","server":"d5ce5cec.92bc8","halt_if":"2","halt_if_type":"str","halt_if_compare":"is_not","override_topic":true,"override_payload":true,"override_data":true,"entity_id":"counter.masterbedroom_wall_left_click","state_type":"str","outputs":2,"x":780,"y":480,"wires":[["5d31232e.6b7a6c"],[]]},{"id":"5d31232e.6b7a6c","type":"switch","z":"9ac58995.e96338","name":"","property":"data.timeSinceChangedMs","propertyType":"msg","rules":[{"t":"gte","v":"501","vt":"num"}],"checkall":"true","repair":false,"outputs":1,"x":930,"y":480,"wires":[["207ff6c7.c79eda"]]},{"id":"207ff6c7. [... cut ...] ervice","z":"9ac58995.e96338","name":"resetCounter","server":"d5ce5cec.92bc8","service_domain":"counter","service":"reset","data":"{\"entity_id\":\"counter.masterbedroom_wall_left_click\"}","render_data":false,"mergecontext":"","output_location":"payload","output_location_type":"msg","x":1270,"y":440,"wires":[[]]},{"id":"d5ce5cec.92bc8","type":"server","z":"","name":"Home Assistant","legacy":false}]
> ```

Is this home assistant? I dont know what do you mean. Where is the firmware you mentioned?

aXionDevelopment, 2019-01-17:
> my WXKG02LM - CMIIT ID 2016DP1247 supports long press, got it few weeks ago and never used it with xiaomi/aqara gateway

Fyi, I have two of these switches, both with de CMIIT ID: 2016DP1247. Only one of the switches supports double and long clicks.

steino, 2019-01-19:
According to https://www.home-assistant.io/components/binary_sensor.xiaomi_aqara/ the 2nd gen model of the WXKG03LM supports single, double and hold and reports zigbee model id as "remote.b186acn01",  gen 1 reports "86sw1".  I do not know how to identify the generation, but my gen1 has a CMIIT ID: 2016DP1334.

aXionDevelopment, 2019-01-19:
Yep sort of,

--
Xiaomi Wall switch WITH double/long click
WXKG02LM 
CMIIT ID: 2016DP1247
zigbee2mqtt:debug 2019-1-19 14:42:49 Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":1}}' of device '**lumi.remote.b286acn01**' (0x00158d000xxxxxx)

--
Xiaomi Wall switch without double  click
WXKG02LM 
CMIIT ID: 2016DP1247
zigbee2mqtt:debug 2019-1-19 14:44:15 Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device '**lumi.sensor_86sw2Un**' (0x00158d000xxxxxx)

closed 2019-03-27 by stale[bot]