Koenkk/zigbee2mqtt #818: CC2531 Coordinator changed behaviour - Not able ro run the bridge

WagnerBeccard, 2019-01-06:
Tryed the last two hours to fix with restarts, pressing the button, changing the configuration, but I think there coordinator now refuses to do his work.
FW revision 20181119 
When the stick gets power both LED's are shining and after approxemetly 30 seccons the green light stops and only the red one is shining.

```
Jan 06 20:25:01 hassbian npm[529]:   zigbee2mqtt:info 2019-1-6 20:25:01 Starting zigbee2mqtt version 1.0.1 (commit #84593a5)
Jan 06 20:25:01 hassbian npm[529]:   zigbee2mqtt:info 2019-1-6 20:25:01 Starting zigbee-shepherd
Jan 06 20:25:04 hassbian npm[529]:   zigbee2mqtt:info 2019-1-6 20:25:04 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
Jan 06 20:26:47 hassbian npm[529]:   zigbee2mqtt:info 2019-1-6 20:26:47 Starting zigbee-shepherd
Jan 06 20:26:50 hassbian npm[529]:   zigbee2mqtt:error 2019-1-6 20:26:50 Error while starting zigbee-shepherd!
Jan 06 20:26:50 hassbian npm[529]:   zigbee2mqtt:error 2019-1-6 20:26:50 Press the reset button on the stick (the one closest to the USB) and start again
Jan 06 20:26:50 hassbian npm[529]:   zigbee2mqtt:error 2019-1-6 20:26:50 Failed to start
Jan 06 20:26:50 hassbian npm[529]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}```

Koenkk, 2019-01-08:
I've never seen this before on a CC2531, perhaps the firmware is corrupted, can you try reflashing?

WagnerBeccard, 2019-01-08:
Flashed one of the other sticks. Something you like to know bout the corrupted one before i reflash this one?

closed 2019-01-08 by WagnerBeccard

Koenkk, 2019-01-09:
As I've never seen this before, let's see this as an incident. We should monitor further if this occurs multiple times.