Koenkk/zigbee2mqtt #819: Cannot read property 'device' of undefined

halkeye, 2019-01-07:
Just flashed the new adapter, hooked it up, and tried to pair my blinds. I have no idea what zigbee adapter the blinds has, but it actually errors out:

Added a console.log right before the error line 

```
Mon, 07 Jan 2019 02:30:05 GMT zigbee-shepherd Device: 0x00124b000bebb951 join the network.
message { type: 'devIncoming',
  endpoints: [],
  data: '0x00124b000bebb951' }
  zigbee2mqtt:error 2019-1-6 18:30:05
        {"message":"Cannot read property 'device' of undefined","stack":"TypeError: Cannot read property 'device' of undefined\n    at Controller.onZigbeeMessage (/home/pi/zigbee2mqtt/lib/controller.js:95:55)\n    at Zigbee.onMessage (/home/pi/zigbee2mqtt/lib/zigbee.js:173:18)\n    at emitOne (events.js:116:13)\n    at ZShepherd.emit (events.js:211:7)\n    at ZShepherd.<anonymous> (/home/pi/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:100:14)\n    at emitOne (events.js:116:13)\n    at ZShepherd.emit (events.js:211:7)\n    at /home/pi/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/event_handlers.js:227:18\n    at _fulfilled (/home/pi/zigbee2mqtt/node_modules/q/q.js:854:54)\n    at self.promiseDispatch.done (/home/pi/zigbee2mqtt/node_modules/q/q.js:883:30)"}
```

The entire log (`DEBUG=* node index.js 2>&1 | tee output.log`) is at https://gist.github.com/halkeye/6468a5cab7c5feb6dbec4b9557d33268

halkeye, 2019-01-07:
Changed it to:
```
if (message.endpoints && message.endpoints.length && message.endpoints[0].device) {
             device = message.endpoints[0].device;
         }
```

Gives me:
```
  zigbee2mqtt:debug 2019-1-6 18:37:15 Received zigbee message of type 'devIncoming' with data '"0x00124b000bebb951"'
  zigbee2mqtt:info 2019-1-6 18:37:15 Device incoming...
  zigbee2mqtt:info 2019-1-6 18:37:15 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2019-1-6 18:37:15 Message without device!
message { type: 'devStatus', endpoints: [], data: 'online' }
  zigbee2mqtt:debug 2019-1-6 18:37:15 Received zigbee message of type 'devStatus' with data '"online"'
  zigbee2mqtt:warn 2019-1-6 18:37:15 Message without device!
```

halkeye, 2019-01-07:
so is the zigbee implementation on this device faulty?

khzd, 2019-01-07:
Hello
Not faulty, may bébés the pairing is not complete or the device is not known !
Please share your configuration and dB for help

halkeye, 2019-01-07:
I'm not sure whats useful in the configuration:

DB (pretty printed through jq): https://gist.github.com/b7c27296c0971775b7da04896447144e

config:
```
# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://<server>'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
  disable_led: true

advanced:
   log_level: debug
```

khzd, 2019-01-07:
hi
your first device is not properly paired : modelId is not there, eplist is empty, :(
please try to paire is as close as possible to your gateway :  you will not progress if this record doesn't provide at least ModelId :(
Good luck and be patient for pairing, some devices required to me more 10 times...
 
{
  "id": 2,
  "type": "Router",
  "ieeeAddr": "0x00124b000bebb951",
  "nwkAddr": 16852,
  "manufId": 0,
  "epList": [],
  "status": "online",
  "joinTime": 1546828205,
  "endpoints": {},
  "_id": "iXG3ONCL0ytvx5cH"
}

halkeye, 2019-01-07:
How do I know what the gateway is? Its mesh like zwave right? so its
probably the first device I was playing with?

AFAIK, the only active zigbee devices I have are these blinds.

sehraf, 2019-01-07:
One of May Xiaomi devices took almost one minute untill it was completly paired...

khzd, 2019-01-08:
Hi
The gateway is your “adapter“

Koenkk, 2019-01-09:
There are 2 issues here:
1. Zigbee2mqtt crashes when the device has no endpoints.
2. There are no endpoints for this device (which I've never seen before).

Thanks to [your comment](https://github.com/Koenkk/zigbee2mqtt/issues/819#issuecomment-4518046850) I've fixed 1. 

2 has yet to be fixed, what device is this?

halkeye, 2019-01-10:
[Finesse.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2743072/Finesse.pdf)

At some point i should call them up and see if I can get more details, but thats the best I've found online.

sebastianheierhoff, 2019-01-14:
> [Finesse.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2743072/Finesse.pdf)
> 

Are you sure this is the correct product? The document says "Built-in 433.92Mhz RF receiver with electronic processor" and doesn't mention Zigbee at all.

halkeye, 2019-01-17:
I've contacted the manufacturer to see what they have, i'm tempted to close this bug for now.

closed 2019-03-25 by stale[bot]