Koenkk/zigbee2mqtt #822: Call for elp on Hue scenes and Home Assistant

alfatag, 2019-01-07:
Hi Friends!

This is not a ticket to open an issue, but a call for help... I've done a lot of research on HA forums, Zigbee2mqtt docs and google research on any exmaples on how to set a simple scene using Phillips Hue lights on zigbee2mqtt with Home Assistant, but without success.

Im trying the following scene config on HA, but it doesnt work:

scene:
name: Dim Lights
    entities:
      light.0x00178801102f0d68_light:
        state:true
        xy_color:["x":0.496,"y":0.415]
        brightness:11

Anyone can help me with a working scene example?

Thanks in advance!

Renato Troitino

AnhDuc85, 2019-01-07:
that's an example by using the HA services: https://pastebin.com/1weSFgCk

alfatag, 2019-01-09:
Hi All!

Anyone has been able to make zigbee2mqtt and Hue bulbs work with Home Assistant scenes?

If so, can you please share your config?

Thanks in advance!

Renato Troitino

alfatag, 2019-01-10:
OK Guys, case closed... syntax error. I will post below the correct config entry for scene with multiple bulbs, just for reference:

`
scene:
  - name: Relax Lights
    entities:
      light.hues01_light:
        state: on
        brightness: 10
        transition: 3
        xy_color: [0.46, 0.41]
      light.hues02_light:
        state: on
        brightness: 10
        transition: 3
        xy_color: [0.46, 0.41]
      light.hues03_light:
        state: on
        brightness: 10
        transition: 3
        xy_color: [0.46, 0.41]
      light.hues04_light:
        state: on
        brightness: 10
        transition: 3
        xy_color: [0.46, 0.41]        
      light.huesofa_light:
        state: on
        brightness: 10
        transition: 3
        xy_color: [0.46, 0.41]
`

closed 2019-01-10 by alfatag