Koenkk/zigbee2mqtt #824: Latest CC2531 firmware for flash via Arduino/ESP8266(alternative flashing method)

simpat1zq, 2019-01-08:
So I was able to flash using the alternative method a couple months ago, and everything has been OK. However, I'm seeing now that this method used a special build of the firmware from a year ago. There have been many updates to it since them. Is there an updated firmware to flash via the alternative method? Can I just use the regular build, or does the arduino method still need a special build?

Chrischi-, 2019-01-08:
you can take the normal .hex version of the firmware from [here](https://github.com/Koenkk/Z-Stack-firmware) and remove the second last line

**-**`:0400000500002D7357`

roger-, 2019-01-08:
Is the only significant change in that version the ability to support more devices?

Chrischi-, 2019-01-08:
for the C2531 there are currently 2 firmware versions available from @Koenkk  

- normal, ~16 devices and ~16 routers
- max devices, 44 devices and 1 router

the dev branch is currently working on an improved firmware. 
therefore i would wait with an update.

simpat1zq, 2019-01-09:
Cool. Thanks guys. I don't have many devices now, so I guess I'll just wait for the latest firmware update before I do anything.

adn77, 2019-01-15:
I read somewhere that the .bin versions of the firmware can be flashed directly via the TI tool (without the debugger).
If that was true, how do we generate a .bin file from the .hex?

Chrischi-, 2019-01-15:
It needs to be compiled. You can't convert the bin file.

closed 2019-03-23 by stale[bot]