Koenkk/zigbee2mqtt #833: Unsupported device: innr rb162

matija88, 2019-01-09:
Hello there, i have two innnr RB162 LED bulbs and would like to add them to my home automation. Has anyone added this device to the "list" of suported devices or can someone help me trough the procedeure I'm a bit stuck at point 2. of this guide https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html i cannot locate this file node_modules/zigbee-shepherd-converters/devices.js I know this are basics but this is just how it is :).
I would appreciate your help very much.

khzd, 2019-01-09:
hi what is your platform ? docker ? Pi or PC?

matija88, 2019-01-09:
Docker.

khzd, 2019-01-09:
on PI?

khzd, 2019-01-09:
I do not have na installation on DOcker but on non Docker install you find it under 
/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters
hop it help

matija88, 2019-01-10:
Yes of course.
{"id":3,"type":"Router","ieeeAddr":"0x00158d0000cfd1ad","nwkAddr":56409,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RB 162\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2],"status":"online","joinTime":1547070530,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"3V8aCLnE8EE96sNv"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b87c10","nwkAddr":31432,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,1026,1027,1029,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"PrvvGIB8hWMeeLKI"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ed1160","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1547069505,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"m5DN1akLInbBD9Fw"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018ed1160","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1547070583,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"m5DN1akLInbBD9Fw"}
{"id":3,"type":"Router","ieeeAddr":"0x00158d0000cfd1ad","nwkAddr":56409,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RB 162\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1,2],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"3V8aCLnE8EE96sNv"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0002b87c10","nwkAddr":31432,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,1026,1027,1029,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"PrvvGIB8hWMeeLKI"}
[... cut ...] d":256,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"acCshH6lhUB0L96c"}

Koenkk, 2019-01-10:
Add to devices.js:

```js
{
zigbeeModel: ['RB 162\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
model: 'RB 162',
vendor: 'Innr',
description: 'E27 Bulb',
supports: generic.light_onoff_brightness().supports,
fromZigbee: generic.light_onoff_brightness().fromZigbee,
toZigbee: generic.light_onoff_brightness().toZigbee,
},
```

matija88, 2019-01-10:
I tried that a couple of times it didn,t work i am still getting:
  zigbee2mqtt:warn 2019-1-10 22:04:20 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:20 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:26 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:26 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:26 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:26 Message without device!
  zigbee2mqtt:warn 2019-1-10 22:04:56 Message without device!
  zigbee2mqtt:info 2019-1-10 22:04:56 Device incoming...
  zigbee2mqtt:info 2019-1-10 22:04:56 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2019-1-10 22:04:56 Device with modelID 'RB 162is not supported.
  zigbee2mqtt:warn 2019-1-10 22:04:56 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-1-10 22:04:56 Device with modelID 'RB 162

matija88, 2019-01-13:
Sure, with a bit of delay. Thanks for your support.
[image]

Koenkk, 2019-01-13:
Looks good, can you post your `devices.js`?

matija88, 2019-01-13:
here you go.
[image]

Koenkk, 2019-01-13:
Also looks good, I have no clue why it's not recognized.

Can you post your zigbee2mqtt log?

matija88, 2019-01-13:
yes.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2753495/log.txt)

henriz, 2019-01-13:
But now support for devices which have ID's which exists ONLY out of bytes, will not work anymore.

matija88, 2019-01-13:
Ok, I updated to zigbee2mqtt-edge stil not ok, here are cofig/log files. Hope i done this right.

[image]
[image]
[image]

Koenkk, 2019-01-14:
Is the device in the network? Only 1 devices is logged on startup. Also for text based files, could you please use: https://pastebin.com/

Koenkk, 2019-01-14:
Please update to the latest dev branch. This will strip the null characters from the model ID.

Putting in devices.js should be sufficient:

```js
{
zigbeeModel: ['RB 162'],
model: 'RB 162',
vendor: 'Innr',
description: 'E27 Bulb',
supports: generic.light_onoff_brightness().supports,
fromZigbee: generic.light_onoff_brightness().fromZigbee,
toZigbee: generic.light_onoff_brightness().toZigbee,
},
```

matija88, 2019-01-14:
I think I understand the problem now but, it seems that I don't know how to update to the latest dev branch I completely remove zigbee2mqtt-edge from hassio and the reinstall it also tried this command git clone https://github.com/koenkk/zigbee2mqtt.git -b dev

This is database.db
[LahttuZX.txt](https://github.com/Koenkk/zigbee2mqtt/files/2757191/LahttuZX.txt)

henriz, 2019-01-14:
just use git clone https://github.com/koenkk/zigbee2mqtt.git, then switch to the folder where it is checked out and do:

`git checkout dev
`

matija88, 2019-01-14:
Tried this command i think is same as the one i mentioned in previous post, `git checkout dev ` gave me `Already on 'dev'
Your branch is up to date with 'origin/dev'.`.
Followed by restart of zigbee2mqtt, again with no promising results log is pretty much same and this is database.db:
[eAsVQdZ0.txt](https://github.com/Koenkk/zigbee2mqtt/files/2757371/eAsVQdZ0.txt)

matija88, 2019-01-15:
tried that, got this.
`bash-4.4# rm -rf node_modules && npm install

> serialport@6.2.2 install /app/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=8.14.0 runtime=node arch=x64 platform=linux)
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:484:19)
gyp ERR! stack     at PythonFinder.<anonymous> (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:406:16)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:16)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Linux 4.15.0-43-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /app/node_modules/serialport
gyp ERR! node -v v8.14.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-15T19_17_08_040Z-debug.log
bash-4.4# 
`

Koenkk, 2019-01-15:
Can you install python? `sudo apt-get install python`.

matija88, 2019-01-15:
What did work was `apk add python` , and then I got this.

`bash-4.4# rm -rf node_modules && npm install

> serialport@6.2.2 install /app/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=8.14.0 runtime=node arch=x64 platform=linux)
gyp ERR! build error 
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:13:12)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:19)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Linux 4.15.0-43-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /app/node_modules/serialport
gyp ERR! node -v v8.14.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-15T19_44_22_669Z-debug.log
bash-4.4# 
`

Koenkk, 2019-01-15:
What os/setup are you using?

henriz, 2019-01-15:
Try:

`apk add --update make`

matija88, 2019-01-15:
I am running ubuntu, where in Docker container I am runing Home Assistant (hassio), then inside of hassio I installed addon zigbee2mqtt to which i connect with this command `sudo docker exec -it  addon_7ad98f9c_zigbee2mqtt-edge bash`.

I tried`apk add --update make` it gave me:

bash-4.4# apk add --update make
fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/x86_64/APKINDEX.tar.gz
(1/1) Installing make (4.2.1-r2)
Executing busybox-1.28.4-r1.trigger
OK: 187 MiB in 47 packages
bash-4.4# rm -rf node_modules && npm install

> serialport@6.2.2 install /app/node_modules/serialport
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=8.14.0 runtime=node arch=x64 platform=linux)
make: Entering directory '/app/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
make: g++: Command not found
make: *** [serialport.target.mk:104: Release/obj.target/serialport/src/serialport.o] Error 127
make: Leaving directory '/app/node_modules/serialport/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.15.0-43-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /app/node_modules/serialport
gyp ERR! node -v v8.14.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@6.2.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@6.2.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-15T20_02_13_012Z-debug.log

henriz, 2019-01-15:
`apk add build-base gcc abuild binutils binutils-doc gcc-doc`

Koenkk, 2019-01-15:
Just update to the latest zigbee2mqtt hassio edge version, (edge is in sync with the latest dev branch).

matija88, 2019-01-15:
[jkdjh8Sr.txt](https://github.com/Koenkk/zigbee2mqtt/files/2761289/jkdjh8Sr.txt)

henriz, 2019-01-15:
`apk add linux-headers`

Probably, @koenkk's answer is more easy for you.

closed 2019-03-23 by stale[bot]