Koenkk/zigbee2mqtt #834: Aqara water leak as undefined

marcotini, 2019-01-09:
`  zigbee2mqtt:warn 2019-1-9 21:31:59 Device with modelID 'undefined' is not supported.
`

zigbee2mqtt version 1.0.1 and I don't know why when the sensor is been triggered it doesn't even send a modelID.

The Aqara water leak should be supported: https://koenkk.github.io/zigbee2mqtt/information/supported_devices.html

sehraf, 2019-01-09:
Have you tried to repair / properly paired it? Xiaomi sensors trend to need a little bit longer to be completely paired.

marcotini, 2019-01-10:
@sehraf and how can I do it? I’ve try to reset it but it remain on static blue/two time blinking.

Koenkk, 2019-01-10:
@marcotini follow the pairing procedure from: https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html, you have to keep pressing the button on the sensor until it is paired.

marcotini, 2019-01-10:
@Koenkk yes it’s already paired (the problem is the “undefined” and “not supported” problem). I don’t know how to reset since if I keep pressing the button, it’s blink like one or two times (even if I unplug the battery).

Koenkk, 2019-01-10:
First hold the button for five seconds, than keep pressing it.

If that doesn't work:

remove all lines of this device from `database.db` and restart zigbee2mqtt ->repair

marcotini, 2019-01-10:
Ok I’ll try. So repair mean delete all lines of this sensor? And what about the other files like state.json and configurarono.yaml?

Koenkk, 2019-01-10:
You only have to delete it from `database.db`

marcotini, 2019-01-10:
Nope it's not working, after 4 second of pressing, the sensor blink one time and then the led it's flickering and then nothing... it won't pair again. I can make a video for explain the situation but I don't know...

sehraf, 2019-01-10:
but you keep pressing the button regularly? :thinking:

marcotini, 2019-01-10:
Yes of course, I've already pair 10 devices.

closed 2019-03-18 by stale[bot]