Koenkk/zigbee2mqtt #836: Disable router led

danpowell88, 2019-01-11:
Is it possible to disable both the router leds completely?

Would be handy if it could somehow be controlled via the yaml configuration for the router.

marcotini, 2019-01-11:
Modify your `configuration.yaml` with:

```
serial:
  disable_led: true
  port: /dev/ttyACM0
```

danpowell88, 2019-01-11:
That is for coordinators only

Koenkk, 2019-01-11:
AFAIK this is not possible with the router firmware.

closed 2019-03-19 by stale[bot]

dmak, 2019-07-13:
Just in case somebody finds this issue: `disable_led: true` worked fine for me for CC2531 on firmare [Z-Stack_Home_1.2 Default](https://raw.githubusercontent.com/Koenkk/Z-Stack-firmware/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip).

emilklindt, 2020-01-08:
In case that anyone is here, looking for ways to disable the LED on a CC2531 (with [Z-Stack router firmware for CC2531](https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531)). It can simply be done by powering on the dongle and pressing the S1 button – this is the button not saying debug next to it.

barrymossel, 2020-04-18:
> In case that anyone is here, looking for ways to disable the LED on a CC2531 (with [Z-Stack router firmware for CC2531](https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531)). It can [...]

Didn't work for me. Heat shrink did work though!

JiriFurda, 2020-06-16:
`disable_led: true` works but it needs to be added to the zigbee2mqtt addon configuration not the general `configuration.yaml`

realjax, 2020-07-11:
> `disable_led: true` works but it needs to be added to the zigbee2mqtt addon configuration not the general `configuration.yaml`

What is 'zigbee2mqtt addon configuration', where do I find that ?

emkamphuis, 2020-08-13:
For the router, pressing S1 button turns green LED on/off, but red LED keeps flashing. Have not found a way to disable flashing light.

Sloth-on-meth, 2020-11-03:
CC2531 routers still have this problem, any news?

877dev, 2020-11-29:
> For the router, pressing S1 button turns green LED on/off, but red LED keeps flashing. Have not found a way to disable flashing light.

This worked for me, thanks.

izocor, 2020-12-22:
You can disable led in zigbee2mqtt add-on configuration. You need install this add-on and configure before this manipulation.
This configuration file you can find Supervisor->Dashboard->Zigbee2mqtt->Configuration in section 
serial:
  disable_led: true #Just add this line
  port: /dev/ttyACM0
Save config and restart add-on

Sloth-on-meth, 2020-12-22:
This is for ROUTERS not COORDINATORS

cagarrinho, 2021-02-24:
disable_led: true
Works for me

Sloth-on-meth, 2021-02-25:
Again this is for ROUTERS NOT COORDINATORS

AbeltjeNL, 2021-03-03:
Facing this same question. PTVO.info has the option to make a configurable firmware:

https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/

Option: 'Status LED'.

Didn't try this yet but planning so on doing this week.

michaelmcdonald, 2021-03-23:
Has there been any progress on this? I find the blinking red LED on my router sticks rather annoying, especially at night. I can use a physical hack to cover it / block it, but would prefer something cleaner.

AbeltjeNL, 2021-03-23:
> Has there been any progress on this? I find the blinking red LED on my router sticks rather annoying, especially at night. I can use a physical hack to cover it / block it, but would prefer something cleaner.

Yes. I've tried the configurable firmware and yes you can disable the led.

michaelmcdonald, 2021-03-27:
> > Has there been any progress on this? I find the blinking red LED on my router sticks rather annoying, especially at night. I can use a physical hack to cover it / block it, but would prefer something cleaner.
> 
> Yes. I've tried the configurable firmware and yes you can disable the led.
> You can also change it from the red to the green led, invert it or change the blinking interval in seconds.

Can you share an example?

AbeltjeNL, 2021-05-19:
@michaelmcdonald and @duceduc,

Sorry. Didn't get a notification for the previous post. What do you want to see? A photo of the CC2531 with the led off or how to use the configurable firmware file?

duceduc, 2021-05-19:
Hi. Thanks for the reply. I have a CC2531 and have flash with the latest router firmware. I wish to disable the red LED.
 
Yes, how to use the configurable firmware file for the CC2531.

duceduc, 2021-05-20:
Hi. Thank you very much for the steps. I have saved the custom firmware with the LED box unchecked.
For steps 12 & 13, do we really need to have zigbee2mqtt `permit join` set to true during the flashing?

AbeltjeNL, 2021-05-20:
> Hi. Thank you very much for the steps. I have saved the custom firmware with the LED box unchecked.
> 
> For steps 12 & 13, do we really need to have zigbee2mqtt `permit join` set to true during the flashing?

Well, no. But the LED is turned off so you couldn't tell when the router is booting, searching for a network or joined a network. To be sure that the router joins your network after flashing, enable permit join. If you don't want to, enable it as soon as the flashing is done.

duceduc, 2021-06-24:
@AbeltjeNL 
Hello. Well, I successfully paired the router to coordinator during the time of flashing. I had to reset my coordinator and repair all devices including the cc2531 router. Now, I cannot get it to pair. There are 2 buttons on the board. S1 and S2. Does pressing the S2 to start repairing works? I have tried, but it doesn't work.

AbeltjeNL, 2021-06-24:
@duceduc, Nice to read that you got your router paired again. 

For the description and image part, i already told @ptvoinfo and @Koenkk.

AbeltjeNL, 2021-08-18:
FYI:

I stopped using these routers. 5 to be exact. My network is much more responsive and quicker. The CC2531 is a bit out of date and the fact that there is still no Zigbee 3.0 firmware out there made me decide to terminate these suckers. 
My network now only communicates with and through 3.0 devices and that is a big jump in performance. 

Not implementing that the Zigbee 1.2 firmware is bad or crap, it just wasn't working very well in my setup 👍🏻