Koenkk/zigbee2mqtt #837: Cannot get the Node Descriptor of the Device: 0x00158d00028b6b2f

sander816, 2019-01-11:
Bug report
 when adding a MCCGQ01LM mijia door sensor i get
 Cannot get the Node Descriptor of the Device: 0x00158d00028b6b2f

I have succesfully connected 2 MCCGQ01LM aqara sensors though

https://www.gearbest.com/smart-light-bulb/pp_257677.html?wid=1433363&currency=EUR&vip=2275151&gclid=Cj0KCQiAmuHhBRD0ARIsAFWyPwiJkmS8xakfysG3pIX6zJjxIPlvwuuvxFGACLu2s41geaEm3cmNnGwaAvB_EALw_wcB

sander816, 2019-01-11:
same goes for the aqara cub:
 Cannot get the Node Descriptor of the Device: 0x00158d000278e797 (Error: Timed out after 10000 ms)

subzero79, 2019-01-11:
Get closer to the coordinator when pairing.  Also if you have a muti meter check the device battery voltage it might be low.

Koenkk, 2019-01-11:
Also see: https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html#xiaomi

sander816, 2019-01-11:
i tried at 1 meter, no difference, 
battery voltage is 2.88v

when pairing and pressing the button every second:
```
 zigbee2mqtt:warn 1/11/2019, 11:17:21 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:17:26 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:26 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:17:31 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:31 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:17:36 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:36 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
 PM Cannot get the Node Descriptor of the Device: 0x00158d00028b6b2f (Error: Timed out after 10000 ms)
  zigbee2mqtt:warn 1/11/2019, 11:17:43 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:43 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:17:47 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:47 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:17:53 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:17:53 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
```

sander816, 2019-01-11:
>    - Stop zigbee2mqtt
>    - Remove all entries of 0x00158d00028b6b2f from database.db
>    - Start zigbee2mqtt
>    - Follow
>    https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html#xiaomi
>    carefully

did that, but still:
  zigbee2mqtt:error 1/11/2019, 11:32:32 PM Cannot get the Node Descriptor
of the Device: 0x00158d00028b6b2f (Error: Timed out after 10000 ms)

sander816, 2019-01-11:
same error when pressin the button:
 zigbee2mqtt:warn 1/11/2019, 11:35:29 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:35:29 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:35:35 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/11/2019, 11:35:35 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 1/11/2019, 11:35:36 PM Message without device!

sensor works fine in mihome

sander816, 2019-01-11:
hmm, after some time it recognized the sensor:
```
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00028b6b2f","nwkAddr":14432,"manufId":4151,"modelId":"lumi.sensor_magnet","epList":[1],"status":"online","joinTime":15472461$
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00028b6b2f","nwkAddr":14432,"manufId":4151,"modelId":"lumi.sensor_magnet","epList":[1],"status":"online","joinTime":15472461$

 zigbee2mqtt:info 1/11/2019, 11:40:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":true,"linkquality":70}'
  zigbee2mqtt:info 1/11/2019, 11:40:52 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":false,"linkquality":73}'
  zigbee2mqtt:info 1/11/2019, 11:40:53 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":true,"linkquality":70}'
  zigbee2mqtt:info 1/11/2019, 11:40:54 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":false,"linkquality":76}'
  zigbee2mqtt:info 1/11/2019, 11:42:26 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":true,"linkquality":70}'
  zigbee2mqtt:info 1/11/2019, 11:42:27 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":false,"linkquality":70}'
  zigbee2mqtt:info 1/11/2019, 11:42:27 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00028b6b2f', payload '{"contact":true,"linkquality":68}'

```

Koenkk, 2019-01-12:
Great!

closed 2019-01-12 by Koenkk

sonuyos, 2019-06-01:
This issue should be fixed, this still happens.

wes1993, 2019-08-05:
Hello, i have the same issue, how can i solve the problem?
i'm trying to add the Xiaomi DJT11LM sensor but i can't (zigbee2mqtt:error 8/5/2019, 9:59:48 AM Cannot get the Node Descriptor of the Device: 0x00158d0002af97b2 (Error: Timed out after 10000 ms)) i don't have any string of this device in the database.
i have also 2 ikea tradfri smart switch (plug) that i think work like a router.
The fw of my coordinator is the latest 3.0
Thanks a lot for all your help.

petjek, 2019-08-20:
Same issue here. I successfully paired a Xiaomi Cube but all of my three Xiaomi Aqara Buttons failed when pairing. This is quite strange because everything worked fine with my former ioBroker-Installation. Now that I'm back to FHEM and installed zigbee2mqtt by hand it doesn't.

Nihvel, 2019-08-26:
Same.
`zigbee2mqtt:error 8/26/2019, 7:43:40 PM Cannot get the Node Descriptor of the Device: 0x00158d000313116c (Error: Timed out after 10000 ms)`

This device is a Aqara vibration sensor.
I'm running on the latest firmware on the coordinator: 20190608
And on the latest software revision: 1.5.1.2

Nihvel, 2019-08-26:
I tried to follow this: https://github.com/Koenkk/zigbee2mqtt/issues/1779#issuecomment-515774316

But I got this:
```
  zigbee2mqtt:info 8/26/2019, 8:09:33 PM New device 'lumi.vibration.aq1' with address 0x00158d000313116c connected!
/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:756
      throw new YAMLException('unacceptable kind of an object to dump ' + type);
      ^
YAMLException: unacceptable kind of an object to dump [object Undefined]
    at writeNode (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:756:13)
    at dump (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:817:7)
    at Object.safeDump (/zigbee2mqtt-1.5.1/node_modules/js-yaml/lib/js-yaml/dumper.js:823:10)
    at Object.writeYaml (/zigbee2mqtt-1.5.1/lib/util/fs.js:13:33)
    at write (/zigbee2mqtt-1.5.1/lib/util/settings.js:139:12)
    at writeRead (/zigbee2mqtt-1.5.1/lib/util/settings.js:122:5)
    at addDevice (/zigbee2mqtt-1.5.1/lib/util/settings.js:206:5)
    at Object.addDevice (/zigbee2mqtt-1.5.1/lib/util/settings.js:390:30)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:68:22)
    at extensions.filter.forEach (/zigbee2mqtt-1.5.1/lib/controller.js:143:31)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm 
ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

And the application restarts due to this crash. This is the second time this happen.

piit79, 2019-10-12:
Just in case somebody finds this and is still struggling, here's how I solved it: after you reset the switch by pressing the reset button for ~5 seconds, wait a couple of seconds and then keep (single) pressing the reset switch every second.

I managed to connect one switch without problems but was then struggling with two other switches and trying 10+ times unsuccessfully... Until I discovered this method. Worked first time for both switches.

CASATAJO, 2019-10-20:
What is the difference between the button and the facepalm reset button? Is not the same one?

piit79, 2019-10-20:
@CASATAJO Not sure if serious... But in case you are: the (facepalm) was a kind of "text emoji" - I kept pressing the actual button instead of the reset button.

[image]

CASATAJO, 2019-10-20:
Thanks. I was. English is not my regular language. So was not sure what you mean

JohnAnderton, 2019-12-06:
Had the same error for door sensor Xiaomi MCCGQ01LM.

`Cannot get the Node Descriptor of the Device: 0x00158d00039c7d4a (Error: Timed out after 10000 ms)`

This helped: https://github.com/Koenkk/zigbee2mqtt/issues/1779#issuecomment-515774316

I suggest to change the pairing guide here: https://www.zigbee2mqtt.io/devices/MCCGQ01LM.html