Koenkk/zigbee2mqtt #839: Unable to pair RTCGQ11LM

kloknibor, 2019-01-11:
I try to add RTCGQ11LM to my set-up. I have 2 new units of these but there seems no way to add these to my set-up. Steps I try to pair : 

Press button for +/- 5 seconds
Light flashes 3 times and 1 more time when button released.
I wait and the log doesn't show anything about a new device... I also tried to keep pressing the button every second without luck and I tried it from a really close distance. I already tried for some hours with any succes, does anybody have any suggestion about what I do wrong?

Thank in advance!

Log :  https://hastebin.com/ovejajopog.apache

kloknibor, 2019-01-11:
Additional, I tried to remove devices.db and start with a fresh one also without any luck... 

Log : https://hastebin.com/ijinuparac.sql

Koenkk, 2019-01-11:
Sometimes it also help to remove the battery of the device for a few seconds and try again.

If that doesn't work, try a new battery.

iganin67, 2019-01-27:
I think I have tried everything including removing and changing the battery and decreasing the distance up to 1cm. RTCGQ01LM was paired in a few seconds. RTCGQ11LM doesn't want to pair. With Xiaomi gateway works flawless.

Koenkk, 2019-01-28:
@iganin67 could it be that your Xiaomi gateway is still powered on?

clockbrain, 2019-01-28:
@iganin67 not sure if this is relevant as it applies to MCCGQ11LM devices but they could follow similar pairing process to RTCGQ11LM.

From [https://community.smartthings.com/t/xiaomi-vs-aqara-door-window-sensors/90738/152](https://community.smartthings.com/t/xiaomi-vs-aqara-door-window-sensors/90738/152)

1 hold sensor’s reset button for 3 seconds until light flashes twice on the Aqara / 3 long flashes on the original Xiaomi) and then if the light flashes again only once,

2 reset (tap the sensor’s reset button, which coincides with a single flash on the Aqara / double-flash on the original Xiaomi), and after tapping the reset button, if the light flashes quickly 3 times, you’re paired, just be patient for the app to show the linked sensor. If you see only one flash, repeat this reset step again, until you see the 3 quick flashes.

closed 2019-04-05 by stale[bot]

errolsancaktar, 2020-03-27:
tried readding one device here is error:
```zigbee2mqtt    | zigbee2mqtt:debug 2020-03-27 14:07:15: Received Zigbee message from 'Basement_Motion', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
zigbee2mqtt    | zigbee2mqtt:error 2020-03-27 14:07:15: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
zigbee2mqtt    |     at /app/lib/extension/deviceReceive.js:106:42
zigbee2mqtt    |     at Array.filter (<anonymous>)
zigbee2mqtt    |     at DeviceReceive.onZigbeeEvent (/app/lib/extension/deviceReceive.js:105:52)
zigbee2mqtt    |     at Controller.callExtensionMethod (/app/lib/controller.js:346:44)
zigbee2mqtt    |     at Controller.onZigbeeEvent (/app/lib/controller.js:231:14)
zigbee2mqtt    |     at Zigbee.emit (events.js:311:20)
zigbee2mqtt    |     at Controller.<anonymous> (/app/lib/zigbee.js:58:52)
zigbee2mqtt    |     at Controller.emit (events.js:311:20)
zigbee2mqtt    |     at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:463:22)
zigbee2mqtt    |     at Generator.next (<anonymous>))
zigbee2mqtt    | 2020-03-27T20:07:15.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

sergimola, 2020-11-13:
Today I received two of these and I am having literally exactly the same problem as the original reporter.
Tried everything and nothing worked. 
@kloknibor did you manage to get it to work?

sergimola, 2020-11-13:
> Today I received two of these and I am having literally exactly the same problem as the original reporter.
> Tried everything and nothing worked.
> @kloknibor did you manage to get it to work?

Finally made it work.. 
1 - Stop zigbee2mqtt
2 - Unplug the usb dongle (cc2531 in my case)
3 - Start zigbee2mqtt
4 - Press the pairing button for 4 seconds until the light flashes 3 times and then a single one. 

It will shortly be detected and start the pairing process. The light will start flashing as well.

kloknibor, 2020-11-13:
I think this might have been solved by using another light switch. Some does not seem to disconnect the power correctly or something...

Merwenus, 2021-08-02:
Thank you sergimola, This is still an issue in the current version of zigbee2mqtt and I'm glad google helped me to find your comment :)

gebauer, 2022-01-28:
Hi,

I had the same problem. 
In my case @clockbrain 's solution helped.

First presse 3+ second until you see the 3 short flashes (+ 1 long flash after you have releases the button)
Then press again shortly the button.

After that it connected...

Perhaps this should be included in the docu?

Best & thanks
Jan

danstreeter, 2022-04-15:
I had this problem with one of my PIR sensors, only after performing a shutdown of zigbee2mqtt, unplugging the USB CC2531 stick, re-plugging it in, and restarting zigbee2mqtt was I then able to pair it via a ~5 second hold of the reset button upon then it paired instantly without issue.

As @gebauer has mentioned - I think this may be worth mentioning in the docs as a potential edge case thing to try when all other advice is not working.

zelial, 2024-07-22:
FTR I had this same issue when migrating my 20+ Aqara sensors from deconz to z2m. Very frustrating experience, took 5-10 pairing attempts (with dongle reboots) to get a sensor paired. Then I upgraded my dongle firmware (from EZSP to ember) via https://darkxst.github.io/silabs-firmware-builder/ (and switched z2m to this adapter) and the issue was gone. With all my devices - I had to repair them all. But it worked on first attempt as it should!