Koenkk/zigbee2mqtt #840: configuration variable "json_attributes" is deprecated, replace with "json_attributes_topic"

h4nc, 2019-01-11:
I replaced all my json_attributes with json_attributes_topic and I still get a bunch of those errors in my ha log.
I think those errors are thrown by the autodiscovered devices which most probably still use json_attributes.

Am I right? How to fix this. Will there be an update soon?

Koenkk, 2019-01-11:
As I understand correctly `json_attributes_topic` is more than only a name change. What exactly did you change/

h4nc, 2019-01-11:
You are right. I use an example to clarify that:

old:
```
- platform: "mqtt" 
  name: zigbee_cube_bat
  state_topic: "zigbee2mqtt/Zigbee Cube"
  value_template: "{{ value_json.battery | round(0) }}"
  device_class: battery 
  json_attributes: 
    - "linkquality"
```
new:
```
- platform: "mqtt" 
  name: zigbee_cube_bat
  state_topic: "zigbee2mqtt/Zigbee Cube"
  value_template: "{{ value_json.battery | round(0) }}"
  device_class: battery 
  json_attributes_topic: "zigbee2mqtt/Zigbee Cube"
```

So it looks like with this it isn't possible any more to chose which attributes you want to add, because this will add all of the attributes. Even the one that is already chosen for the value_template.

Koenkk, 2019-01-11:
Yes, what happens with the battery in this case? Is that also added as a json attribute? (because it's also the value of the sensor itself)

h4nc, 2019-01-11:
The state of the battery will be shown as state and as an attribute.

I added a picture here:
https://community.home-assistant.io/t/json-attributes-topic-json-attributes-deprecated/90652

I liked the solution before better, but it seems like we have to live with that (because json_attributes is deprecated)

Koenkk, 2019-01-11:
Yes, it even get's worse when you think of e.g. the xiaomi weather sensor, which has 4 sensors (humidity, temperature, pressure, battery). All these 4 sensors will have the humidity, temperature, pressure and battery as a json attribute.

h4nc, 2019-01-11:
That's really annoying. Hopefully there will be a satisfying solution. 

By the way: I remember another topic about binary_sensors and json_attributes. Json_attributes where not supported by binary_sensors. With json_attributes_topic now it looks like this is added. Maybe this is the reason for the change?
https://www.home-assistant.io/components/binary_sensor.mqtt/

tarbax, 2019-01-13:
how do i fix this i have zigbee2mqtt with autodiscovery so i dont have the yaml

Koenkk, 2019-01-14:
Fixed in the dev branch, can we close this issue?

h4nc, 2019-01-14:
Looks like this fix only deletes the Jason attributes?
Am I right? This would mean attributes are gone? Not really a step forward, right?

emontnemery, 2019-01-14:
@Koenkk I opened a PR to add support for `json_attributes` to MQTT binary sensor, and balloob (Home Assistant owner) requested to instead add support for a `json_attributes_topic` and also do the same change to sensor.
Apparently this had been discussed several times before in the hass community, and my PR was the last drop.

Is there a reason why a `json_attributes_topic` does not work for zigbee2mqtt?

Koenkk, 2019-01-14:
@emontnemery  thanks for letting us know

@h4nc it has been replaced with json_attributes_topic

h4nc, 2019-01-14:
@Koenkk can't see that it has been replaced here:
https://github.com/Koenkk/zigbee2mqtt/commit/6912f8108d034b5233b00e834180a7cdf66b339c
This only looks like it was removed.

@emontnemery so cherrypicking will not be available any more? Are there plans to readd this?

Koenkk, 2019-01-14:
@h4nc its added dynamically based on the type https://github.com/Koenkk/zigbee2mqtt/commit/6912f8108d034b5233b00e834180a7cdf66b339c#diff-0c056233b68fd3653afb296f98947c8fR516

h4nc, 2019-01-14:
Do you know how long it usually takes to get this changes to the hassio addon?

jonydebu, 2019-01-14:
Same issue here with hassio , hope the changes will be made as soon as posible . Great job Koenkk .

emontnemery, 2019-01-15:
@Koenkk would it be useful to have `json_attrubutes_topic` enabled also for other platforms than sensor and binary_sensor?

emontnemery, 2019-01-15:
@Koenkk Can you list the platforms you need it for, and I'll try to get it done within the next couple of weeks. Another way would be to create a sensor only with the additional statistics and link it with the real device through device.identifier. I.e. for each device create a real entity (switch, light, sensor) and an additional sensor entity only with the stats.

Koenkk, 2019-01-15:
@emontnemery `light`, `switch`, `cover`

h4nc, 2019-01-15:
@Koenkk I'm afraid if i switch to etch I will have to reconfigure all my sensors again. I'm not sire about that. This would be a lot of work (many sensors). Is there an easy way to switch?

emontnemery, 2019-01-17:
@Koenkk PRs now submitted to home-assistant:
home-assistant/home-assistant#20190
home-assistant/home-assistant#20191
home-assistant/home-assistant#20192

vpomax, 2019-01-18:
I use Hassio Supervisor 142 with HA v. 0.85.1
it show me "configuration variable "json_attributes" is deprecated, replace with "json_attributes_topic""
Have to wait a upgrade of zigbee2mqtt to solve the problem?

h4nc, 2019-01-18:
This is already solved in edge (dev addon).

You can easily switch to edge or wait for an update.

closed 2019-01-18 by h4nc

vpomax, 2019-01-18:
must I replace addon zigbee2mqtt with zigbee2mqtt-edge?

h4nc, 2019-01-18:
yes, i did this some days ago. 

1) stop zigbee2mqtt 
2) copy your config (in zigbee2mqtt  addon)
3) install zigbee2mqtt-edge 
4) paste your config in zigbee2mqtt-edge
5) start zigbee2mqtt-edge
6) everything should work as before

don't forget to disable autostart of zigbee2mqtt

hawaltie, 2019-01-26:
Thank you for all the good work!

When will this patch be be available in the stable branch?

hawaltie, 2019-01-26:
Thanx. Then I'll just have to be patient.

lassieee, 2019-02-08:
I have these errors in my homeassistant logs too. I'm running latest HA (docker), latest zigbee2mqtt (docker) and latest z-stack firmware. I don't have any files with "json_attributes" so i'm kinda lost. I just tried the latest arm32v6-dev but I still have these errors.
Any idea's?
Thanks

Koenkk, 2019-02-08:
@lassieee you should clear the retained messages from your MQTT broker.

lassieee, 2019-02-09:
Thanks @Koenkk . I ended up deleting mosquitto.db and all my troubles went away :-)
Thank you very much for this awesome project 🥇

marcgarciamarti, 2019-03-09:
@Koenkk I'm a bit confused with this. Could you clarify whether we need to do something when we see warning messages like this
`[homeassistant.components.mqtt.sensor] configuration variable "json_attributes" is deprecated, replace with "json_attributes_topic"`
pop up in HA log file?
I really tried to understand the conclusions here but I'm not sure.
I would appreciate it if you clarified this.
Thanks

Koenkk, 2019-03-09:
@marcgarciamarti you need to clear the retained MQTT messages. (e.g. for mosquitto: https://community.openhab.org/t/clearing-mqtt-retained-messages/58221)

zepixir, 2019-03-10:
@lassieee @Koenkk  I also deleted the mosquitto database file. But now I get these error messages in mosquitto addon logs:
```
1552246264: |-- aclcheck(zepixir, zigbee2mqtt/+/+/+/set, 4) CACHEDAUTH: 0
1552246264: |-- mosquitto_auth_acl_check(..., client id not available, zepixir, zigbee2mqtt/+/+/+/get, MOSQ_ACL_WRITE)
```

Zigbee2mqtt addon logs says it is connected to the server. But nothing of my zigbee devices automations works. The devices are also unavailable in homeassistant integrations under mqtt.

Anything you experienced?

Koenkk, 2019-03-10:
Did you restart zigbee2mqtt afterwards?

zepixir, 2019-03-10:
@Koenkk yes, tried to restart the zigbee2mqtt (edge version), and restartet my intel nuc 2 times without luck. I could find the new mosquitto.db file after I reinstalled the mosquitto add-on.

Koenkk, 2019-03-11:
Can you provide the zigbee2mqtt log when it publishes messages?

zepixir, 2019-03-17:
@Koenkk the problem was with the update of mosquitto addon from v4 to v4.1. No problem with v4.

jeppper, 2020-11-29:
What is the correct procedure for implementing the cube into Home Assistant?
I have tried this from the guide https://www.zigbee2mqtt.io/devices/MFKZQ01LM.html
Pairing with z2m is okay.

sensor:
  - platform: "mqtt"
    state_topic: "zigbee2mqtt/xiaomi_cube"
    availability_topic: "zigbee2mqtt/bridge/state"
    value_template: "{{ value_json.action }}"

but i guessed its outdated and tried to change it to:

sensor:
  - platform: "mqtt"
    name: "xiaomi_cube"
    state_topic: "zigbee2mqtt/xiaomi_cube"
    availability_topic: "zigbee2mqtt/bridge/state"
    value_template: "{{ value_json.action }}"
    json_attributes_topic: "zigbee2mqtt/xiaomi_cube"

Anyway nothing happens in HA. State is always "unknown"

In configuration.yaml in z2m I simply have:
'0x00158d00010b28a2':
  friendly_name: xiaomi_cube

Must something be added?