Koenkk/zigbee2mqtt #841: router not showing in config anymore

danpowell88, 2019-01-12:
Using the latest router firmware for cc2531 when I pair it with the network it no longer seems to add an entry into the configuration.yaml, I've also checked database.db and nothing in there has an address that matches.

The router however does appear to be on the network as its flashing every few seconds once and debug output shows 

```
PANID:1A62,Ch:11,Join:1
PANID:EA6A,Ch:25,Join:0

▒A▒▒PANID:1A62,Ch:11,Join:1
PANID:EA6A,Ch:25,Join:0

/0x0017880103434C6E,NEIGHBOR
9091/0x0017880104E90115,NEIGHBOR
55965/0x00178801035609F4,NEIGHBOR
9995/0x001788010356087D,NEIGHBOR
23010/0x0017880102B0A9D7,NEIGHBOR
/0x0017880103434C6E,NEIGHBOR
9091/0x0017880104E90115,NEIGHBOR
55965/0x00178801035609F4,NEIGHBOR
9995/0x001788010356087D,NEIGHBOR
23010/0x0017880102B0A9D7,NEIGHBOR
/0x0017880103434C6E,NEIGHBOR
9091/0x0017880104E90115,NEIGHBOR
55965/0x00178801035609F4,NEIGHBOR
9995/0x001788010356087D,NEIGHBOR
23010/0x0017880102B0A9D7,NEIGHBOR
/0x0017880103434C6E,NEIGHBOR
9091/0x0017880104E90115,NEIGHBOR
55965/0x00178801035609F4,NEIGHBOR
9995/0x001788010356087D,NEIGHBOR
23010/0x0017880102B0A9D7,NEIGHBOR
```

Ch11 is my zigbee2mqtt network and 25 is my hue one im migrating over.

I tried to do a mqtt publish to remove the address of the router, it completed successfully but had no change.

Is it possible that the co-ordinator itself still has this device in its memory somewhere from when I had it previously paired, and if so is there anyway to remove it without flashing and restarting the co-ordinator from scratch.

Koenkk, 2019-01-12:
Does the router produce any messages? Normally it should oublish to `genOnOff` every minute.

danpowell88, 2019-01-12:
Watching the logs I couldn't see any, I did see it flash 3 times every
minute or so though.

Is there a way to alter the IEEE address and see if its something related
to that

danpowell88, 2019-01-12:
I also just reset and reflashed the stick and it seemed to join the network without zigbee2mqtt running (also unplugged my hue hub)

Koenkk, 2019-01-12:
That's possible if the CC2531 (coordinator) is still powered.

closed 2019-03-21 by stale[bot]