Koenkk/zigbee2mqtt #842: zigbee2mqtt unreliable: queuing is wrong

henriz, 2019-01-12:
Hello,

I'd like to get in to the development of this plugin, to help improve it.

At it's current state, in my opninion, it's unreliable. It works for about half a day, and then starts acting up. And when it works, there's something wrong in the queing mechanism. In situations with multiple commands, it somehow seems to get clogged with commands.

This results in it doing nothing for a certain amount of times, and then all the sudden process the queue of commands. I understand there has been a change recently for situations with multiple commands, I think something went wrong there?

I haven't really looked into the code yet, but it feels to me like this app needs threading.

Also what I've noticed; when started the first time, on/off commands for lights do not work. First you have to set a level, which does work, before on/off commands to work. In general on/of does not work very wel, set level works way better.

Are you open to have other devvers join?

Koenkk, 2019-01-12:
Its open-source, meaning that everybody can join. 

However the above mentioned problems mostly occur due to bad network stability. 

About the bulb on/off not working, which bulb is this? This has not been reported before.

henriz, 2019-01-12:
Dear Koen,

Thanks, I'm not the kind of complain only and do nothing. I want to help, so I will.

Networking is not the issue here. There are 2 networks at play; TCP/IP and Zigbee. As far as TCP/IP goes, the MQTT broker is running on the same device as zigbee2mqtt, and the messages get to the MQTT broker just fine. So that excludes TCP/IP networking issues.

I monitor the Zigbee network with a secondary device, and the commands are simply not sent. Also, before switching to a pi with zigbee2mqtt and so forth, I used a hue hub, which has been running this network for like 2 year without ever having any hickup. So that also excludes Zigbee networking issues.

Also, note that eventually all commands ARE executed and that once this happens, it processes ALL the commands at once, in sequence, for all lights.  So it's hung up on something. There's a device or piece of code in there, which halts execution for like 20-40 seconds. This is also why I think this app needs threading. Is it waiting for an ack somewhere?

I'm pondering ditching domoticz and starting my own development for it, as I also want to use the new group functionality (is the scenes cluster also supported in zigbee-shepherd?).

As far as joining in, maybe I asked it wrong, but I'd like to be a more structural member of the development team, discuss with you what to do next, which area's to do and so forth. In stead of being a separate island developing and then doing pull requests.

henriz, 2019-01-12:
Is this the issue I'm having?
```

 cat configuration.yaml  | grep retain | wc
     34      68     612

```

So I have 34 devices. And all hue bulbs are routers.

marklagendijk, 2019-01-12:
My setup suffers from exactly the same issue. It works fine until I sent a lot of subsequent commands (for example dimming down and up again). It then doesn't respond for about a minute, and then executes all the commands at once.

henriz, 2019-01-12:
I might have found the culprit, but haven't confirmed this yet by lookint at and/or changing the code.

As I was on the track of network size, i first started looking at the firmware on the cc2531. The seller (marktplaats) told me he flashed the latest firmware version, so I didn't do anything in that respect. But this issue came along, so I flashed the firmware, the 'normal' firmware from Koen's repository.

Now I get log messages which I have never had before:

`zigbee2mqtt:error 1/12/2019, 7:24:52 PM Cannot get the Node Descriptor of the Device: 0x001788010211444a (Error: Timed out after 10000 ms)`

The device in question is a Philips Hue Motion sensor, and I get these messages every now and then for all of them (I have 4 of 'm in my network).

Might it be that that timeout is what is clogging the system?

Mark, do you have Hue motion sensors?

My zigbee setup contains only Philips devices, a mixture of different bulbs (all of the recent types and with the lates firmware), Philips dimmer switches and Philips motion sensors.

marklagendijk, 2019-01-12:
> Mark, do you have Hue motion sensors?
No motion sensors, just Hue Dimmers and Hue lights.

henriz, 2019-01-12:
Well then there is a similarity in our setup, as in we both use hue dimmers and lights.

What do you use to control them? Domoticz or ie. hass.io?

henriz, 2019-01-12:
Those PM message without device are normal, they are a side-effect of devices pairing.

@Koenkk Ik have now confirmed this is an issue in zigbee2mqtt and not a networking issue.

If I run a subscription to all mosquitto channels alongside the zigbee2mqtt output, I can see mosquitto is getting the messages, and that zigbee2mqtt processes them. Until it freezes, mosquitto is then still showing the messages immediatelly, but zigbee2mqtt is NOT.

As mosquito, zigbee2mqtt and the 'all channels subscription' are all running on the same server, there can be only one conclusion. Something is clogging zigbee2mqtt.

Here you can see this:
https://youtu.be/tMWPtOGWnqE

Koenkk, 2019-01-12:
The queue hangup can easily be solved by chaning this line https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/devicePublish.js#L24 to e.g. `this.queue.concurrency = 100;`

I implemented this because when sending commands to multiple devices (e.g. group of bulbs) at exactly the same time, some bulbs failed to respond. This is of course can now be solved with group support.

@henriz can you make the above mentioned change and check if you have any improvement?

henriz, 2019-01-12:
Well..diving into the code, directly I see an issue.  In the controller.js, the onMQTTMessage function:

```

const results = this.extensions
            .filter((e) => e.onMQTTMessage)
            .map((e) => e.onMQTTMessage(topic, message));
```

Basicaly what this does, if a MQTT message is received, this is forwarderd to ALL device objects, and the result of the call is mapped.

So in my case, I have 34 devices, this means 34 objects are getting this message, processing it, and returning the result.

The javascript .filter function is incredibly slow. Also, suppose in the extensions array, the 2nd extension is the one the message was intended for, it will still keep on going until the 34th message.

This can easily be optimized. I'll see if I can impove on it.

While I'm typing this, I see your answer, Koenkk. I'll try that right now!

henriz, 2019-01-12:
@Koenkk The suggested change does not change the behaviour in any way, unfortunatelly.

Koenkk, 2019-01-13:
Can you try removing the whole queueing mechanism from devicePublish?

```
const results = this.extensions
            .filter((e) => e.onMQTTMessage)
            .map((e) => e.onMQTTMessage(topic, message));
```

Will not call devices, it will just call extensions which can decide to handle this message.

henriz, 2019-01-13:
I'll try. In the mean time I'm starting to think this is more a zigbee-shepherd issue.

If I remove this part from zigbee.js:
```

if (cmdType === 'functional' && entity.functional) {
            entity.functional(cid, cmd, zclData, cfg, callback_);
        } else if (cmdType === 'foundation' && entity.foundation) {
            entity.foundation(cid, cmd, zclData, cfg, callback_);
        } else {
            logger.error(`Unknown zigbee publish cmdType ${cmdType}`);
        }
```

It functions fine and keeps on functioning. Well, it doesn't function, as nothing actually gets sent to the lights, but you get what I mean.

Also this new firmware has worsened the situation a lot. So right now I'm more battling to keep it to work at all, as oppose to testing this issue.

I get a lot of:
` zigbee2mqtt:error 1/13/2019, 8:01:11 PM Cannot get the Node Descriptor of the Device: 0x0017880102030d92 (Error: Timed out after 10000 ms)`

Especially on motion sensors. I flashed this firmware:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin

Using a Arduino Due and the alternate flashing method. If all goes well my 'real' flasher will arrive this week.

What firmware do you recommend?

This is my network, with a few devices missing:
 ```
zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102a817c6 (0x0017880102a817c6): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x00178801021d8dbf (0x00178801021d8dbf): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010385ee8d (0x001788010385ee8d): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x00178801021d7fac (0x00178801021d7fac): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010291d85b (0x001788010291d85b): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010356a7d0 (0x001788010356a7d0): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010209c9a0 (0x001788010209c9a0): 9290011370 - Philips Hue white A60 bulb E27 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010291cf97 (0x001788010291cf97): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102b2749b (0x0017880102b2749b): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102f14d18 (0x0017880102f14d18): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880103569fe9 (0x0017880103569fe9): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102b14ba8 (0x0017880102b14ba8): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102db5f82 (0x0017880102db5f82): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010286afe3 (0x001788010286afe3): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102e7700b (0x0017880102e7700b): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010356a156 (0x001788010356a156): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102dae44c (0x0017880102dae44c): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102e76fbd (0x0017880102e76fbd): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010291825d (0x001788010291825d): 8718696695203 - Philips Hue white ambiance E14 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x00178801022431b2 (0x00178801022431b2): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880103e7964e (0x0017880103e7964e): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880103e72fd7 (0x0017880103e72fd7): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x00178801020d6a11 (0x00178801020d6a11): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102d59136 (0x0017880102d59136): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102030d92 (0x0017880102030d92): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010212678e (0x001788010212678e): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102114e4e (0x0017880102114e4e): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x001788010211444a (0x001788010211444a): 9290012607 - Philips Hue motion sensor (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102fc9416 (0x0017880102fc9416): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880102fcb484 (0x0017880102fcb484): 8718696449691 - Philips Hue White Single bulb B22 (Router)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x0017880103e730ab (0x0017880103e730ab): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:info 1/13/2019, 7:39:41 PM 0x00178801021b46cd (0x00178801021b46cd): 324131092621 - Philips Hue dimmer switch (EndDevice)

```

Koenkk, 2019-01-13:
@henriz I'm currently running this firmware: https://drive.google.com/open?id=1Iu-vVJw_d0bKeQKQhcZbkpQCfGB3z-Bi

It's almost the same as the current z-stack-firmware/dev but with increased XDATA.

lolorc, 2019-01-14:
I would also wish there was an easier solution to prevent rebinding when upgrading the firmware. I'm shielding the stick and stop every routers to make sure it doesn't detect the existing panid/chanid. But it's far from convenient :)

All the issues you're talking about, have they appeared with latest version ? I'm still using dev from end of october, I'm a bit hesitant to upgrade.

about the firmware, I'm also a bit lot, I'm running a "optimised for larger network" cc2530+cc2591 firmware. Does the latest dev one with group support also have the optimisations ? (increased xdata i guess)

henriz, 2019-01-14:
So far, all newer versions have been a total disaster for me. Asides from the queue appearing to be clogging, I had it working fine. Since starting to upgrade, I haven't had it working fine again.

Oten NPM won't start, bound devices aren't reached, and so forth.

Koenkk, 2019-01-14:
@henriz this can be fixed by unplugging, plugging, press the button on the cc2531 (close to the usb) and start zigbee2mqtt.

lolorc, 2019-01-14:
I've just upgraded (silly me) I also had the "TypeError: Cannot read property 'close' of undefined" error.
Having to unplug/plug/press button to resolve that kind of issue is awful.
With my cc2530 I just had to restart zigbee2mqtt.

henriz, 2019-01-14:
I can't get it to start at all anymore. I'm starting to feel the end-devices (dimmerswitches/motion sensors) are reaking havoc on the firmware in the device.

@lolorc: I have already beaten myself for not noting the firmware version which was on the stick when I got it, as that was working reasonably well. Upgrading seems to have opened a whole can of worms for me. WIth no devices bound, it worked fine, I was able to bind every device, but as soon as I added the dimmerswitches and motion sensors, it simply stopped working at all.

Even if I now run the zigbee-shepherd simple sample script, it doesn't work.

So even this script:

```
var ZShepherd = require('zigbee-shepherd');
var shepherd = new ZShepherd('/dev/ttyACMZigbee');  // create a ZigBee server

shepherd.on('ready', function () {
    console.log('Server is ready.');

    // allow devices to join the network within 60 secs
    shepherd.permitJoin(60, function (err) {
        if (err)
            console.log(err);
    });
});

shepherd.start(function (err) {                // start the server
    if (err)
        console.log(err);
});

```

Won't start. What I think happens is that the device, when powered on, takes it's role and starts communicating. Somehow the motion sensors or dimmerswitches are causing it to time out.

This might also be the underlying issue that started this whole thread in the first place.

But it's all guesses.

tmarquespt, 2019-01-14:
@henriz 
I had the exact same issue as you. Sensors, switches and bulbs on/off functions worked as expected, but dimming lights was a pain. The 'network clogs', and from 20sec to 1 minute later, all the previous commands act in sequence. I ended up changing all the bulbs (including Ikea ones) to a Hue bridge, but still hoping to see them working in zigbee2mqtt one day

henriz, 2019-01-14:
Well I'm a persistant bugger and a developer, so I won't stop until this works.. it has become a personal matter now! >:)

henriz, 2019-01-14:
Well.. i finally got my zigbee network working again. The only way to accomplish that, was to re-flash the stick. With the same firmware.

To be on the save side, I also re-checked out the dev branch of this project.

I removed the batteries from the dimmer switches and motion sensors, to prevent then to rejoin.

I have a suspicion the issue is with endpoints. The switches and motion sensors are endpoints, not routers. So now I'm first going to test without, I can simulate the button presses to the dimmer switches.

As a sidenote. I first tried flashing the firmware of the zigbee-shepherd project. This always fails at 46% of the first big part. Then I reflashed the firmware koenkk linked here. I could flash this without issue, however, I noticed that on startup of zigbee2mqtt there was a red led flashing on the stick. I have never seen that led before, it has never done anything.

This might also make it possible it's an issue with the alternative flashing method. My real flasher is still on its way from chine, so I use the alternate flashing method. As I understand there are some values set in the flashing software, which might not be part of the flash itself. The zigbee-shepherd firmware has like 10 bytes it writes before big chuncks, maybe those contain the proper values?

In any case, I got it running again, so I can now continue testing.

henriz, 2019-01-15:
Well, it's not the endpoints, but I have confirmed it's clogging.

When running debug output, in a series of commands, eventually you get this:
```

serialport:unixRead Starting read +2m
  serialport:unixRead Finished read 10 bytes +1ms
  serialport:main binding.read finished +2m
  cc-znp { sof: 254,
  cc-znp   len: 5,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'srcRtgInd',
  cc-znp   payload: { dstaddr: 47929, relaycount: 1, relaylist: <Buffer a3 3b> },
  cc-znp   fcs: 159,
  cc-znp   csum: 159 } +2m
  serialport:main _read reading +6ms
  serialport:bindings read +7ms
  serialport:unixRead Starting read +6ms
  cc-znp:AREQ <-- ZDO:srcRtgInd, { dstaddr: 47929, relaycount: 1, relaylist: <Buffer a3 3b> } +2ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +2m

```

So at some instances, it halts reading from the serial port, in this case even for 2 minutes. This is unrelated to the actual command which is being sent.

Koenkk, 2019-01-15:
@henriz nice finding, perhaps it's indeed a zigbee-shepherd issue.

trekker25, 2019-01-15:
nice to find it, but zigbee-shepherd isn't updated anymore i assumed? So it's not easy to fix at all..?

Koenkk, 2019-01-15:
@trekker25 that means we have to update it :)

philhawthorne, 2019-01-16:
@henriz

My zigbee network decided to giveup with this same error overnight, and now I can't get my network to reboot. Hopefully re-flashing the stick will fix it for me.

Best of luck trying to trace this down! You're not the only one experiencing this!

philhawthorne, 2019-01-16:
So just to add some more information to the puzzle. I re-flashed the CC2530, and of course lost all my pairings. However, I did try to wrap the stick in foil, as suggested to try and keep everything paired.

Now I've noticed, that if I restart zigbee2mqtt, I will get the `Cannot read property 'close' of undefined` **unless** I have the stick wrapped in foil on boot. Once booted I can remove the foil, and then start re-pairing my devices.

Very strange

runningman84, 2019-01-16:
When reading your comments I ask myself if the CC2530is the right device for zigbee2mqtt in the long run or if it wouldn't be possible to use the ConBee stick in the future.

henriz, 2019-01-16:
It's not the device which is causing all of this. It's a mixture of firmware and software. 

@philhawthorne how do you flash? with the programmer or using the alternate flashing method?

philhawthorne, 2019-01-16:
I used the programmer to flash it

Jofagi, 2019-01-20:
Hi, my test setup is very small; I've only got a "Philips Hue white A60 Bulb E27" and a CC2531 attached to a Raspi and I'm also experiencing the described queueing problem.

I've got my "production system" Hue Bridge running in parallel for my other bulbs. Could the Hue be interfering with the CC2531 even when no lights are being switched? Does zigbee have "keep alive" packages or something like that?

Can you maybe point me to another project using the CC2351, so I could try whether the problem exists there as well? Something also based on zigbee-shepherd? Or something that's not based on it? This could help in identifying the problematic component.

henriz, 2019-01-21:
It's a tough issue to troubleshoot. I've built in log messages for about every function, but it somehow just halts at times.

From the looks of it, it's the device itself which is getting backed up, so it appears to be a firmware issue, not an issue with zigbee2mqtt or zigbee-shepherd.

Jofagi: does your 'production' setup include dimmer switches and/or motion sensors? Those do strange stuff. Like the motion sensors can be all the sudden bound, without any binding messages and without resetting them. Dimmerswitches often fail to get configured.

Jofagi, 2019-01-21:
@henriz I've only got actors such as bulbs and stripes, but no sensors or switches in my setup.

I'm not convinced, that the dongle itself is just not fit. After all there appear to be quite a lot of people using it. I came to this project from a article of the German c't magazine. I assume they had good experience with it; otherwise they would have at least mentioned the problems it in the article.

And since I'm already having problems with a single bulb, something really weird must be going on...

Another wild guess: Could it be a power issue? I've got the stick attached to a raspi, which also runs some other services and has a HDD attached. I assumed if there was a problem with the power supply, the raspi would simply reboot. But since it doesn't do that; I didn't worry.
However, if the low-power behavior is different, for example by cutting the dongle off in order to save the rest of the system, could this result in such behavior?

Do you also run it on a raspi that's possibly under-powered?

tmarquespt, 2019-01-21:
@Jofagi 

I have mine on a usb port of a QNAP. I'm running Hassio on a Ubuntu vm and can confirm that the issue remains (comming from the raspi). So probably not a power issue to the CC
Since I changed all the bulbs from the zigbee2mqtt network to the Philips hue I have no complaints. The temp sensors are not a good way to check, but all my switches and motion sensors act immediately and I don't have any more clogs in the zigbee2mqtt network that I'm aware of. I only had clogs when using dimming function on bulbs (both Philips and Ikea).

Koenkk, 2019-01-22:
I expect that at least some of the above mentioned clogging issues should be fixed in the dev branch.

henriz, 2019-01-22:
That would be very cool. I just haven't had the time to really dive into it yet. Gonna test it immediatelly.

henriz, 2019-01-22:
It appears now after whatever is clogging has happened, the serial link is disconnected, which is causing it not to function anymore. I waited about 15 minutes but none of the devices would respond. Then when I tried stopping zigbee2mqtt I got this, on an endless loop. The only way to get past it, was to log in on another console and kill the node process.
```

  serialport:main close attempted, but port is not open +0ms
  serialport:main close attempted, but port is not open +0ms
  serialport:main close attempted, but port is not open +0ms
  serialport:main close attempted, but port is not open +0ms
  serialport:main close attempted, but port is not open +1ms
 [..] and so on
```

henriz, 2019-01-23:
It's the same thing. If there's no ACK, it will timeout.

Koenkk, 2019-01-23:
@henriz that doesn't seem to be true in all situations, see the example where I plugged the power plug of the `living_room_standing_lamp`: https://hastebin.com/ecuxefudoc.sql

closed 2019-03-31 by stale[bot]