Koenkk/zigbee2mqtt #844: How to add support for 'undefined' modelID device?

YamiFrankc, 2019-01-12:
I got some light bulbs from Sylvania ([these ones](https://www.amazon.com/gp/product/B07576J38L/ref=ppx_yo_dt_b_asin_title_o00__o00_s00?ie=UTF8&psc=1))

I have tried pairing all of the 4 bulbs that I have many times, but I always get a "Device with modelID 'undefined' is not supported" message in the log, I reseted them many times and also put them just besides the raspberry pi.
There are some other Sylvania bulbs that are supported so I guess this must be really similar?

Anyway, looking around the database.db file, I found an entry with an modelID of "A19 W 10 year", so I tried adding support for that, and it didnt work.

henriz, 2019-01-12:
How did you try to add it?

Most important file here is 
node_modules/zigbee-shepherd-converters/devices.js

For instance, I added support for NXP devices by adding:

    {
        zigbeeModel: ['ZLL-ExtendedColo'],
        model: 'ZLL-ExtendedColo',
        vendor: 'NXP',
        description: 'NXP extended color light',
        supports: generic.light_onoff_brightness_colortemp_colorxy.supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy.fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy.toZigbee,
    },

I suspect you're using the wrong modelID.

henriz, 2019-01-12:
Addendum:

what might help, I think the logging from zigbee2mqtt when a device is unknown, is incomplete.

To find out WHAT device it was talking about, edit
./lib/extension/deviceReceive.js

find the section where it sais:
```
 if (!mappedDevice) {
            logger.warn(`Device with modelID '${device.modelId}' is not supported.`);

```

and add the following line:

` logger.warn(`Device address '${device.ieeeAddr}'`);`

So that the full code reads:

```
 if (!mappedDevice) {
            logger.warn(`Device with modelID '${device.modelId}' is not supported.`);
           logger.warn(`device address '${device.ieeeAddr}'`);
            logger.warn(`Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html`);
            return;
        }

```

Now when you get the 'device not supported' message, it will also tell you the address of the device.

Once you have that, open ./data/database.db, and look for the device address. It might be in there several times.

One of these lines will also contain the modelID you have to use.

ryanbeaton, 2019-01-12:
‘undefined’ modelid has been happening to a few devices recently. Probably a bug introduced but I am on an old version for stability so can not investigate. If you search the issues you will find more.

YamiFrankc, 2019-01-12:
@henriz that worked! For that one bulb. The other bulbs arent working yet.

```
{
    zigbeeModel: ['A19 W 10 year'],
    model: 'A19-Dimmable',
    vendor: 'Sylvania',
    description: 'Dimmable bulb,
    supports: generic.light_onoff_brightness_colortemp_colorxy.supports,
    fromZigbee: generic.light_onoff_brightness_colortemp_colorxy.fromZigbee,
    toZigbee: generic.light_onoff_brightness_colortemp_colorxy.toZigbee,
}
```
Its how I added it, I had the modelID right (A19 W 10 year), but it worked after I deleted it from all of the files you mentioned.

henriz, 2019-01-12:
So you have multiple bulbs of this type, and one of them is now working, but the rest aren't?

Are they reset so they are trying to pair? do you see ANY messages about these lights?

8xiom, 2019-01-12:
> ‘undefined’ modelid has been happening to a few devices recently. Probably a bug introduced but I am on an old version for stability so can not investigate. If you search the issues you will find more.

In fact that happened to me, too (w/Aqara Cube). It turned out that the reason was an incomplete pairing process. After trying 5-6 times again and again it worked for me, no changes necessary. Of course it‘s worth mentioning that the cube is on the list of supported devices.

henriz, 2019-01-12:
I only got it on devices that were not supported. But from those, on all of them.

YamiFrankc, 2019-01-13:
Yes, I have 4 of these bulbs. I have reseted and tried to pair them many times(they blink twice when they enter pairing mode, and blink again once paired). I'm getting the undefined modelID with them too.

Koenkk, 2019-01-27:
@YamiFrankc 

- Can you post your `database.db`
- The Xiaomi temperature sensors report these nulls, nothing to worry about, this is normal.

closed 2019-04-04 by stale[bot]