Koenkk/zigbee2mqtt #845: Two way switching with Xiaomi Single Gang Wall switches

joshdinsdale, 2019-01-12:
Hi Guys

Im wondering if anyone has had any success with two way switching? I have a landing light that has a switch at the top and bottom of the stairs. I'd like to replace both of these with Xiaomi Zigbee switches (i have used these elsewhere in the house) but am sure about the logic behind making this work.

I am using Domoticz as my HA controller, so am thinking i'd probably be best using a variable to set a fake switch each time the Xiaomi switches are toggled.

Does anyone have any suggestions or ways of accomplishing this? 

Thanks

Josh

henriz, 2019-01-12:
You could use nodered to accomplish this. You can set flow variables in node-red, which will retain the status.

Bojan023, 2019-01-14:
Somewhat related: the **wireless** Xiaomi switches work as a pulse switch; when a button is pressed, a Zigbee message is broadcasted with the event (for instance: left button clicked twice). In HomeAssistant (similair to Domoticz) I've automation script running for consequences of these events (turn on light x).

What I assume is that the wired version of the wall switch work the same way, so there is no direct connection between the buttons and the relais. So if a button is pressed, only a event is broadcasted. The relais is switched by receiving certain commands over Zigbee (e.g. switch line 1). Normally this command is send from the Xiaomi hub but now you need to automate this from Domoticz.

Not sure is this helps you, I just give you my 2 cents. Good luck!

joshdinsdale, 2019-01-18:
Hey bojan023 thanks for the reply, however I don't think that is the same with the wired version as it switches the light even if not paired with a ZigBee controller.
I think Node Red is the answer, but the logic to do it is escaping me at the moment. If anyone has any suggestions as to how to get started that would be great! :S

closed 2019-03-26 by stale[bot]