Koenkk/zigbee2mqtt #847: Osram Classic Light 60

Luftloch80, 2019-01-12:
During pairing i get a lot „Message without device“ errors but then it Says paired 
I can also sent payload for on and off but no brightness
Also no payload from device Is Shown when i Turn the Lamp,on or off

Koenkk, 2019-01-13:
This message without device message can be ignored, it will not be shown in the next version of zigbee2mqtt.

Luftloch80, 2019-01-13:
Okay. But why do i not see any payload from the device 
When i Turn Off the Osram Plug i See payload but Not with the lightbulb

Luftloch80, 2019-01-13:
zigbee2mqtt:info 2019-1-13 18:33:57 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-13 18:33:57 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-13 18:33:57 Coordinator firmware version: '20181024'
  zigbee2mqtt:info 2019-1-13 18:33:57 Currently 1 devices are joined:
  zigbee2mqtt:info 2019-1-13 18:33:57 0x8418260000073d64 (0x8418260000073d64): AC03641 - OSRAM LIGHTIFY LED Classic A60 clear (Router)
  zigbee2mqtt:warn 2019-1-13 18:33:57 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-1-13 18:33:57 Allowing new devices to join.
  zigbee2mqtt:warn 2019-1-13 18:33:57 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-1-13 18:33:57 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-1-13 18:33:57 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2019-1-13 18:33:58 zigbee-shepherd ready
  zigbee2mqtt:info 2019-1-13 18:33:58 Connected to MQTT server
  zigbee2mqtt:info 2019-1-13 18:33:58 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:warn 2019-1-13 18:34:05 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:06 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:06 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:06 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:07 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:07 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:07 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:07 Message without device!
  zigbee2mqtt:warn 2019-1-13 18:34:08 Message without device!
  zigbee2mqtt:info 2019-1-13 18:34:08 Device incoming...
  zigbee2mqtt:info 2019-1-13 18:34:08 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

After Message it stops. 
Everytime new entry in dB 
I can sent payload but do Not Receive any

Koenkk, 2019-01-13:
Can you post the log while sending a payload?

Koenkk, 2019-01-14:
What do you send to turn it on? There is a strange value in the payload: `030`

Koenkk, 2019-01-17:
Why are you publishing `zigbee2mqtt/0x8418260000073d64/state` `off` then? You should publish: `zigbee2mqtt/0x8418260000073d64/state`

```js
{
"brightness": 30
}
```

snoopyCode, 2019-01-23:
Is there really `zigbee2mqtt/0x8418260000073d64/state`? When you look at `light_brightness` in [https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js](url) only `/set` and `/get` is available. 

I guess with `zigbee2mqtt/0x8418260000073d64/set` and the payload
```
{
"brightness": 30
}
```
should work.

Koenkk, 2019-01-24:
@snoopyCode you are right, `zigbee2mqtt/0x8418260000073d64/set` should be used.

closed 2019-04-01 by stale[bot]