Koenkk/zigbee2mqtt #848: eCozy caching issue

Zagitta, 2019-01-13:
Hey so I'm trying out the experimental eCozy support using the dev branch and working on integrating it with homeassistant.

One of the issues I'm running into currently is that despite forcing `advanced.cache_state` to true in `configuration.yaml` it seems that the mqtt messages sent by zigbee2mqtt for the eCozy thermostat contains null for everything except the changed key as can be seen in the following gist of the log file: https://gist.github.com/Zagitta/b69883e4ddcf62c2bf7c3324be3b77c1

As seen from the log file i send a mqtt get message of `system_mode`:
```
zigbee2mqtt:debug 13/01/2019, 00.43.26 Received MQTT message on 'zigbee2mqtt/0x70b3d5de00000df5/get' with data '{"system_mode": ""}'
zigbee2mqtt:info 13/01/2019, 00.43.26 Zigbee publish to device '0x70b3d5de00000df5', hvacThermostat - read - [{"attrId":28}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:debug 13/01/2019, 00.43.27 Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"systemMode":1}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 00.43.27 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":null,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":34,"last_seen":"2019-01-13T00:43:27.177Z","voltage":0.27,"system_mode":1}
```
At this point i send a mqtt get message of `occupied_heating_setpoint` at which point it promptly has forgotten about `system_mode`:
```
zigbee2mqtt:debug 13/01/2019, 00.43.30 Received MQTT message on 'zigbee2mqtt/0x70b3d5de00000df5/get' with data '{"occupied_heating_setpoint": ""}'
zigbee2mqtt:info 13/01/2019, 00.43.30 Zigbee publish to device '0x70b3d5de00000df5', hvacThermostat - read - [{"attrId":18}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:debug 13/01/2019, 00.43.31 Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"occupiedHeatingSetpoint":2000}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 00.43.31 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":null,"local_temperature_calibration":null,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":34,"last_seen":"2019-01-13T00:43:31.198Z","voltage":0.27}'
```
It can also be seen that when waiting or the next `local_temperature` update from the device it has again forgotten `occupied_heating_setpoint`:
```
zigbee2mqtt:debug 13/01/2019, 00.43.42 Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":2616}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 00.43.42 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":26.16,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":34,"last_seen":"2019-01-13T00:43:42.135Z","voltage":0.27}'
```
I've done some further digging by inserting some rudimentary logging around the `objectAssignDeep.noMutate` call in `controller.js` and it seems some weird stuff is happening:
```
zigbee2mqtt:info 13/01/2019, 01.38.09 messagePayload before: {"local_temperature":null,"local_temperature_calibration":null,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"last_seen":"2019-01-13T01:38:09.565Z"}
zigbee2mqtt:info 13/01/2019, 01.38.09 state before: {"local_temperature":null,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":39,"last_seen":"2019-01-13T01:38:06.550Z","system_mode":1}
zigbee2mqtt:info 13/01/2019, 01.38.09 messagePayload after: {"local_temperature":null,"local_temperature_calibration":null,"occupied_heating_setpoint":20,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":39,"last_seen":"2019-01-13T01:38:09.565Z"}
```

Although not shown in the log it doesn't seem to be an issue with my xiaomi aqara temperature/humidity sensors so what on earth is happening?

Chrischi-, 2019-01-13:
i will make a PR.

i had the same with my bitron thermostat converter.

Zagitta, 2019-01-13:
Thanks  @Chrischi- I can confirm your fix works! 🎉🎉 
What's the consensus on writing a value to a device where it does not automatically send back the updated value? For example setting `system_mode` does not result in an updated value from my eCozy thermostat. I can see this is handled for transitions in `devicePublish.js`, should there perhaps be a way to enable this behavior in `zigbee-shepherd-converters` for values which needs it?

Last but not least what about the `system_mode` value? Currently it's published simply as the raw integer value rather than a translated string such as "Off" or "Auto" etc.
Should I make a PR to change this or is it intentionally the raw value? I'm asking because if we're not translating the value it'll have to be done in home assistant which is a bit of a pain.

Chrischi-, 2019-01-13:
@Zagitta 
> What's the consensus on writing a value to a device where it does not automatically send back the updated value? For example setting `system_mode` does not result in an updated value from my eCozy [...]

in most cases the endpoint gives a response and the converter is also prepared to process it. 
Do you have a log of (while) setting the mode?

Zagitta, 2019-01-13:
Here's a filtered log of first writing the `system_mode` value, waiting two minutes and then getting it:

```
zigbee2mqtt:debug 13/01/2019, 20.47.19 Received MQTT message on 'zigbee2mqtt/0x70b3d5de00000df5/set' with data '{"system_mode": 0}'
zigbee2mqtt:info 13/01/2019, 20.47.19 Zigbee publish to device '0x70b3d5de00000df5', hvacThermostat - write - [{"attrId":28,"dataType":48,"attrData":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:debug 13/01/2019, 20.47.20 Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":2463}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 20.47.20 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":24.63,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":2,"last_seen":"2019-01-13T20:47:20.944Z","system_mode":1,"remote_sensing":0}'
zigbee2mqtt:debug 13/01/2019, 20.47.50 Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":2463}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 20.47.50 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":24.63,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":0,"last_seen":"2019-01-13T20:47:50.946Z","system_mode":1,"remote_sensing":0}'
zigbee2mqtt:debug 13/01/2019, 20.49.48 Received MQTT message on 'zigbee2mqtt/0x70b3d5de00000df5/get' with data '{"system_mode": ""}'
zigbee2mqtt:info 13/01/2019, 20.49.48 Zigbee publish to device '0x70b3d5de00000df5', hvacThermostat - read - [{"attrId":28}] - {"manufSpec":0,"disDefaultRsp":0} - null
zigbee2mqtt:debug 13/01/2019, 20.49.49 Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"systemMode":0}}' of device 'Thermostat' (0x70b3d5de00000df5)
zigbee2mqtt:info 13/01/2019, 20.49.49 MQTT publish: topic 'zigbee2mqtt/0x70b3d5de00000df5', payload '{"local_temperature":24.72,"local_temperature_calibration":null,"occupied_heating_setpoint":null,"unoccupied_heating_setpoint":null,"setpoint_change_amount":null,"linkquality":0,"last_seen":"2019-01-13T20:49:49.097Z","system_mode":0,"remote_sensing":0}'
```

I'll try your suggestion now.

Zagitta, 2019-01-13:
@Koenkk What's your opinion on supporting a mqtt topic like: `zigbee2mqtt/[deviceID]/set/[parameter]`? I'm asking because the homeassistant [MQTT HVAC component ](https://www.home-assistant.io/components/climate.mqtt/) does not support a data template for the various command topics. Another solution would be to extend homeassistant with data templates for all the command topics in MQTT HVAC component however I thought I'd ask your opinion first since I think it's better to adapt zigbee2mqtt than homeassistant.

Chrischi-, 2019-01-13:
my Bitron thermostat gives me a value. This doesn't seem to work with the eCozy. That's why I can't test it any further for you.

@sebastianheierhoff 
can you take a look at it?

sebastianheierhoff, 2019-01-14:
I've just tested getting and setting "system_mode" of my ecozy via MQTT (running latest version of zigbee2mqtt/dev and zigbee-shepherd-converters) and got no errors.
```
{"system_mode":""}
{"system_mode":1,"last_seen":"2019-01-14T17:18:30.232Z"}
```
Adding `(cb) => device.report('hvacThermostat', 'systemMode', 1, 0, 0, cb),` should not be possible as systemMode is not reportable. 

Are you maybe polling an attribute that hasn't changed, see [link](https://github.com/Koenkk/zigbee2mqtt/issues/518)?

Zagitta, 2019-01-14:
I'm not polling anything, I'd just like a MQTT response that the value has changed successfully after writing it. This seems to happen for the `occupied_heating_setpoint` value but not for `system_mode` which I guess boils down to what you're saying about it not being reportable.
This is an issue with regards to the homeassistant integration as it requires a mqtt message to confirm the value has changed after a write.

@sebastianheierhoff a completely different question: Have you experienced that the eCozy thermostat  seems to turn down after a while?
I'm seeing the following: [image]
Where if I set the `occupied_heating_setpoint` to some value it'll keep it around that temperature for a while but then eventually do as in the picture and drop down to a lower temperature.
I can't quite figure it out which is why I started investigating `system_mode` in the first place as I thought it might be turning off itself or going into the sleep mode but it doesn't seem like it.
I've tried investigating `occupancy` value but trying to set/get `unoccupied_heating_setpoint` didn't do anything so I'm guessing neither does `occupancy`.

sebastianheierhoff, 2019-01-15:
Okay, I didn't understand this. I can confirm that my ecozy acts the same.

Regarding the problem of not keeping the temperature: have you tried setting the system_mode to 4?

Chrischi-, 2019-01-15:
After reading this i'am also sure, that Homeassistant or whatever backend you use has to implement the intelligence of polling a "unreportable attribute" after setting a value.
The bahavior of zigbee2mqtt is working as it should

closed 2019-03-23 by stale[bot]