Koenkk/zigbee2mqtt #849: update.sh get error

itajackass, 2019-01-13:
when I run sudo ./update.sh I get this error message.

```
Stopping zigbee2mqtt...
Creating backup of configuration...
Updating...
Aggiornamento di b0d3c2f..9380bbc
error: Your local changes to the following files would be overwritten by merge:
        .eslintignore
        .github/ISSUE_TEMPLATE.md
        .github/stale.yml
        .travis.yml
        README.md
        docker/Dockerfile.amd64
        docker/Dockerfile.arm32v6
        docker/Dockerfile.arm64v8
        lib/controller.js
        lib/extension/devicePublish.js
        lib/extension/networkMap.js
        lib/extension/routerPollXiaomi.js
        lib/extension/softReset.js
        lib/mqtt.js
        lib/state.js
        lib/util/settings.js
        lib/zigbee.js
        package.json
        test/data.test.js
        test/devicePublish.test.js
Please commit your changes or stash them before you merge.
error: The following untracked working tree files would be overwritten by merge:
        docgen/docgen.js
        docgen/integrating-with-homeassistant.js
        docgen/supported-devices.js
        docs/_config.yml
        docs/configuration/configuration.md
        docs/configuration/device_specific_configuration.md
        docs/getting_started/flashing_the_cc2531.md
        docs/getting_started/pairing_devices.md
        docs/getting_started/running_zigbee2mqtt.md
        docs/getting_started/what_do_i_need.md
        docs/how_tos/how_to_debug.md
        docs/how_tos/how_to_support_new_devices.md
        docs/how_tos/how_to_support_new_devices_on_hassio.md
        docs/images/cc2530.jpg
        docs/images/cc2530_cc2591.jpg
        docs/images/cc2530_rfx2401.png
        docs/images/cc2531.jpg
        docs/images/cc_debugger.jpg
        docs/images/connected.jpg
        docs/images/devices/22670.jpg
        docs/images/devices/3210-L.jpg
        docs/images/devices/3216331P5.jpg
        docs/images/devices/324131092621.jpg
        docs/images/devices/3261030P7.jpg
        docs/images/devices/3320-L.jpg
        docs/images/devices/3326-L.jpg
        docs/images/devices/4052899926110.jpg
        docs/images/devices/4052899926158.jpg
        docs/images/devices/4058075036147.jpg
        docs/images/devices/4058075036185.jpg
        docs/images/devices/4058075816718.jpg
        docs/images/devices/4058075816794.jpg
        docs/images/devices/421786.jpg
        docs/images/devices/4256251-RZHAC.jpg
        docs/images/devices/433714.jpg
        docs/images/devices/45852GE.jpg
        docs/images/devices/45857GE.jpg
        docs/images/devices/50045.jpg
        docs/images/devices/50049.jpg
        docs/images/devices/53170161.jpg
[... 136 lines cut ...]
        docs/information/alternative_flashing_methods.md
        docs/information/cc_sniffer_devices.md
        docs/information/docker.md
        docs/information/mqtt_topics_and_message_structure.md
        docs/information/supported_devices.md
        docs/information/zigbee_network.md
        docs/integration/home_assistant.md
        images/architecture.png
        images/logo.png
        lib/extension/bridgeConfig.js
        lib/extension/deviceConfigure.js
        lib/extension/deviceReceive.js
        lib/extension/extensionTemplate.js
        lib/extension/homeassistant.js
        lib/extension/markOnlineXiaomi.js
        lib/zapp/cie.js
        test/controller.test.js
        test/deviceReceive.test.js
        test/docgen.test.js
        test/homeassistant.test.js
        test/images.test.js
        test/utils.js
Please move or remove them before you merge.
Aborting
```

Koenkk, 2019-01-13:
Before https://github.com/Koenkk/zigbee2mqtt/blob/master/update.sh#L10 can you add `git reset --hard HEAD` and try again?

itajackass, 2019-01-14:
same result: Aborting

New log:

Stopping zigbee2mqtt...
Creating backup of configuration...
Updating...
HEAD ora si trova a b0d3c2f 0.2.0
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 45 (delta 21), reused 43 (delta 19), pack-reused 0
Unpacking objects: 100% (45/45), done.
Da https://github.com/Koenkk/zigbee2mqtt
   dccdc5a..9605e72  dev        -> origin/dev
Aggiornamento di b0d3c2f..9380bbc
error: The following untracked working tree files would be overwritten by merge:
        docgen/docgen.js
        docgen/integrating-with-homeassistant.js
        docgen/supported-devices.js
        docs/_config.yml
        docs/configuration/configuration.md
        docs/configuration/device_specific_configuration.md
        docs/getting_started/flashing_the_cc2531.md
        docs/getting_started/pairing_devices.md
        docs/getting_started/running_zigbee2mqtt.md
        docs/getting_started/what_do_i_need.md
        docs/how_tos/how_to_debug.md
        docs/how_tos/how_to_support_new_devices.md
        docs/how_tos/how_to_support_new_devices_on_hassio.md
        docs/images/cc2530.jpg
        docs/images/cc2530_cc2591.jpg
        docs/images/cc2530_rfx2401.png
        docs/images/cc2531.jpg
        docs/images/cc_debugger.jpg
        docs/images/connected.jpg
        docs/images/devices/22670.jpg
        docs/images/devices/3210-L.jpg
        docs/images/devices/3216331P5.jpg
        docs/images/devices/324131092621.jpg
        docs/images/devices/3261030P7.jpg
        docs/images/devices/3320-L.jpg
        docs/images/devices/3326-L.jpg
        docs/images/devices/4052899926110.jpg
        docs/images/devices/4052899926158.jpg
        docs/images/devices/4058075036147.jpg
        docs/images/devices/4058075036185.jpg
        docs/images/devices/4058075816718.jpg
        docs/images/devices/4058075816794.jpg
        docs/images/devices/421786.jpg
        docs/images/devices/4256251-RZHAC.jpg
        docs/images/devices/433714.jpg
        docs/images/devices/45852GE.jpg
        docs/images/devices/45857GE.jpg
        docs/images/devices/50045.jpg
        docs/images/devices/50049.jpg
        docs/images/devices/53170161.jpg
        docs/images/devices/7146060PH.jpg
        docs/images/devices/7199960PH.jpg
        docs/images/devices/72922-A.jpg
        docs/images/devices/7299355PH.jpg
        docs/images/devices/7299760PH.jpg
        docs/images/devices/73693.jpg
        docs/images/devices/73740.jpg
        docs/images/devices/73742.jpg
        docs/images/devices/74282.jpg
        docs/images/devices/74283.jpg
        docs/images/devices/74696.jpg
        docs/images/devices/8718696449691.jpg
        docs/images/devices/8718696485880.jpg
[... 122 lines cut ...]
        docs/information/3d_cases.md
        docs/information/alternative_flashing_methods.md
        docs/information/cc_sniffer_devices.md
        docs/information/docker.md
        docs/information/mqtt_topics_and_message_structure.md
        docs/information/supported_devices.md
        docs/information/zigbee_network.md
        docs/integration/home_assistant.md
        images/architecture.png
        images/logo.png
        lib/extension/bridgeConfig.js
        lib/extension/deviceConfigure.js
        lib/extension/deviceReceive.js
        lib/extension/extensionTemplate.js
        lib/extension/homeassistant.js
        lib/extension/markOnlineXiaomi.js
        lib/zapp/cie.js
        test/controller.test.js
        test/deviceReceive.test.js
        test/docgen.test.js
        test/homeassistant.test.js
        test/images.test.js
        test/utils.js
Please move or remove them before you merge.
Aborting

Koenkk, 2019-01-15:
I would recommend doing a fresh checkout of zigbee2mqtt and copy your `data` directory.

closed 2019-03-23 by stale[bot]