Koenkk/zigbee2mqtt #852: Adding new device

P2mb2, 2019-01-13:
HI, 

i'm trying to add a device on Hassio version. I added the   "zigbee_shepherd_devices": true  parameter on config and created a devices.js file with the content below:

```
core-ssh:/share/zigbee2mqtt# cat devices.js
{
    zigbeeModel: ['LST001'],
    model: '915004463301',
    vendor: 'Philips',
    description: 'Hue ambiance LightStrip',
    supports: 'll see ' ,
    fromZigbee: [], // We will add this later
    toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee, // Should be empty, unless device can be controlled (e.g. lights, switches)
},
```
When starting, I get this error:

```
  zigbee2mqtt:info 2019-1-13 18:13:46 Logging to directory: '/share/zigbee2mqtt/log/2019-01-13.18-13-45'
/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/devices.js:3
    model: '915004463301',
         ^
SyntaxError: Unexpected token :
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:617:28)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/index.js:3:17)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm
 
ERR!
 A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-13T18_13_47_826Z-debug.log
```

Can you please help ? :) For sure i'm missing something.
Note: I know this particular equipment is already on the dev branch, but i'm using the master :(  

Many Thanks.

Koenkk, 2019-01-13:
@P2mb2 why not switch to the dev branch? (-edge for hassio)

P2mb2, 2019-01-13:
When i first installed i guess i choose stability over faster updates. I'm now using a lot of automations and need reliability on my instalation.
Is there any way to change to dev branch without the need to add all devices again? I currently have 20+ devices. :))

pwrangles, 2019-01-13:
Im really struggling to get things running with these "Unsupported" devices. would be good to understand how to do it

Koenkk, 2019-01-14:
@P2mb2 you don't have to readd devices when changing to the dev branch

@pwrangles somebody else is already working on the salus thermostat.

pwrangles, 2019-01-14:
Hi Kornkk, thanks for the info. Could you share the link to this please? I’m using the TRV’s and not the Thermostat.

Koenkk, 2019-01-14:
@pwrangles I thought it was the same as the issue where you commented on.

pwrangles, 2019-01-14:
Yes, Sorry about that. I found it shortly after asking for the link. Thanks for your help!

Koenkk, 2019-01-14:
@P2mb2 it seems that there is a syntax error in his `devices.js` (which I cannot see from the OP).

pwrangles, 2019-01-14:
I get the same issue, This is my entire Devices.js file:

```
{
    zigbeeModel: ['SAL6EM1'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TRV10RFM', // Vendor model number, look on the device for a model number
    vendor: 'Salus', // Vendor of the device (only used for documentation and startup logging)
    description: 'Wireless Radiator Controller', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'Temperature', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
}
```

Koenkk, 2019-01-14:
could you post your whole log?

pwrangles, 2019-01-14:
Sure
```
  zigbee2mqtt:info 2019-1-14 16:32:00 Logging to directory: '/share/zigbee2mqtt/log/2019-01-14.16-32-00'
/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/devices.js:3
    model: 'TRV10RFM', // Vendor model number, look on the device for a model number
         ^
SyntaxError: Unexpected token :
    at createScript (vm.js:80:10)
    at Object.runInThisContext (vm.js:139:10)
    at Module._compile (module.js:617:28)
    at Object.Module._extensions..js (module.js:664:10)
    at Module.load (module.js:566:32)
    at tryModuleLoad (module.js:506:12)
    at Function.Module._load (module.js:498:3)
    at Module.require (module.js:597:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/zigbee2mqtt-1.0.1/node_modules/zigbee-shepherd-converters/index.js:3:17)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-14T16_32_01_316Z-debug.log
2019-01-14T16:32:02: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-01-14T16:32:02: PM2 log: App [npm:0] starting in -fork mode-
2019-01-14T16:32:02: PM2 log: App [npm:0] online
> zigbee2mqtt@1.0.1 start /zigbee2mqtt-1.0.1
> node index.js
```

Koenkk, 2019-01-14:
I don't see anything wrong here, @danielwelch could this be an issue of the addon?

danielwelch, 2019-01-14:
Addressing @P2mb2 and @pwrangles since the other issue seems unrelated. 

@P2mb2 is that your entire devices.js file, or just the part you added? @pwrangles the add-on does not “merge” in your new device, it just overwrites zigbee-shepherd’s devices.js file, so the intended workflow is to edit devices.js. You’ll need to start with a working devices.js and make your custom edits from there.

And just to clarify/reiterate, the edge add-on tracks dev branch, and the standard (non-edge) version tracks releases.

pwrangles, 2019-01-14:
@danielwelch Thanks for the response! That makes complete sense!! Ive done that and its started successfully. Complete noob here! Can you tell! 

Thanks for your help!

P2mb2, 2019-01-14:
Thanks @danielwelch 

thats my problem too... I Thought devices.js only had one structure and it would add to the original one,
Thanks for the explanation... meanwhile i've installed dev version... one more to help testing new versions ;)

danielwelch, 2019-01-14:
Happy to help!

pwrangles, 2019-01-14:
After adding my device to the devices..js I get the below:
```

  zigbee2mqtt:info 2019-1-14 17:14:00 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
/zigbee2mqtt-1.0.1/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');
        ^
Error: Unrecognized cluster
    at new FuncPayload (/zigbee2mqtt-1.0.1/node_modules/zcl-packet/lib/functional.js:27:15)
    at /zigbee2mqtt-1.0.1/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/zigbee2mqtt-1.0.1/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at Dissolve.emit (events.js:211:7)
    at Dissolve.<anonymous> (/zigbee2mqtt-1.0.1/node_modules/dissolve-chunks/index.js:73:29)
    at emitNone (events.js:106:13)
    at Dissolve.emit (events.js:208:7)
    at emitReadable_ (/zigbee2mqtt-1.0.1/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
npm ERR! 
code ELIFECYCLE
npm ERR!
 errno 1
npm
 ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm 
ERR!     /root/.npm/_logs/2019-01-14T17_14_24_336Z-debug.log
2019-01-14T17:14:25: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-01-14T17:14:25: PM2 log: App [npm:0] starting in -fork mode-
2019-01-14T17:14:25: PM2 log: App [npm:0] online
> zigbee2mqtt@1.0.1 start /zigbee2mqtt-1.0.1
> node index.js
```

Koenkk, 2019-01-14:
@pwrangles that is a zigbee2mqtt issue. Your device is using a non-standard cluster which should be added. 

Can you add to: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L344 `console.log(frameCntl, manufCode, seqNum, cmd, zclData, cId);`

pwrangles, 2019-01-16:
Hi @Koenkk, Sorry, I don't understand where that is or where to add it on the Hass.IO add-on?

pwrangles, 2019-01-16:
I have installed the whole system on a dedicated Pi and added the line requested but I don't get any further. Im struggling to even get the device to pair.

Koenkk, 2019-01-16:
@pwrangles that's because the devices are joining your other network (the one you already had). You should get out of range of your current network.

pwrangles, 2019-01-16:
My current network is turned off.

Koenkk, 2019-01-16:
On hassio you should be able to find the `af.js` under `/app/node_modules/zigbee-shepherd/lib/components/af.js` or `node_modules/zigbee-shepherd/lib/components/af.js`

deennoo, 2019-01-25:
i have the same salus trv full error is : 

```
 cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 64512, srcaddr: 64158, srcendpoint: 6, dstendpoint: 1, wasbroadcast: 0, linkquality: 115, securityuse: 0, timestamp: 6218074, transseqnumber: 0, len: 81, data: <Buffer 0d 78 10 62 36 03 49 44 33 20 6f 74 61 45 72 72 3a 30 20 65 72 72 43 6f 64 65 3a 30 30 30 30 30 30 30 30 20 6d 63 75 45 72 72 3a 30 34 43 30 30 30 30 ... > } +3ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');
```
Once again a wrong clusterid : 64512

Koenkk, 2019-01-26:
@deennoo similar to #917. Please try with zcl-id and zcl-packet form https://github.com/Koenkk/zcl-id/commits/salus and https://github.com/Koenkk/zcl-packet/tree/salus

closed 2019-04-03 by stale[bot]