Koenkk/zigbee2mqtt #853: It breaks everything during pairing

marcotini, 2019-01-13:
When I try to pair a double switch from Xiaomi, everything stop working. If I try to restart Zigbee2mqtt it give me the error "try to fix 60 seconds...". I solve rebooting the entire system.

How can be possible? I've just 13 end devices and they should be 15, right?

Chrischi-, 2019-01-13:
please show us a log to help you better.

marcotini, 2019-01-13:
2019-1-13 20:11:31 - info: Logging to directory: '/app/data/log/2019-01-13.20-11-31'
2019-1-13 20:11:31 - info: Starting zigbee2mqtt version 1.0.1 (commit #84593a5)
2019-1-13 20:11:31 - info: Starting zigbee-shepherd
2019-1-13 20:11:34 - info: Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)

Chrischi-, 2019-01-13:
the log while you pairing the device.

the above error will normaly fix after 60 seconds (einfach laufen lassen).
If not unplug and replug your zigbee stick.

of course you could also reflash the stick and pair all devices again.

marcotini, 2019-01-13:
There are no log, the last entry is a change value from a different device. It’s just freeze everything when it should pair the double switch.

Re-pair all the devices is a pain, I prefer not to do that. Btw there is a way to export all the configuration to a different machine? Just copy the configuration.yaml and database.db files is enough?

marcotini, 2019-01-14:
Ok now, without doing nothing (I've just added the option for the debug) it's work and it's let me add two other devices (double switch included) with a total of 15 devices. I really don't know why but I'm not complain lol.

I can add the 16th device without other routers?

Btw the log now is:
```
  zigbee2mqtt:debug 2019-1-14 20:40:06 Saving state to file /app/data/state.json
  zigbee2mqtt:debug 2019-1-14 20:40:07 Received zigbee message of type 'devInterview' with data '"0x00..."'
  zigbee2mqtt:info 2019-1-14 20:40:07 Connecting with device...
  zigbee2mqtt:info 2019-1-14 20:40:07 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 2019-1-14 20:40:07 Message without device!

```

closed 2019-03-22 by stale[bot]