Koenkk/zigbee2mqtt #856: Dimming transition with tradfri lights?

ericmiller777, 2019-01-14:
I have been trying to get a tradfri bulb to transition when I dim it, it transitions on and off, but not to different levels, sorry if this is something easy.

Chrischi-, 2019-01-14:
what is your `zigbee2mqtt/[DEVICE_ID]/set` command?

ericmiller777, 2019-01-14:
> what is your `zigbee2mqtt/[DEVICE_ID]/set` command?

Pretty new to this, using hass.io, in /share/zigbee2mqtt I have four files:

configuration.yaml
database.db
log
state.json

in configuration i have:

'0x90fd9ffffe6e8ec3':
    friendly_name: '0x90fd9ffffe6e8ec3'
    retain: false

in database.db I have:

{"id":2,"type":"Router","ieeeAddr":"0x90fd9ffffe6e8ec3","nwkAddr":52281,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E26 WS opal 980lm","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":544,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"MvuaArCONf37ke2n"}

there are two listings for it.

Chrischi-, 2019-01-14:
sorry! i dont know how to set the transition in hass.io. I'am not using it.

It should send a payload with transition to your MQTT topic. 

maybe @Koenkk can help you

ericmiller777, 2019-01-14:
> sorry! i dont know how to set the transition in hass.io. I'am not using it.
> 
> It should send a payload with transition to your MQTT topic.
> 
> maybe @Koenkk can help you

Thank you been trying to figure this out for over a week >.< the addon github issue tracker sent me here. I can set "transition from the home assistant states area and send the command that way and it works fine, but I don't know how to make "transition" permanent.

Chrischi-, 2019-01-14:
it's possible this is not implemented atm but @Koenkk knows more about hass.io.

i only send the payload via node-red to mqtt

ericmiller777, 2019-01-15:
> AFAIK the "transition" can not be send when using the on/off/brightness buttons of a light via the home assistant UI. This is a limitation of home assistant.

So what would be the best way to fix this, if I install node red in my hassio configuration will that help?

Koenkk, 2019-01-15:
I don't have any experience with node red.

henriz, 2019-01-16:
the 'transtime' property doesn't work properly. I have experimented with this as well, but even if you get it into the mqtt message, it doesn't get translated into the actual command. I was planning to debug and fix it, until I ran in into a bigger issue.

CypherMK, 2019-01-17:
Transition through node red seems to work if you send "transition": put_transition_time_in_seconds_here.
But I can't get it to work reliably with 3 lights at the same time. Sometimes it works, sometimes it doesn't. Waiting for the group function, maybe that makes a difference.

Bojan023, 2019-01-19:
> 
> 
> AFAIK the "transition" can not be send when using the on/off/brightness buttons of a light via the home assistant UI. This is a limitation of home assistant.

Do the IKEA light bulbs actually support this transition? When broadcasting a transition manually from HA, Zigbee2MQTT confirms the broadcast including the payload but it "has no converter"?
[image]
[image]

.

CypherMK, 2019-01-20:
When I add a different delay to each light in my group of lights, transition seems to work reliably.

ericmiller777, 2019-01-24:
I'll try seperateing it.

closed 2019-04-01 by stale[bot]

Zabov79, 2019-04-05:
So is it possible to have a transition between different levels of a single light bulb? And how?

ericmiller777, 2019-04-22:
Yep any updates on this, the only thing ive found is I think node red works, but i dont really wanna download node red just for that....

ericmiller777, 2019-05-23:
@Koenkk Sorry to bother you, just wondering, will the 1.5 update fix this? I saw that the upcoming breaking change would allow more options in the hass.io addon?

Koenkk, 2019-05-23:
@ericmiller777 it can be done with `light.turn_on` via `transition` (https://www.home-assistant.io/components/light/)

Koenkk, 2019-05-23:
@ericmiller777 you should be able to do this via:

```yaml
devices:
  '0x84182600000e9afd':
    friendly_name: '0x84182600000e9afd'
    retain: false
    homeassistant:
      light:
        transition: 2
```

Note that you have to restart home assistant afterwards.

StimpyMGS, 2020-01-25:
@Koenkk Sorry to open this again, but in your last comment you suggest to add the transition value to devices.yaml. However, this breaks the entity in HA. When I remove the lines it works again. After every change in devices.yaml I restart the z2m docker and the HA server. 
I too would like to add a transition when turning a light on/off by default.

Koenkk, 2020-01-29:
@StimpyMGS implemented, see #2850