Koenkk/zigbee2mqtt #857: Hue bulb full factory reset command / Wireshark Dump

Chrischi-, 2019-01-14:
Does anyone have or can provide a wireshark capture while resetting a hue bulb?

I hope we can somehow recreate the reset command for people who don't have a hue-switch.
If someone has more information if this is possible at all, please let us know here.

An alternative would be to modify the firmware for the CC2531 so that it does this. 
After a reset you could flash the Coordinator firmware again.

Just a few thoughts on that.

henriz, 2019-01-16:
I am waiting for a second CC2531 to arrive, which it should have already. When I get that, and no-one has done this yet, I'll capture a reset for you.

kiall, 2019-01-18:
This isn't exactly what you asked for - but may be useful anyway... I've been using this to factory reset bulbs using an older ZigBee stick I have laying around.

https://github.com/vanviegen/hue-thief

Chrischi-, 2019-01-18:
Nice! Looks like it used the normal `Reset to Factory New Request Command Frame`

closed 2019-03-22 by Chrischi-