Koenkk/zigbee2mqtt #858: Docs: Xiaomi wireless wall switch has new events/click types

sapkra, 2019-01-14:
I ordered the Xiaomi Aqara double key wireless wall switches WXKG02LM an noticed that some of them (newer models) are also supporting the following click types:

- left_double
- left_long
- right_double
- right_long
- both_double
- both_long

The model ID is the same only the packaging is different. The reaction time of the newer models is a lot slower because they have to wait for a possible double click.
I think we have to add these information to the [documentation](https://koenkk.github.io/zigbee2mqtt/information/supported_devices.html).

Koenkk, 2019-01-14:
Is this already supported by zigbee2mqtt? (I guess this requires some changes)?

sapkra, 2019-01-14:
Yes it's already supported.

```
payload '{"battery":100,"voltage":3125,"linkquality":92,"click":"left_double"}'
payload '{"battery":100,"voltage":3125,"linkquality":94,"click":"right_double"}'
payload '{"battery":100,"voltage":3125,"linkquality":92,"click":"both_double"}'
payload '{"battery":100,"voltage":3125,"linkquality":94,"click":"left_long"}'
payload '{"battery":100,"voltage":3125,"linkquality":94,"click":"right_long"}'
payload '{"battery":100,"voltage":3125,"linkquality":92,"click":"both_long"}'
```

Koenkk, 2019-01-14:
Great, could you make a pr to add them here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L160 ? They will automatically endup in the documentation.

sapkra, 2019-01-14:
Do you have any suggestions how to express it? We have one device with different versions!

My device from AliExpress only supports only supports the current documented events. My new device from GearBest also supports the new events. I think when I just add the it will end in some confusions.

The GearBest model is `lumi.remote.b286acn01\u0000\u0000\u0000` which is supported because it was added to the older device `lumi.sensor_86sw2\u0000Un`. These events are supported because `WXKG02LM_click_multistate` was also added. https://github.com/Koenkk/zigbee-shepherd-converters/commit/d4059ccfeccec32eb351400abf581aed49f049d7

In my opinion we should split them in two different devices. Are there any problems by doing it?

Koenkk, 2019-01-14:
Splitting is not possible as we would have duplicated modelId's.

We can do the same as here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L124

sapkra, 2019-01-14:
I think the double and long click events are also missing for the single switch. Should I also add them to this device?

https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L148