Koenkk/zigbee2mqtt #861: Multiple coordinators on one service

ABeumer525, 2019-01-14:
Unsupported device
First check: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

ABeumer525, 2019-01-15:
Two reasons:
1. Multiple devices at first
2. Having a seperate group that when power fails still works. Modem and pi are battery backed up and in case of power outage I will still be notified about someone entering the house.

henriz, 2019-01-16:
Within a single network. As I understand him, he doesnt neccesarily want a single network, he wants multiple networks. And this is perfectly possible.

The easiest way to accomplish this is first to set different channels for each coordinator. Then make sure only one coordinator is ON and only the devices that need to pair to that coordinator. Pair them. Then switch it all of. Switch on the second coordinator and all the devices you want paired to it .Pair them, switch it all off.

And so forth. And before switching them all on, disable pairing on all of them.

After this you have 3 coordinators with 3 separate networks.

khzd, 2019-01-16:
Thanks Henriz for this feedback
I have a network on channel 26, so I can use the channel 11 to update my Xiaomi devices if needed without shutdown the primary one ?

henriz, 2019-01-16:
yes you can, however, you'll have to bind the devices to the network on channel 11, and then rebind them to your normal network.

I do roughly the same thing to send OTA updates to my hue stuff using the hue hub.

ABeumer525, 2019-01-16:
@henriz you are totally getting it.

ABeumer525, 2019-01-21:
You guys gave me an idea. On updating everything I tried to re-pair everything in a certain order. It looks like I only need one coordinator now. Everything that is alarm critical I have paured first and then I started pairing routers and other stuff.

closed 2019-03-29 by stale[bot]

wimpie007, 2020-02-07:
it is not a silly question in some cases...
My network is becoming to big, and there is an area in my house (actually the car-garage) where i always have reception troubles. (wife shutting down some zigbee lights doesn't help to get there a stable netwerk either)
I would opt to run a second zigbee network for those reasons. However, maintaing one zigbee2mqtt instance would be perfect. (instead of running a second instance on another pi)
a cc2530+esp -> socat would be perfect. 
(now i'm testing tasmota zigbee with the cc2530, but integrating it in HA is a pain in the ^&*)

jd1900, 2020-12-15:
I would like a second network as testing, so it would be great to have support. 
But right now I'm using a virtual machine in my PC to achieve this.

Char-r, 2021-12-09:
Wanted to chime in on that as well.

I have a house with huge walls and even with 5 IKEA extenders I still cannot go through and reach the coordinator. I purchased some Zigstar Ethernet coordinators/routers in the hope that I could set-up multiple networks for various areas of the house but then realised this was not feasible with Z2M :(

Another solution I had in mind and wondering about, Would it be feasible to have a router connect back via Ethernet to Z2M with the ethernet acting as another Zigbee route? (and thus making it transparent to the Zigbee Protocol since as far as it is concerned it would just be another router?)

thanks

Char-r, 2021-12-15:
@Koenkk would adding multiple TCP coordinators (so multiple independant networks managed within one Z2M interface) be feasible or are there technical considerations that would make this impossible?

Koenkk, 2021-12-15:
@Char-r that would be a fundamental change to the zigbee-herdsman design (not seeing that happen tbh)

Char-r, 2021-12-15:
> @Char-r that would be a fundamental change to the zigbee-herdsman design (not seeing that happen tbh, just too much work)

Ok thanks. I was hoping not having to run and maintain multiple RPI instances but seems like it is the only solution. 

Not sure if there is an option to do this on github but I would be willing to sponsor the development of such a feature if it ever was something that might become a possibility in the future.