Koenkk/zigbee2mqtt #864: Osram Classic A60 RGBW (AA69697) - can't set state

chettai, 2019-01-14:
I got the bulb paired:
```
Jan 15 00:30:22 veettamma npm[16818]:   zigbee2mqtt:info 1/15/2019, 12:30:22 AM MQTT publish: topic 'homeassistant/light/0x8418260000cb0bc7/light/config', payload '{"brightness":true,"color_temp":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/bedroom_light_rgbw/set","state_topic":"zigbee2mqtt/bedroom_light_rgbw","availability_topic":"zigbee2mqtt/bridge/state","name":"bedroom_light_rgbw_light","unique_id":"0x8418260000cb0bc7_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x8418260000cb0bc7","name":"bedroom_light_rgbw","sw_version":"Zigbee2mqtt 1.0.1","model":"Classic A60 RGBW (AA69697)","manufacturer":"OSRAM"}}'
```
light is defined as in the documentation (lights.yaml):
```
 -  platform: "mqtt"
    name: "bedroom_light_rgbw_light"
    unique_id: "0x8418260000cb0bc7_light_zigbee2mqtt"
    state_topic: "zigbee2mqtt/bedroom_light_rgbw"
    command_topic: "zigbee2mqtt/bedroom_light_rgbw/set"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    schema: "json"
```

But in HA the bulb is showed always as OFF. If tried to set the state, it goes back to off.
The log shows the following:

```
Jan 15 00:40:47 veettamma npm[16818]:   zigbee2mqtt:info 1/15/2019, 12:40:47 AM Zigbee publish to '0x8418260000cb0bc7', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 15 00:40:48 veettamma npm[16818]:   zigbee2mqtt:info 1/15/2019, 12:40:48 AM MQTT publish: topic 'zigbee2mqtt/bedroom_light_rgbw', payload '{"state":"ON"}'
```

I don't know what's wrong.
Can someone point me to the right direction?

khzd, 2019-01-15:
Hi
You need to post a log with debug option.

chettai, 2019-01-15:
> Hi
> You need to post a log with debug option.

After turning the light in HA to on : 

```
Jan 15 21:09:41 veettamma npm[14683]:   zigbee2mqtt:debug 1/15/2019, 9:09:41 PM Received MQTT message on 'zigbee2mqtt/bedroom_light_rgbw/set' with data 'ON'
Jan 15 21:09:41 veettamma npm[14683]:   zigbee2mqtt:info 1/15/2019, 9:09:41 PM Zigbee publish to '0x8418260000cb0bc7', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 15 21:09:41 veettamma npm[14683]:   zigbee2mqtt:info 1/15/2019, 9:09:41 PM MQTT publish: topic 'zigbee2mqtt/bedroom_light_rgbw', payload '{"state":"ON"}'
Jan 15 21:09:41 veettamma npm[14683]:   zigbee2mqtt:debug 1/15/2019, 9:09:41 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}}' of device 'Classic A60 RGBW' (0x8418260000cb0bc7)
Jan 15 21:09:41 veettamma npm[14683]:   zigbee2mqtt:info 1/15/2019, 9:09:41 PM MQTT publish: topic 'zigbee2mqtt/bedroom_light_rgbw', payload '{"state":"ON"}'
```

Koenkk, 2019-01-15:
What HA version are you running?

chettai, 2019-01-15:
> What HA version are you running?

Home Assistant 0.75.3

Koenkk, 2019-01-16:
From the zigbee2mqtt side, everything looks OK.

Are you using manual or discovery configuration?

chettai, 2019-01-16:
> From the zigbee2mqtt side, everything looks OK.
> 
> Are you using manual or discovery configuration?

The discovery is set to true.
But I have added the bulb manually in lights.yaml according to the documentation.

Koenkk, 2019-01-16:
When using discovery, this is not needed, can you remove this?

chettai, 2019-01-18:
> When using discovery, this is not needed, can you remove this?

I have removed the entry in lights.yaml. But the result is more or less the same:
```
Jan 18 21:10:41 veettamma npm[1510]:   zigbee2mqtt:debug 1/18/2019, 9:10:41 PM Received MQTT message on 'zigbee2mqtt/bedroom_light_rgbw/set' with data 'ON'
Jan 18 21:10:41 veettamma npm[1510]:   zigbee2mqtt:info 1/18/2019, 9:10:41 PM Zigbee publish to '0x8418260000cb0bc7', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 18 21:10:41 veettamma npm[1510]:   zigbee2mqtt:info 1/18/2019, 9:10:41 PM MQTT publish: topic 'zigbee2mqtt/bedroom_light_rgbw', payload '{"state":"ON"}'

```

Koenkk, 2019-01-18:
Can you provide the complete log?

chettai, 2019-01-19:
> Can you provide the complete log?

The log when I switch the light on. Afterwards the switch goes of OFF

HA log:
```
Jan 19 10:45:58 veettamma hass[330]: 2019-01-19 10:45:58 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: service_data=entity_id=light.bedroom_light_rgbw_light, service=turn_on, domain=light>
Jan 19 10:45:58 veettamma hass[330]: 2019-01-19 10:45:58 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]>
Jan 19 10:45:58 veettamma hass[330]: 2019-01-19 10:45:58 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: service_data=payload=ON, topic=zigbee2mqtt/bedroom_light_rgbw/set, retain=False, qos=0, service=publish, domain=mqtt>
Jan 19 10:45:58 veettamma hass[330]: 2019-01-19 10:45:58 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]>
```
zigbee2mqtt log:
```
Jan 19 10:45:58 veettamma npm[1510]:   zigbee2mqtt:debug 1/19/2019, 10:45:58 AM Received MQTT message on 'zigbee2mqtt/bedroom_light_rgbw/set' with data 'ON'
Jan 19 10:45:58 veettamma npm[1510]:   zigbee2mqtt:info 1/19/2019, 10:45:58 AM Zigbee publish to '0x8418260000cb0bc7', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Jan 19 10:45:58 veettamma npm[1510]:   zigbee2mqtt:info 1/19/2019, 10:45:58 AM MQTT publish: topic 'zigbee2mqtt/bedroom_light_rgbw', payload '{"state":"ON"}'
```

Koenkk, 2019-01-19:
> > What HA version are you running?
> 
> Home Assistant 0.75.3

Could you update to the latest version? (0.85.1)

chettai, 2019-01-19:
After updating to version 0.86.0b0 it is working now.
Thanks for all of your support.

closed 2019-01-19 by chettai