Koenkk/zigbee2mqtt #867: Can't pair SmartThings STS-PRS-251

battosai30, 2019-01-15:
Hi,
I can't pair my new STS-PRS-251. I press the button until LED stops being ON, I got a "Pairing" status in status log or MQTT but even after a long time it's never paired. Does anyone else get this problem ?

My zigbee2mqtt is up to date and I check that the tag is well present in device.js. I got no problem with Xiaomi or Ikea Tradfri devices.

Regards

battosai30, 2019-02-18:
Hi, some updates : I have not the reason, but changing to another platform solved the problem :s (both were fresh install but ...).
I noticed another thing : pairing sometimes does not work. I got so much better results removing battery, push button while placing agaain the battery. 
Last thing : reported battery voltage seems to have a problem. I checked with a voltmeter, a real 2,74V gives 2000mV and a real 3,28V gives 3035mV.
Thanks

Koenkk, 2019-02-18:
great!

closed 2019-02-18 by Koenkk