Koenkk/zigbee2mqtt #868: Get current version from Home Assistant

itajackass, 2019-01-15:
I'd like to show version installed from HA frontend.

I've set a sensor like:

  - platform: file
    file_path: /opt/zigbee2mqtt/npm-shrinkwrap.json
    name: zigbee2mqtt_current_version
    value_template: "{{ value_json.version }}"

But I get "unknown" value.
No error from HA log.

I've set also in configuration.yaml (HA)

whitelist_external_dirs:
    - '/opt/zigbee2mqtt'

I don't know if is a problem with the json file or HA...

subzero79, 2019-01-16:
The instances published to ha discovery have their version in the payload so you can use any device
```
homeassistant/switch/0x00158d0001e7e42b/switch/config {"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/fan_olivia_e42b/set","state_topic":"zigbee2mqtt/fan_olivia_e42b","name":"fan_olivia_e42b_switch","unique_id":"0x00158d0001e7e42b_switch_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0001e7e42b","name":"fan_masterbedroom_e42b","sw_version":"Zigbee2mqtt 1.0.1","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}
```

Notice the sw_version key

itajackass, 2019-01-23:
Ok but it's not very elegant...i'll prefer find version by reading inside the file

closed 2019-03-31 by stale[bot]