Koenkk/zigbee2mqtt #869: disabling joining new device

norbgazo, 2019-01-16:
Hello from France

i have a problem to add new xiaomi sensor and I think  it's because I have the message "  zigbee2mqtt:info 2019-1-16 14:04:28 Zigbee: disabling joining new devices." ..... 
but why I have this message. Evrything seems OK, for me

Sorry for my bad English
Thank for your help
Norb

pi@raspberrypi:/opt/zigbee2mqtt $ npm start
> zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt
> node index.js
  zigbee2mqtt:info 2019-1-16 14:04:26 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-16.14-04-26'
  zigbee2mqtt:info 2019-1-16 14:04:27 Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
  zigbee2mqtt:info 2019-1-16 14:04:27 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-16 14:04:27 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-16 14:04:27 Coordinator firmware version: '20181024'
  zigbee2mqtt:info 2019-1-16 14:04:27 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-1-16 14:04:27 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-1-16 14:04:27 Allowing new devices to join.
  zigbee2mqtt:warn 2019-1-16 14:04:27 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-1-16 14:04:27 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-1-16 14:04:27 Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 2019-1-16 14:04:28 zigbee-shepherd ready
  zigbee2mqtt:info 2019-1-16 14:04:28 Connected to MQTT server
  zigbee2mqtt:info 2019-1-16 14:04:28 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-1-16 14:04:28 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[]}'
  zigbee2mqtt:info 2019-1-16 14:04:28 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2019-1-16 14:04:28 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":false}'

-----------------------------------------------------------------------------

# Required: Home Assistant integration (MQTT discovery)
homeassistant: false
# Required: allow new devices to join
permit_join: true
# Required: MQTT settings
mqtt:
  # Required: MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # Required: MQTT server URL
  server: 'mqtt://localhost:1883'
  # Optional: MQTT server authentication user
  user:
  # Optional: MQTT server authentication password
  password:
  # Optional: MQTT client ID
  client_id: 'MY_CLIENT_ID'
  # Disable self-signed SSL certificates
  reject_unauthorized: true
  # Optional: Include device information to mqtt messages (default: false)
  include_device_information: true

khzd, 2019-01-16:
Salut norb,
Can you share your configuration.yaml? Seems that permit_join is set to false
Bien à toi 
Khaled

henriz, 2019-01-16:
@khzd, in his logs:
zigbee2mqtt:warn 2019-1-16 14:04:27 permit_join set to true in configuration.yaml.

norbgazo, 2019-01-16:
thank you Khaled and Henriz for your help

I think too that it should be in a mqtt client but where I can Check this setting?

khzd, 2019-01-16:
Hi
It happen to me with HA :(
When HA put it in false, I had to reboot to have it in true :(

Does your automation system put it in false?
What do you use as home auto ?

henriz, 2019-01-16:
I use Domoticz, but that also sets allow join to true as default. Which is the correct default setting.

Domoticz also requires a restart for it to work correctly.

You can also just set it on the commandline using the mosquitto tools.

norbgazo, 2019-01-16:
I use Domoticz but I don't know where in Domotics I can find this setting ... I am searching and i'll tell you if i've found something
thanks

henriz, 2019-01-16:
In domoticz you find it under settings, hardware, then the zigbee-mqtt device, at the bottom 'zigbee pairing'.

However, domoticz does not pick up on changes in the network without a restart, forcing it to repoll zigbee2mqtt. This is also why you don't see a just paired device in domoticz.

norbgazo, 2019-01-16:
YYYYYYYeeeeeeesssss
It was so simple .... my problem was in field 'zigbee pairing' in domoticz/settings/hardware but why is it on False at the installation?
Thank you very much for your help ......

khzd, 2019-01-17:
Bon courage

closed 2019-01-17 by norbgazo