Koenkk/zigbee2mqtt #870: aqara temperature/humidity sensor round value for auto discovery

h4nc, 2019-01-16:
The temperature and the humidty value of the aqara temperature sensor is reported with two decimal places. 

I think this sensor has an accuracy around 0.5 degrees, so it does make sense to show two decimal places. 

Sensor name: WSDCGQ11LM

`value_template: "{{ value_json.humidity | round(0) }}"`

This would round the reported data to a plausible value.

khzd, 2019-01-16:
Hi
You can set the precision in configuration.yaml for all devices or for a specific one, see the doc

h4nc, 2019-01-16:
I couldn't find it in the docs.

Not here 
https://koenkk.github.io/zigbee2mqtt/configuration/device_specific_configuration.html
https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html

Could you point to the right url or show an example?

h4nc, 2019-01-16:
@Koenkk is there a way to set the precision in the configuration.yaml file of zigbee2mqtt?

ryanbeaton, 2019-01-16:
In the dev branch only at the moment

https://github.com/Koenkk/zigbee2mqtt/issues/708#issuecomment-451907453

Koenkk, 2019-01-17:
@khzd yes, next release will be 1.1, planned for 30 January if all firmware stability tests are OK.

h4nc, 2019-01-17:
```
devices:
  '0xID1':
    friendly_name: Zigbee Humidity
    retain: false
    device_options: 
      humidity_precision: 0

```
Do I have to format it like this?

Also I have a motion sensor set up like this:

```
  '0xID2':
    friendly_name: Zigbee Bewegung WZ
    occupancy_timeout: 60
```
Should I change it to this?

```
  '0xID2':
    friendly_name: Zigbee Bewegung WZ
    device_options:
     occupancy_timeout: 60
```