Koenkk/zigbee2mqtt #871: remove entity doesn't work

h4nc, 2019-01-16:
I have an osram ledstripe and I wanted to use it with zigbee2mqtt. Unfortunately this did not work.

See: https://github.com/Koenkk/zigbee2mqtt/issues/774

In the meanwhile I moved my hue bridge and now it works (connected to hue). 

In home assistant integration (mqtt) I still see this light as an entity. Also it is shown in states.

I looked in the database.db file and configuration file in the share section. But nothing there anymore, I already deleted those.

In the config section of ha I went to .storage and looked for the entity in `core.entity_registry` and `core.device_registry` and also deleted the entity. 

I tried setting autodiscovery on and off. I restartet HA.

I tried removing the entity with `zigbee2mqtt/bridge/config/remove` and the friendly name shown in state. 

But all of that doesn't help, the entity still shows up. 

How can I delete it?

Koenkk, 2019-01-16:
You need to remove the retained MQTT messages of your MQTT broker.

h4nc, 2019-01-16:
I use mosquitto as an addon in home assistant, can you give me some advice how to do that?

h4nc, 2019-01-16:
Thanks for your advice. Had to reinstallthe mosquitto addon and relink the integration mqtt.

closed 2019-01-16 by h4nc