Koenkk/zigbee2mqtt #872: Unable to start latest-dev

timota, 2019-01-16:
Hi, 

when trying to run latest-dev docker image got error.

Command to run docker is:
docker run -d --restart unless-stopped -v /apps/zigbee2mqtt-data-dev:/app/data --device=/dev/ttyACM0 --name zigbee2mqtt-edge koenkk/zigbee2mqtt:latest-dev

> zigbee2mqtt@1.0.1 start /app
> node index.js

  zigbee2mqtt:info 2019-1-16 18:57:22 Logging to directory: '/app/data/log/2019-01-16.18-57-22'
  zigbee2mqtt:info 2019-1-16 18:57:22 Starting zigbee2mqtt version 1.0.1 (commit #28e3288)
  zigbee2mqtt:info 2019-1-16 18:57:22 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-16 18:57:25 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
/app/node_modules/q/q.js:155
                throw e;
                ^

TypeError: Cannot read property 'close' of undefined
    at shepherd.start (/app/lib/zigbee.js:45:47)
    at /app/node_modules/q/q.js:2059:17
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
    at process._tickCallback (internal/process/next_tick.js:181:9)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-16T18_57_25_938Z-debug.log

No matter what firmware flashed on CC2531.

How it can be fixed, as i would like to test Group functionality. 

Thanks

Koenkk, 2019-01-16:
- stop zigbee2mqtt
- unplug CC2531
- plug CC2531 
- press reset button on CC2531 (the one close to the usb port, green light should go off)
- start zigbee2mqtt

timota, 2019-01-16:
it works. Many thanks. 

Issue can be closed.

closed 2019-01-16 by timota