Koenkk/zigbee2mqtt #873: make Zigbee interface agnostic from MQTT

apiel, 2019-01-16:
This library is really cool but I don't really want to use MQTT. However, zigbee2mqtt help me a lot for my own implementation to use the zigbee-shepherd. But since I am also using your zigbee-shepherd-converters library, I always have to look to your changes to be able to update my code with your last changes.
Would be great to extract the Zigbee logic from zigbee2mqtt in another library, to be able to use it with something else than MQTT, for example to make a REST api. In my case I don't even expose my zigbee coordinator https://github.com/apiel/mhb-js/blob/master/src/zigbee/zigbee.js . As you can see my code, is mainly copy paste from your code. Ideally it would be awesome to just use a part of your library. I hope you can take this comment in conciderantion for future refactoring. I might even give you some help for that... (just need to find the time ^^)

Koenkk, 2019-01-17:
I hope that the zigbee-shepherd-converters API will get more stable now. Most of the zigbee logic is already in zigbee-shepherd, I think the rest can be quite application specific.

apiel, 2019-01-18:
I still think that you have some logic, we could re-use. I am thinking about to make a zigbee API in typescript with [nestjs](https://nestjs.com/). With using this framework, I will have no choice to extract the zigbee logic in some services, that I could then share in some external packages. But let see if I find the motivation :p

closed 2019-01-18 by apiel

apiel, 2019-01-19:
Yesterday, I was playing a bit https://github.com/apiel/zigbee-api
I copy a bit of your readme and a lot of your code... It's still need lot of refactoring, typing, (maybe unit test). For the moment I was able to pair the Ikea outlet and switch it on/off with the rest API. Now I have to see how I could handle event, cause rest it's not really intended for that ^^. After that, if I am still motivated, I will implement web socket and graphql interface.