Koenkk/zigbee2mqtt #876: Device with modelID 'undefined' is not supported.

Kryzek, 2019-01-17:
I paired Trust Wireless Contact Sensor with Zigbee2MQTT and it seems to be paired nicely.

I get state information nicely from the device but there is also error "Device with modelID 'undefined' is not supported." every time it's state changes.
Ex:
```
  zigbee2mqtt:debug 1/17/2019, 5:17:47 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'undefined' (0x00124b0014d9d7f3)
  zigbee2mqtt:warn 1/17/2019, 5:17:47 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 1/17/2019, 5:17:47 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 1/17/2019, 5:17:47 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cca9aa)
  zigbee2mqtt:info 1/17/2019, 5:17:47 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cca9aa', payload '{"contact":true,"linkquality":0}'
```

This is kind of weird since the device "0x00124b0014d9d7f3" is my CC2531 USB stick. When I change state my other Zigbee-device (Airam Led bulb) it does not trigger this warning.

JLFN, 2019-01-17:
I get the same problem with my Heiman Smoke detector after i updated to 1.0.1 It worked with 0.9 flawless.

Koenkk, 2019-01-17:
Duplicate of #801, I mentioned a probable solution here: https://github.com/Koenkk/zigbee2mqtt/issues/801#issuecomment-452831847

closed 2019-01-17 by Koenkk