Koenkk/zigbee2mqtt #877: uninstall

norbgazo, 2019-01-17:
Hello from France
After 3/4 days to try and test a lot of things, I havegot some problems and I prefer to restart with a new and fresh version . 
Is There somewhere a list to uninstall all and erase in a good condition all Zigbee2mqtt and components in my rapsberry?. I am the very best of newbe of the north of France and my english is bad but I learn :-)

Thanks for your help

Koenkk, 2019-01-17:
`rm -rf /opt/zigbee2mqtt`

norbgazo, 2019-01-19:
> Thank you for your help Koen ....
rm -rf /opt/zigbee2mqtt is not ok for me, I have a message that I have not permit?

norbgazo, 2019-01-19:
No it's OK  .... i am stupid sorry ..... :-(
thank you very much for your help

closed 2019-01-19 by Koenkk

sebi5361, 2020-06-04:
Maybe https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html could be updated with the way to uninstall zigbee2mqtt (`rm -rf /opt/zigbee2mqtt`) + the way to remove zigbee2mqtt systemctl settings...

spidgrou, 2021-08-09:
Hi,
I remove as describe above and also remove zigbee2mqtt service.... but I'm still seeing in my MQTT zigbee2mqtt publish.
How can I stop everything?

Thanks

Jacksonbm1, 2022-07-16:
> Hi, I remove as describe above and also remove zigbee2mqtt service.... but I'm still seeing in my MQTT server zigbee2mqtt publish. How can I stop everything?
> 
> Thanks

I know this is a closed topic, but this should explain why you still are seeing them appear in your mqtt: https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466

> . You follow all the normal procedures to delete an entity yet it always gets ‘re-discovered’. Why? Because when its ‘discovery’ payload was published, it was published as a retained message. That’s the part where you guessed correctly that, somehow, mosquitto was storing it. That’s the very definition of a ‘retained message’, namely it’s a message (payload) that is retained (stored) by the MQTT broker (mosquitto).