Koenkk/zigbee2mqtt #878: Unable to pair battery devices using max_devices firmware

philhawthorne, 2019-01-17:
I recently had to re-flash my stick with the firmware again after an error. This time, I decided to use the alternate max_devices firmware, here: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/alternatives/max_devices/bin

This firmware is great. I've noticed that I am able to pair bulbs from everywhere in my house, without needing to move the Pi which houses zigbee2mqtt closer. This tells me that devices are connecting to zigbee2mqttt via their nearest neighbors, which are other Hue bulbs.

However, I am unable to get any Xiaomi battery devices to pair with either a router, or even the controller itself. For example, right next to the Pi I have a `MCCGQ11LM` door/window sensor. After factory resetting it, and then clicking the button, it never pairs with zigbee2mqtt. There are also no logs in zigbee2mqtt to denote that a device is incoming.

However, I can easily pair a Hue lightbulb (I was able to pair `7299760PH`, `433714`, `8718696449691` and `74282` with ease. 

After several attempts, I went back to the original firmware here: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

This firmware only allows one router, so I ensured the first device I connected was a centrally located Hue lightstrip. No problem. I was then also able to connect the previous devices above, and immediately I was able to start re-pairing my battery devices, including the `MCCGQ11LM` and `WXKG03LM` wireless switch. 

However, I now can't get all of my devices to pair without physically moving the Pi around the house. This once again tells me that things are only connecting through the one router, which is a shame.

The new max_devices firmware is great, but for the moment I can't get it to sync battery devices :(

Koenkk, 2019-01-18:
@glentakahashi do you recognise these issues?

glentakahashi, 2019-01-18:
I haven't used the newest firmware, and I have a slightly custom built version of the max_devices firmware that I built a long time ago, but I haven't had issues with any battery devices.
I have a lot of xiaomi ones like the wall switch, door sensors, etc. and they all pair fine.

This weekend I wanted to test out the dev branch and new firmware and see if I can get group support working, so will report back if I notice anything on the max_devices firmware

glentakahashi, 2019-01-19:
Just as an update, migrated to latest firmware and latest dev branch on @Koenkk 's max_devices branch, and battery devices and all other devices work fine for me

philhawthorne, 2019-01-20:
Thanks Glen!

Just out of curiosity, when you say "max_devices branch" is that different to this firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/alternatives/max_devices/bin

What order did you pair your devices in? 

As some background I was able to pair Philips Hue bulbs and Sylvania downlights, but nothing battery powered from Xiaomi. 

Did you pair a Xiaomi power socket by chance? Maybe I will try that next.

glentakahashi, 2019-01-20:
Correct, that is the exact version i just flashed. Can't remember the exact order, but it wasn't all lights -> all battery or vice versa, it was mostly random.

I didn't try a Xiaomi Socket, but beware those are routers not end devices, so they aren't suitable for use with the` max_devices` firmware. You could also try and and build your own firmware that still has higher max_devices but keeps max_routers at something like 5-6

philhawthorne, 2019-01-20:
Interesting.

So would a Philips Hue bulb be considered a router as well?

I was under the impression, that if I add a router to the network, then that would _increase_ the number of end devices I could add to my network, as they should (in theory) connect via the router. ( see https://github.com/Koenkk/Z-Stack-firmware/issues/23#issuecomment-452839123 )

sehraf, 2019-01-20:
Usually everything constantly powered like a bulb of plug is a router. And yes it increases the amount of devices you can add.

ryanbeaton, 2019-01-20:
I thought the purpose of the max devices firmware is the router limit is only 1. How the works with lights that are routers I don’t know but that could explain the weirdness you’re experiencing.

https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/CC2531/alternatives/max_devices/README.md

philhawthorne, 2019-01-20:
@ryanbeaton I think you might be right. I originally understood it as allowing more devices to link up, but didn't realize it was restricted to just one router.

I'm going to re-flash everything again, and try have things connect through their nearest router and see if that improves. For now I'll close this.

Cheers!

closed 2019-01-20 by philhawthorne

glentakahashi, 2019-01-20:
yea @philhawthorne the `max_devices` firmware was specifically designed for people who basically don't have routers and want to connect as many end devices as possible to the coordinator directly. I originally set it up because I exclusively use Sengled A19 lights, and xiaomi battery devices, of which none are routers.