Koenkk/zigbee2mqtt #879: NYCE Motion Sensor

simplextech, 2019-01-17:
I've gone through the docs on how to add a device, however I've had no luck with various combinations.  I do have the docs from NYCE if they would help?  I've been fighting to get this supported in deCONZ without any success.

Device Info:
NYCE Ceiling Motion Sensor (ZHA)
Model:  NCZ-3043-HA
Supports: occupancy, temperature, humidity

It will pair but I can't get it to enroll and configure properly.  

Here's the DB entry for it.

`{
  "id": 2,
  "type": "EndDevice",
  "ieeeAddr": "0x000d6f000e301ad2",
  "nwkAddr": 44816,
  "manufId": 4281,
  "manufName": "NYCE",
  "powerSource": "Battery",
  "modelId": "3043",
  "epList": [
    1
  ],
  "status": "online",
  "joinTime": 1547766455,
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 1026,
      "inClusterList": [
        0,
        1,
        3,
        32,
        1026,
        1029,
        1030,
        1280
      ],
      "outClusterList": [],
      "clusters": {
        "genBasic": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "genPowerCfg": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "batteryVoltage": 26,
            "batteryPercentageRemaining": 102,
            "batterySize": 4,
            "batteryQuantity": 2,
            "batteryAlarmMask": 0,
            "batteryVoltMinThres": 24,
            "batteryVoltThres1": 0,
            "batteryVoltThres2": 0,
            "batteryVoltThres3": 0,
            "batteryAlarmState": 0
          }
        },
        "genIdentify": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "identifyTime": 0
          }
        },
        "genPollCtrl": {
          "dir": {
            "value": 1
          },
          "attrs": {}
        },
        "msTemperatureMeasurement": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "measuredValue": 2070,
            "minMeasuredValue": -1000,
            "maxMeasuredValue": 8500,
            "tolerance": 40
          }
        },
        "msRelativeHumidity": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "measuredValue": 4200,
            "minMeasuredValue": 0,
            "maxMeasuredValue": 10000,
            "tolerance": 600
          }
        },
        "msOccupancySensing": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "occupancy": 0,
            "occupancySensorType": 0,
            "pirOToUDelay": 10,
            "pirUToODelay": 0,
            "pirUToOThreshold": 1
          }
        },
        "ssIasZone": {
          "dir": {
            "value": 1
          },
          "attrs": {
            "zoneState": 0,
            "zoneType": 13,
            "zoneStatus": 0,
            "iasCieAddr": "0x0000000000000000",
            "zoneId": 255
          }
        }
      }
    }
  },
  "_id": "dhFcER30shcQuCJD"
}`

simplextech, 2019-01-18:
I've done that.  Here's a couple entries I tried.  Neither worked.

For starters I really only care about the motion, the temp/humidity are bonus.  I tried the below and it did not configure,  I also tried another with temp and it would not configure either.

`{
        zigbeeModel: ['3043'],
        model: 'NCZ-3043-HA',
        vendor: 'NYCE',
        description: 'Ceiling Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [fz.ias_zone_motion_dev_change, fz.ias_zone_motion_status_change],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
    },`

Another try....

`{
        zigbeeModel: ['3043'],
        model: 'NCZ-3043-HA',
        vendor: 'NYCE',
        description: 'Ceiling Motion Sensor',
        supports: 'occupancy and temperature',
        fromZigbee: [
            fz.generic_temperature, fz.ignore_temperature_change, fz.ias_zone_motion_dev_change,
            fz.ias_zone_motion_status_change,
        ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, 1000, 0, cb),
            ];
            execute(device, actions, callback);
        },
    },`

Pointers on what next?  Starting with simple and expand with battery, temp, humidity?  I'm new to Zigbee development and ZCL most of my gear is z-wave.

khzd, 2019-01-18:
Ok you can focus on only motion, look on device.js a motion sensor ie Xiaomi 
What is your log without debug ?

simplextech, 2019-01-18:
A basic devices.js entry without a configure and it doesn't join.

standard log info is below. 

`1/17/2019, 9:52:56 PM - warn: Message without device!
1/17/2019, 9:52:58 PM - warn: Message without device!
1/17/2019, 9:53:00 PM - warn: Message without device!
1/17/2019, 9:53:00 PM - info: Device incoming...
1/17/2019, 9:53:00 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
1/17/2019, 9:54:14 PM - warn: Message without device!
1/17/2019, 9:54:20 PM - warn: Message without device!
1/17/2019, 9:54:23 PM - warn: Message without device!
1/17/2019, 9:54:25 PM - warn: Message without device!
1/17/2019, 9:54:56 PM - warn: Message without device!
1/17/2019, 9:54:57 PM - warn: Message without device!
1/17/2019, 9:54:59 PM - warn: Message without device!
1/17/2019, 9:55:01 PM - warn: Message without device!
1/17/2019, 9:55:04 PM - warn: Message without device!
1/17/2019, 9:55:04 PM - info: Device incoming...
1/17/2019, 9:55:04 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
1/17/2019, 9:56:21 PM - warn: Message without device!`

khzd, 2019-01-18:
did you creat an entry in from zigbee : NCZ-3043-HA ?

khzd, 2019-01-18:
Ok so the config section is probably ok. Let’s keep it in the device.js

What is the log ?

simplextech, 2019-01-18:
Log was in the previous comment.  Nothing striking.  This was without debug.

simplextech, 2019-01-18:
Here's some new log info with configure

zigbee2mqtt:info 1/17/2019, 11:06:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 1/17/2019, 11:06:34 PM New device with address 0x000d6f000e301ad2 connected!
  zigbee2mqtt:info 1/17/2019, 11:06:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000e301ad2"}'
  zigbee2mqtt:error 1/17/2019, 11:06:35 PM Failed to configure 0x000d6f000e301ad2 0x000d6f000e301ad2

khzd, 2019-01-18:
Hi
With « PM Failed to configure » means your config section is ko
You may need to bind only motion and with debug mode see witch section is incorrect.
Debug=* ....

closed 2019-03-26 by stale[bot]