Koenkk/zigbee2mqtt #880: Pair xbee-PRO S2B as router

makuser, 2019-01-18:
Having an xbee joining the PAN yields this:

```
  serialport:poller received "readable" +7s
  serialport:bindings read +7s
  serialport:unixRead Starting read +7s
  serialport:unixRead Finished read 17 bytes +0ms
  serialport:main binding.read finished +7s
  cc-znp { sof: 254,
  cc-znp   len: 12,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'tcDeviceInd',
  cc-znp   payload:
  cc-znp    { nwkaddr: 59254, extaddr: '0x0013a20040000001', parentaddr: 0 },
  cc-znp   fcs: 91,
  cc-znp   csum: 91 } +7s
  serialport:main _read reading +8ms
  serialport:bindings read +10ms
  serialport:unixRead Starting read +10ms
  cc-znp:AREQ <-- ZDO:tcDeviceInd, { nwkaddr: 59254, extaddr: '0x0013a20040000001', parentaddr: 0 } +5ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +2ms
  serialport:unixRead waiting for readable because of code: EAGAIN +3ms
  serialport:poller Polling for "readable" +15ms
  serialport:poller received "readable" +33ms
  serialport:bindings read +37ms
  serialport:unixRead Starting read +34ms
  serialport:unixRead Finished read 18 bytes +1ms
  serialport:main binding.read finished +39ms
  cc-znp { sof: 254,
  cc-znp   len: 13,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'endDeviceAnnceInd',
  cc-znp   payload:
  cc-znp    { srcaddr: 59254,
  cc-znp      nwkaddr: 59254,
  cc-znp      ieeeaddr: '0x0013a20040000001',
  cc-znp      capabilities: 142 },
  cc-znp   fcs: 78,
  cc-znp   csum: 78 } +41ms
  serialport:main _read reading +9ms
  serialport:bindings read +11ms
  serialport:unixRead Starting read +10ms
  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 59254, nwkaddr: 59254, ieeeaddr: '0x0013a20040000001', capabilities: 142 } +5ms
  zigbee-shepherd spinlock: false [] +2ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +3ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 59254, nwkaddrofinterest: 59254 } +1ms
  serialport:main _write 9 bytes of data +11ms
  serialport:bindings write 9 bytes +11ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +5ms
  serialport:unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +7s
  serialport:unixRead waiting for readable because of code: EAGAIN +14ms
  serialport:poller Polling for "readable" +25ms
  serialport:unixWrite write returned null 9 +2ms
  serialport:unixWrite wrote 9 bytes +0ms
  serialport:unixWrite Finished writing 9 bytes +1ms
  serialport:main binding.write write finished +6ms
  serialport:poller received "readable" +10ms
  serialport:bindings read +13ms
  serialport:unixRead Starting read +11ms
  serialport:unixRead Finished read 6 bytes +0ms
  serialport:main binding.read finished +10ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'nodeDescReq',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 102,
[... 269 lines cut ...]
  cc-znp      numinclusters: 0,
  cc-znp      inclusterlist: <Buffer >,
  cc-znp      numoutclusters: 0,
  cc-znp      outclusterlist: <Buffer > },
  cc-znp   fcs: 235,
  cc-znp   csum: 235 } +44ms
  serialport:main _read reading +11ms
  serialport:bindings read +12ms
  serialport:unixRead Starting read +12ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 59254, status: 0, nwkaddr: 59254, len: 8, endpoint: 232, profileid: 49413, deviceid: 0, deviceversion: 0, numinclusters: 0, inclusterlist: <Buffer >, numoutclusters: 0, outclusterlist: <Buffer > } +5ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +1ms
  serialport:unixRead waiting for readable because of code: EAGAIN +9ms
  serialport:poller Polling for "readable" +23ms
  zigbee-shepherd Device: 0x0013a20040000001 join the network. +26ms
  zigbee2mqtt:debug 2019-1-17 23:57:08 Received zigbee message of type 'devIncoming' with data '"0x0013a20040000001"' of device 'undefined' (0x0013a20040000001)
  zigbee2mqtt:info 2019-1-17 23:57:08 Device incoming...
  zigbee2mqtt:info 2019-1-17 23:57:08 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2019-1-17 23:57:08 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-1-17 23:57:08 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 2019-1-17 23:57:08 Received zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0x0013a20040000001)
  zigbee2mqtt:warn 2019-1-17 23:57:08 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-1-17 23:57:08 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
```

I was trying to get a buffer like here: https://github.com/Koenkk/zigbee2mqtt/issues/516#issuecomment-432199639, so I used `DEBUG=* npm start`, to find out why the modelID is undefined.

closed 2019-03-26 by stale[bot]

makuser, 2019-08-27:
@thompsa how did you manage to pair it? What's the exact procedure you did to have the S2B to join the network?

thompsa, 2019-08-28:
I have paired my XBee-PRO S2B as a router in zigbee2mqtt and it is working fine. It shows up as an unknown device but that does not seem to affect operation.

The settings I used were
ZS = 2
NJ = 5A
NO = 3
EE = 1
EO = 1
SN = AF0
SP = 64

Its the blue box on the networkmap.

[image]

thompsa, 2019-08-28:
I put zigbee2mqtt in paring mode and then just powered up the Xbee again. It found the network and joined it. I had the PanID unset on the Xbee so it will join any network. You can then refresh the operating network parameters in XCTU to ensure AI,OP,OI,CH show its connected and on the right network.

thompsa, 2019-08-28:
I am using the firmware XBP24-ZB, ZigBee Router API mode, version 23A7

bernardolm, 2019-08-29:
Unfortunately, I could not use my XBee Pro S2B yet, @thompsa...

With `data/configuration.yaml`:
```yml

# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

advanced:
  rtscts: false
  log_level: debug
  report: true
```
When I try start zigbee2mqtt I get some errors:
```bash
01:02 in zigbee2mqtt on master [✘!] ➜ DEBUG=* npm start

> zigbee2mqtt@1.5.1 start /home/bernardo/workspaces/bernardolm/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 8/29/2019, 1:02:25 AM Logging to directory: '/home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/log/2019-08-29.01-02-25'
  zigbee2mqtt:debug 8/29/2019, 1:02:25 AM Removing old log directory '/home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/log/2019-08-29.00-28-49'
  zigbee2mqtt:debug 8/29/2019, 1:02:25 AM Loaded state from file /home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 8/29/2019, 1:02:25 AM Saving state to file /home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 8/29/2019, 1:02:25 AM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 8/29/2019, 1:02:25 AM Starting zigbee-shepherd
  zigbee2mqtt:debug 8/29/2019, 1:02:25 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/database.db","coordBackupPath":"/home/bernardo/workspaces/bernardolm/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":false}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +2ms
  serialport/stream _write 1 bytes of data +0ms
  serialport/binding-abstract write 1 bytes +3ms
  serialport/stream _read reading +0ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +1ms
  serialport/bindings/unixWrite wrote 1 bytes +0ms
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms
  serialport/stream binding.write write finished +2ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +2ms
  serialport/bindings/poller Polling for "readable" +3ms
  cc-znp The serialport /dev/ttyACM0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  unpi:writer --> frame [254,0,33,2,35] +0ms
  serialport/stream _write 5 bytes of data +1s
[... 79 lines cut ...]
  zigbee2mqtt:error 8/29/2019, 1:03:39 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 8/29/2019, 1:03:39 AM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/home/bernardo/workspaces/bernardolm/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/home/bernardo/workspaces/bernardolm/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 8/29/2019, 1:03:39 AM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/bernardo/.npm/_logs/2019-08-29T04_03_39_778Z-debug.log
```
I'm trying archive with this setup:

[image]

Searching on internet I realized that my XBee (XBP24BZ7, XBee Pro S2**B**) isn't same yours (XBP24-ZB, XBee Pro **S2** [without **B**]) but I can't believe that this is a problem, the firmware version 23A7 is the same.

Do you have any idea?
And you are using only zigbee2mqtt or it's together with Hassio, ioBroker, Domoticz...?

Thanks by support!

thompsa, 2019-08-29:
Oh, You can't use a XBee for the zigbee2mqtt controller.  You need to use a CC2531, see here
 https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html

I am using my XBee-PRO S2B as a router (ie zigbee repeater) to increase the range of the CC2531. If you want to use a XBee as the coordinator then you need to use the zha component on Home Assistant. https://www.home-assistant.io/components/zha/

bernardolm, 2019-08-29:
😞 I understand, @thompsa... I was believing that you and @makuser are using some XBee shield instead CC2531. Now I did understand how is your setup.

So I'll read these links and about how to use XBee as coordinator.

Just one more question, I'll need one XBee as coordinator **and another one** as router? Or **just one** as coordinator is enought to connect to my XBee devices like Xiaomi Aqara sensor?

Again, thank you so much by help!

makuser, 2019-08-29:
@bernardolm you can not use an XBee as coordinator with zigbee2mqtt. For that if you are using zigbee2mqtt with Home Assistant, you could to use the `zha` component directly or some other solution.

One thing is for sure: The XBee platform is not supported by zigbee2mqtt. You'll have to use a software relying on for example https://github.com/zigpy/zigpy-xbee to use the XBee.

What were you planning to use the XBee with? Home Assistant? Domoticz?

bernardolm, 2019-08-29:
Thank you so much, @makuser.

Ok, I did understand about can't use XBee directly with zigbee2mqtt. I will read and try one these another approachs to do that.

About the app I planning to use, I can say that I will use any that this setup works. I don't have preference, today I don't use any of them.

jpjodoin, 2019-10-12:
@thompsa :
I'm trying to get some XBee Pro S3B to act as router. I installed the latest firmware for those, then I set the CR (Routing/Messaging Mode to standard). 

I tried a few things
1) Set the PAN ID to 0 in XCTU, press on the join button on the coordinator, reset the device.
2) Set the ID (Network Id) to the same value as pan_id in my configuration.yaml + set the Encryption Enable (EE) and KY (AES Encyption Key) to the network_key value of configuration.yaml. I also tried to align my settings with yours for those that exist in my version (S3B):
NO = 3
SP = 64
SN = AF0
EE = 1

I would appreciate if you have any more insight/idea on why it might not work. Also, on your side, did you provide the PanId or did your Xbee was able to join automatically ?

Thanks!

wvolz, 2020-12-06:
Make sure you set the KY setting to "5A6967426565416C6C69616E63653039". That was the setting that made everything work for me.