Koenkk/zigbee2mqtt #882: set device class in configuration.yaml (zigbee2mqtt)

h4nc, 2019-01-18:
Most of my aqara contact sensors are mounted to windows. With auto discovery all of them show up as doors in home assistant. 

I know that it's possible to set the device_class in customize.yaml, but I'd like to know if this could also be done in the configuration.yaml of the addon (zigbee2mqtt). 

something like this? I did not find something like that in the docs. If its not implemented yet, will it be in future?

```
  '0x00ID':
    friendly_name: Zigbee window SZ
    device_class: window
    retain: false
```

Koenkk, 2019-01-18:
The problem is that e.g. a Xiaomi weather sensor will publish 4 sensors (battery, temperature, humidity, pressure) and therefore the above configuration will not work. Of course this is not impossible, but I don't see a need to do it as it can already be override via `customize.yaml`.

h4nc, 2019-01-18:
ok, thanks just wanted to know if there is that way too.

closed 2019-01-18 by h4nc