Koenkk/zigbee2mqtt #883: Linkquality of routers doesn't show as graph

h4nc, 2019-01-18:
I noticed that the linkquality isn't shown as graph in home assistant. Looks like homeassistant takes the state as a string because every number gehts its on color in a bar (see picture).

[image]

Koenkk, 2019-01-18:
Thanks, fixed

closed 2019-01-18 by Koenkk

h4nc, 2019-01-19:
I have zigbee2mqtt edge. Do I myself have to change something too fix that in HA, but for now I see it like before.

Koenkk, 2019-01-19:
Update to latests edge and restart home assistant should be enough.

h4nc, 2019-01-19:
I still get not graph. Do I have to delete the old data from the database?

h4nc, 2019-01-20:
After reinstalling edge the linkquality shows as graph.

But now I noticed that it says 
Linkquality: x RSSI 

RSSI isn‘t a unit and also we get values for linkquality AND RSSI

Koenkk, 2019-01-21:
So in order to have the graph shown the sensor should have a unit, what would be a good unit for the linkquality?

h4nc, 2019-01-21:
I see, so we need to set a unit so that the graph is shown. Isn't there a way to set it unitless.

There are some value in physics that don't have a unit like the Mach-number.

If it's not possible to set it a unitless number than your solution to set it as RSSI is good enough until there is a way to set it unitless.
Maybe something like that should be implemanted in ha, unless it it isn't already.

Do you knwo which units are allowed in ha?

Koenkk, 2019-01-21:
@h4nc there isn't really a unit, https://www.avrfreaks.net/comment/1162131#comment-1162131

Koenkk, 2019-01-21:
yes, updated to `-`

h4nc, 2019-01-21:
I updated edge and the unit is gone in the attributes but I still get the unit RSSI in the state.

[image]
[image]

Koenkk, 2019-01-21:
Did you restart home assistant?

h4nc, 2019-01-21:
Yes, I did. Tried it again, still RSSI

MarsWarrior, 2019-01-22:
> Yes, I did. Tried it again, still RSSI

I think you have to remove the sensor from home assistants device registry in the .strorage folder. You should finf the rssi sensor in one or more of the core.* files.

Remove entry and restart home assistant!

Koenkk, 2019-01-22:
A build issue prevented the build from regenerated, should be up in an hour.

h4nc, 2019-01-22:
Nice, thanks for letting us know.