Koenkk/zigbee2mqtt #884: QBKG03LM does not find route after some time installed

hanross2323, 2019-01-18:
Hi

Aqara Wall Switch (no neutral - double key) loses routing after some time correctly connected to zigbee2mqtt.

`  zigbee2mqtt:error 2019-1-18 13:33:32 Zigbee publish to '0x00158d0002c03208', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 3 failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.`

Coordinator version:

`Coordinator firmware version: '20181024'`

Zigbee2mqtt 1.0.1

Koenkk, 2019-01-18:
When having `log_level` `debug` (https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html) do you see `Check online` logging of this device?

brtwrs, 2019-01-22:
I'm having the same problems as @hanross2323. I do see "Check online" logging for this device in my logs

2019-1-22 09:24:12 - debug: Check online 0x00158d0001660bfe 0x00158d0001660bfe
2019-1-22 09:37:13 - debug: Check online 0x00158d0001660bf9 0x00158d0001660bf9

Both my switches worked perfect for a few days

Im on your latest dev branch btw

Koenkk, 2019-01-23:
@brtwrs what's the linkquality of the device, could you also post the networkmap?

brtwrs, 2019-01-24:
They dont show linkquality. But there are mulitple IKEA lightbulbs in the same room which are always working so I guess range is not the problem.

Here is my networkmap:

`{
node[shape=record];
  "0x00124b0009eb0446" [label="{0x00124b0009eb0446|Coordinator|No model information available|online}"];
  "0x00124b0009eb0446" -> "0x90fd9ffffe35aaf7" [label="76"]
  "0x00124b0009eb0446" -> "0xd0cf5efffe7a4825" [label="66"]
  "0x00124b001b6aec4e" [label="{0x00124b001b6aec4e|Router|No model information available|offline}"];
  "0x000b57fffed9c663" [label="{0x000b57fffed9c663|Router|IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (LED1536G5)|offline}"];
  "0x000b57fffed9c663" -> "0x00124b0009eb0446" [label="92"]
  "0x000b57fffed9c663" -> "0x000b57fffe9a3260" [label="50"]
  "0x000b57fffed9c663" -> "0xd0cf5efffe7a4825" [label="14"]
  "0x000b57fffe9a3260" [label="{0x000b57fffe9a3260|Router|IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (LED1649C5)|offline}"];
  "0x000b57fffe9a3260" -> "0x00124b0009eb0446" [label="19"]
  "0x000b57fffe9a3260" -> "0x90fd9ffffe35aaf7" [label="94"]
  "0x000b57fffe9a3260" -> "0xd0cf5efffe7a4825" [label="37"]
  "0x000b57fffeae4fba" [label="{0x000b57fffeae4fba|Router|IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (LED1624G9)|offline}"];
  "0x00158d00016da6c8" [label="{0x00158d00016da6c8|Router|Xiaomi Aqara double key wired wall switch (QBKG03LM)|online}"];
  "0x000b57fffe9a8c12" [label="{0x000b57fffe9a8c12|Router|IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)|offline}"];
  "0x000b57fffe9a8c12" -> "0x00124b0009eb0446" [label="97"]
  "0x000b57fffe9a8c12" -> "0x000b57fffe9a3260" [label="108"]
  "0x000b57fffe9a8c12" -> "0xd0cf5efffe7a4825" [label="66"]
  "0x00158d00016c4843" [label="{0x00158d00016c4843|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x000b57fffe9de3df" [label="{0x000b57fffe9de3df|Router|IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (LED1624G9)|offline}"];
  "0x000b57fffe9de3df" -> "0x00124b0009eb0446" [label="35"]
  "0x000b57fffe9de3df" -> "0x000b57fffe9a3260" [label="60"]
  "0x000b57fffe9de3df" -> "0x90fd9ffffe35aaf7" [label="166"]
  "0x000b57fffe9de3df" -> "0xd0cf5efffe7a4825" [label="91"]
  "0x000b57fffe9c2eb6" [label="{0x000b57fffe9c2eb6|Router|IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (LED1624G9)|offline}"];
  "0x000b57fffe9c2eb6" -> "0x00124b0009eb0446" [label="85"]
  "0x000b57fffe9c2eb6" -> "0x000b57fffe9a3260" [label="92"]
  "0x000b57fffe9c2eb6" -> "0x90fd9ffffe35aaf7" [label="148"]
  "0x000b57fffe9c2eb6" -> "0xd0cf5efffe7a4825" [label="90"]
  "0x000b57fffe48ba2a" [label="{0x000b57fffe48ba2a|Router|IKEA TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (LED1649C5)|offline}"];
  "0x000b57fffe48ba2a" -> "0x00124b0009eb0446" [label="69"]
  "0x000b57fffe48ba2a" -> "0x000b57fffe9a3260" [label="204"]
  "0x000b57fffe48ba2a" -> "0x90fd9ffffe35aaf7" [label="133"]
  "0x000b57fffe48ba2a" -> "0xd0cf5efffe7a4825" [label="60"]
  "0x000b57fffe3635d7" [label="{0x000b57fffe3635d7|Router|IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (LED1623G12)|offline}"];
  "0x000b57fffe3635d7" -> "0x00124b0009eb0446" [label="72"]
  "0x000b57fffe3635d7" -> "0x000b57fffe9a3260" [label="93"]
  "0x000b57fffe3635d7" -> "0x90fd9ffffe35aaf7" [label="62"]
  "0x90fd9ffffe676907" [label="{0x90fd9ffffe676907|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|offline}"];
  "0x90fd9ffffe676907" -> "0x000b57fffe9a3260" [label="49"]
  "0x90fd9ffffe676907" -> "0x90fd9ffffe35aaf7" [label="214"]
  "0x90fd9ffffe676907" -> "0xd0cf5efffe7a4825" [label="25"]
  "0x90fd9ffffe16ad09" [label="{0x90fd9ffffe16ad09|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x90fd9ffffe16ad09" -> "0x000b57fffe9a3260" [label="91"]
  "0x90fd9ffffe16ad09" -> "0x90fd9ffffe35aaf7" [label="255"]
  "0x90fd9ffffe16ad09" -> "0xd0cf5efffe7a4825" [label="118"]
  "0x90fd9ffffe35aaf7" [label="{0x90fd9ffffe35aaf7|Router|IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)|online}"];
  "0x90fd9ffffe35aaf7" -> "0x00124b0009eb0446" [label="36"]
  "0x90fd9ffffe35aaf7" -> "0x000b57fffe9a3260" [label="98"]
  "0x90fd9ffffe35aaf7" -> "0xd0cf5efffe7a4825" [label="105"]
  "0x000b57fffed740f7" [label="{0x000b57fffed740f7|Router|IKEA TRADFRI driver for wireless control (10 watt) (ICPSHC24-10EU-IL-1)|offline}"];
  "0x000b57fffea15468" [label="{0x000b57fffea15468|Router|IKEA TRADFRI driver for wireless control (10 watt) (ICPSHC24-10EU-IL-1)|offline}"];
  "0x000b57fffea15468" -> "0x00124b0009eb0446" [label="63"]
  "0x000b57fffea15468" -> "0x90fd9ffffe35aaf7" [label="91"]
  "0x000b57fffea15468" -> "0xd0cf5efffe7a4825" [label="136"]
  "0x90fd9ffffe29ea1e" [label="{0x90fd9ffffe29ea1e|Router|IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (LED1623G12)|offline}"];
  "0x90fd9ffffe29ea1e" -> "0x00124b0009eb0446" [label="55"]
  "0x90fd9ffffe29ea1e" -> "0x000b57fffe9a3260" [label="81"]
  "0x00158d0001df0140" [label="{0x00158d0001df0140|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x000b57fffe301837" [label="{0x000b57fffe301837|Router|IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (LED1623G12)|online}"];
  "0x000b57fffe301837" -> "0x000b57fffe9a3260" [label="39"]
  "0x000b57fffe301837" -> "0xd0cf5efffe7a4825" [label="0"]
[... cut ...]
  "0x00158d000238798f" [label="{0x00158d000238798f|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d0001209e60" [label="{0x00158d0001209e60|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"];
  "0x00158d0001d6be87" [label="{0x00158d0001d6be87|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d0001d6be8a" [label="{0x00158d0001d6be8a|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
}'`

Koenkk, 2019-01-26:
Let's continue in #943

closed 2019-01-26 by Koenkk