Koenkk/zigbee2mqtt #885: do not recognize Aqara Human Body sensor RTCGQ11LM

vpomax, 2019-01-18:
Pairing the Xiaomi Aqara Human Body sensor (RTCGQ11LM)
using Hassio 0.85.1 ... it show me the following error:

zigbee2mqtt:error 2019-1-18 15:32:57 Cannot get the Node Descriptor of the Device: 0x00158d0001e47d52 (Error: Timed out after 10000 ms)

Koenkk, 2019-01-18:
https://koenkk.github.io/zigbee2mqtt/getting_started/pairing_devices.html#xiaomi

vpomax, 2019-01-18:
ok I read it...
I click on the little pairing button of Aqara Human Body sensor... for 4 seconds... then it blink blue light 3 times. I release the button, it blink 1 time ... then wait 2 o 3 seconds and blink 3 times.
But it show the message: 
zigbee2mqtt:error 2019-1-18 15:53:57 Cannot get the Node Descriptor of the Device: 0x00158d0001de728b (Error: Timed out after 10000 ms)

JLFN, 2019-01-18:
> ok I read it...
> I click on the little pairing button of Aqara Human Body sensor... for 4 seconds... then it blink blue light 3 times. I release the button, it blink 1 time ... then wait 2 o 3 [...]

You have to continue to try until it pairs that message mean it havent been paired. Sometime Aqara takes  many tries to pair.

vpomax, 2019-01-18:
The working procedure is... for Mijia or Aqara device...
1) press pairing button for 4 second
2) release button
3) press button for one second and release
4) repeat many time point 3.....
in 10 o 20 seconds it will be recognized.

Koenkk, 2019-01-18:
Great!

closed 2019-01-18 by Koenkk

michel72, 2019-01-19:
Wel, not that great. I have exactly the same sensor, it pairs, but does not work:

```
 zigbee2mqtt:warn 2019-1-19 12:32:35 Failed to remove '0x00158d000200XXXX', trying force remove...
  zigbee2mqtt:info 2019-1-19 12:32:35 Successfully removed 0x00158d000200XXXX
  zigbee2mqtt:info 2019-1-19 12:32:35 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d000200XXXX"}'
  
  
  zigbee2mqtt:info 2019-1-19 12:35:09 New device with address 0x00158d000200XXXX connected!
  zigbee2mqtt:info 2019-1-19 12:35:09 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d000200XXXX"}'
  zigbee2mqtt:warn 2019-1-19 12:35:09 Device with modelID 'undefined' is not supported.
  
  "0x00158d000200bb12" [label="{0x00158d000200XXXX|EndDevice|No model information available|online}"];
```

Replaced the last 4 digits with XXXX

michel72, 2019-01-19:
Is has worked in previous versions btw.

chaertl1, 2019-01-21:
@michel72 
what coordinator hardware are you using ?
Is it a cc2530 ?

michel72, 2019-01-21:
Yes it is.  I just upgraded the firmware to 28-11-2018 and had to re-pair all devices. Now it works, but I had some real issues pairing other devices that previously where pairing without any problems.

chaertl1, 2019-01-21:
Which firmware version are you using now where it is working ?
There is already a topic on this issue.
#786

tringler, 2019-01-22:
I'm using a `CC2531` and faced to the same problem:
```
  zigbee2mqtt:warn 2019-1-21 23:43:32 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-1-21 23:43:32 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-1-21 23:43:39 Message without device!
  zigbee2mqtt:error 2019-1-21 23:45:28 Cannot get the Node Descriptor of the Device: 0x00158d00016db094 (Error: Timed out after 10000 ms)
```

cela96, 2019-02-12:
I have the same problem with `CC2531 `

```
zigbee2mqtt:warn 2019-2-12 09:12:52 Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 2019-2-12 09:12:52 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html``
```

cela96, 2019-02-12:
I have the same problem with `CC2531 `
Other zigbee devices work properly.

```
zigbee2mqtt:warn 2019-2-12 09:12:52 Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 2019-2-12 09:12:52 Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html``
```

Koenkk, 2019-02-12:
- Stop zigbee2mqtt
- Remove all entries of this device from `database.db`
- Start zigbee2mqtt
- Follow http://zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi carefully.

schumi2004, 2019-04-18:
I have it paired and i see in logging that it does something and detects movement but it's not available under Domoticz devices although i do see logging information in Domoticz aswell.
Any ideas?