Koenkk/zigbee2mqtt #886: Choose the right ZigBee network

leobel96, 2019-01-18:
Reading the guide and the how-to about how easy it is to connect a new ZigBee device to the zigbee2mqtt coordinator, a question flashes in my head: is it possible to choose, during first device connection, which ZigBee coordinator should the device use? How could I, for example, force the device to connect to my ZigBee coordinator and not to the one of my neighbour? Is there a protection against malicious ZigBee devices that an ill-intentioned could use to control my ZigBee devices?

Skeletorjus, 2019-01-18:
You typically tell the coordinator, be it zigbee2mqtt, Trådfri, Hue or whatever, to scan for incoming devices for a short amount of time. Once it's paired it cannot be controlled by another network.
If you somehow mange to pair something to another zigbee network, however unlikely it might be, you just need to reset the device to unpair it and it'll be removed from the network.

leobel96, 2019-01-18:
Yes but by default "permit_join: true". In my opinion this could led to conflicts or vulnerabilites if a user isn't warned about the risks. It should be false by default and a timer should automatically set it to false after some time for me.

closed 2019-01-19 by leobel96

leobel96, 2019-01-19:
Thank you, now it is clearer.