Koenkk/zigbee2mqtt #887: Installed latest DEV branch (for HA 0.85 compatibility) and now network map stopped working....

quarcko, 2019-01-18:
Dev branch with fix for 0.85 HA. Now when requesting network map in the debug console there is just:

  zigbee2mqtt:debug 2019-1-18 17:17:31 Received MQTT message on 'zigbee2mqtt/bridge/networkmap' with data 'graphviz'
  zigbee2mqtt:info 2019-1-18 17:17:31 Starting network scan...
  zigbee2mqtt:info 2019-1-18 17:17:42 Network scan completed

And nothing more... I always used to extract graphviz text from console, and now there is nothing...
bug or intended change?

JLFN, 2019-01-18:
it works for me,
Open two shells and first type on the first shell dont close it.
```.bash
mosquitto_sub -V mqttv311 -C 1 -u 'USERNAME' -P 'PASSWORD' -h 'HOSTNAME' -t zigbee2mqtt/bridge/networkmap/graphviz | sed -e 's/|EndDevice//' -e 's/|[^|]\+(.*)|/|/' -e 's/No [^|]\+|//' | sfdp -Tpng > /tmp/map.$(date +%Y%m%d%H%M%S).png
```
Type on second shell
```.bash
mosquitto_sub -V mqttv311 -C 1 -u 'USERNAME -P 'PASSWORD' -h 'HOSTNAME' -m "graphviz" -t zigbee2mqtt/bridge/networkmap
```

Chrischi-, 2019-01-18:
I think you mean this to publish a message:

`mosquitto_pub -V mqttv311 -u 'USERNAME -P 'PASSWORD' -h 'HOSTNAME' -m "graphviz" -t zigbee2mqtt/bridge/networkmap`

Skeletorjus, 2019-01-18:
I started experiencing the same issue a couple of days ago after reinstalling the dev branch of the hassio addon. Cant't see any commits that could be the culprit.
The network scan starts and finishes in one second without any graphviz output.

Koenkk, 2019-01-18:
Thanks for reporting, should be fixed in latest dev branch.

Koenkk, 2019-01-18:
@nerdosity can you provide the log?

quarcko, 2019-01-18:
I just reinstalled edge version of addon and still no network map...

⁣Sent from BlueMail ​

nerdosity, 2019-01-18:
I will try, I deleted the dev docker and made it again with non-dev to see if it was a problem with dev or with my configuragion. The crash disappeared.
Tomorrow I will fire up a docker from the dev version again.

Kryzek, 2019-01-19:
https://github.com/Koenkk/zigbee2mqtt/issues/887#issuecomment-455686203 - I was missing my map too (non HA-version), but with latest dev version I see map again.
```
digraph G {
node[shape=record];
  "0x00124b0014d9d7f3" [label="{0x00124b0014d9d7f3|Coordinator|No model information available|online}"];
  "0x00124b0014d9d7f3" -> "0x00158d0001cded1c" [label="25"]
  "0x00158d0001cded1c" [label="{0x00158d0001cded1c|Router|Airam LED OP A60 ZB 9W/827 E27 (4713407)|offline}"];
  "0x00158d0001cded1c" -> "0x00124b0014d9d7f3" [label="27"]
  "0x00158d0001b70237" [label="{0x00158d0001b70237|EndDevice|Airam CTR.U remote (can only be used to control the Airam 4713407 bulb) (AIRAM-CTR.U)|offline}"];
  "0x00158d0001b70237" -> "0x00158d0001cded1c" [label="68"]
  "0x00158d0001cca9aa" [label="{0x00158d0001cca9aa|EndDevice|Trust Wireless contact sensor (ZCTS-808)|online}"];
  "0x00158d0001cca9aa" -> "0x00158d0001cded1c" [label="68"]
}
```

Another question - Why does my router (Airam LED) seems to be offline in the map?

Koenkk, 2019-01-19:
@Kryzek probably because the device was sleeping while the network scan was executed

h4nc, 2019-01-20:
Had the same issue, reinstalling edge did solve it.

Mind that you might not be able copy and paste your config of the addon to the new version, as a new device section is added.

@Koenkk could you tell be what this is for? 

`  "devices": [],`

quarcko, 2019-01-21:
I also reisntalled one more time, and now it works
maybe i was too quick to install it first time and there was no fix atm.

Now it seems to report much richer map with devices mapping to more than one router at the time.
But can confirm that some devices that are working perfectly are shown without connection to any of the routers. (out of 30 devices, 6 are without connection, but working  okay.)
My setup involves only Xiaomi products (end devices and wall switches)

Also curiuos about this new option mentioned above.
I added it to config, and my old setup seems to be not altered and works okay.
But still what is it for?

Koenkk, 2019-01-22:
> Had the same issue, reinstalling edge did solve it.
> 
> Mind that you might not be able copy and paste your config of the addon to the new version, as a new device section is added.
> 
> @Koenkk could you tell be what this is for?
> 
> ` "devices": [],`

I'm not sure what you mean, the `devices`, it was always in (https://koenkk.github.io/zigbee2mqtt/configuration/device_specific_configuration.html)

h4nc, 2019-01-22:
In the config of the addon

had to add this line to get no error. 

[image]

Koenkk, 2019-01-22:
@h4nc that's really strange, it basically means no devices paired, @danielwelch do you recognize this?

h4nc, 2019-01-22:
@Koenkk I do have 12 devices paired, all of them work fine.

[image]

All I can say that I recently switched to edge, it's like two weeks ago and I did not have this devices[] section in that version. 
Updating edge to the newest version brought this section and I did notice it because my old configuration without devices[] wasn't accepted. Only with adding it to my config I was able to save the config and start the addon again.

Koenkk, 2019-01-22:
@danielwelch thanks, as this issue has been solved I will close this issue.

closed 2019-01-22 by Koenkk

h4nc, 2019-01-22:
> @Koenkk @h4nc yes, this is a new option to add device-specific config support to the add-on. The json entry he or she is talking about is new in a recent version of the add-on in the _add-on_ configuration, nothing new to zigbee2mqtt itself. With the way nested configuration works in Hassio (config schema validation, etc) the empty list will show up by default, and can be left empty. It may be possible to change this, but the documentation around nested config isn’t great.
> 
> If you’re not using device-specific config, you can leave it blank and ignore it.

Is there a documentation somewhere on how to use this (format)?

danielwelch, 2019-01-22:
@h4nc Not yet, as it's only in the edge version for now, but I'll update the documentation when it's released in the next stable build. You can infer the format for your config from [the schema](https://github.com/danielwelch/hassio-zigbee2mqtt/blob/master/zigbee2mqtt-edge/config.json#L43). Please test it out! If you have any problems, please open a new issue in the add-on repository.